AravisProject/aravis #672: arv-viewer doesn't render properly

tpwrules, 2022-05-10:
**Describe the bug**
When attempting to view a camera stream using the included viewer, the video shows up flickering and flashing under the UI elements, if it shows up at all, and desktop elements get "stuck" in the area the video would have shown as if it were never redrawn. When closing the view to go back to the list of cameras, the default white background does not show up and instead there are static of video behind the UI elements.

**To Reproduce**
Open the viewer and start a stream on a problematic platform.

**Expected behavior**
I expect the video to be viewable clearly when the stream is started and disappear completely when the stream is closed.

**Camera description:**
Any

**Platform description:**
Various,
including: Ubuntu 18.04 with Gnome desktop and NVidia drivers, Lubuntu 20.04 with LXQt desktop and Intel drivers (though much less severely), NixOS 21.05 with XFCE desktop and Mesa drivers on ARM.

**Additional context**
This is more or less a duplicate of #522 , but under a more searchable name. There are various screenshots there too.

EmmanuelP, 2022-05-11:
Please have a look at #345 

Anyway, even if it is not a dup of #345, please don't reopen this one, I'm going to rename #522.

closed 2022-05-11 by EmmanuelP

tpwrules, 2022-05-11:
Thank you for this clue. It looks like the viewer only works if there is a `gst-plugins-good` that has been compiled with the GTK3 video sink plugin. Perhaps this should be mentioned in the documentation somewhere.