AravisProject/aravis #684: FLIR GS3 parameter access modes appear wrong (GcAccessMode) - Can write to read-only nodes

nznobody, 2022-06-16:
**Describe the bug**
It appears possible that the `GcAccessMode` is being reported wrong

**To Reproduce**
Steps to reproduce the behavior:
```bash
arv-tool-0.8 --name "Point Grey Research-1E10010C45DC-010C45DC" description TriggerSource
        Enumeration  : [RO] 'TriggerSource'
Specifies the internal signal or physical input line to use as the trigger source. The selected trigger must have its TriggerMode set to On.
arv-tool-0.8 --name "Point Grey Research-1E10010C45DC-010C45DC" control TriggerSource=Line3
TriggerSource = Line3
arv-tool-0.8 --name "Point Grey Research-1E10010C45DC-010C45DC" control TriggerSource
TriggerSource = Line3
```

**Expected behavior**
I expect the node to be writable, and it is. However code that checks GcAccessMode will think it is not.
E.g. with a Lucid camera (GigE, desktop XEON host):
```
arv-tool-0.8 --name "Lucid Vision Labs-TRI023S-C-213902307" description TriggerSource
        Enumeration  : [RW] 'TriggerSource'
```

**Camera description:**
 - Point Grey Research FLIR GS3-U3-23S6C-C (USB3)

**Platform description:**
 - Aravis version: 0.8.21
 - OS: Debian GNU/Linux 11 (bullseye) (docker container)
 - Hardware: Raspberry Pi 4

**Attached files**
I've enabled debugging of what I thought would be useful and attached the outputs.
***FLIR***
[GS3-U3-23S6C-C-010C45DC-arv-tool-describe-TriggerSource.txt](https://github.com/AravisProject/aravis/files/8923342/GS3-U3-23S6C-C-010C45DC-arv-tool-describe-TriggerSource.txt)
```
export ARV_DEBUG=genicam:4,cp:4
vscode ➜ /src (feature/python-datacapture ✗) $ arv-tool-0.8 --name "Point Grey Research-1E10010C45DC-010C45DC" control TriggerSource &> GS3-U3-23S6C-C-010C45DC-arv-tool-describe-TriggerSource.txt
```
***Lucid***
[TRI023S-C-213902307-arv-tool-describe-TriggerSource.txt](https://github.com/AravisProject/aravis/files/8923343/TRI023S-C-213902307-arv-tool-describe-TriggerSource.txt)
```
export ARV_DEBUG=genicam:4,cp:4
vscode ➜ /src (feature/python-datacapture ✗) $ arv-tool-0.8 --name "Lucid Vision Labs-TRI023S-C-213902307" description TriggerSource &> TRI023S-C-213902307-arv-tool-describe-TriggerSource.txt
```

nznobody, 2022-08-10:
In interest of a possibly related bug, I am also having issues with using the `TriggerSoftware` with a `FLIR-Blackfly S BFS-PGE-31S4C` camera.

# Bug:
GigEVision write_register error when trying to use software trigger. This occurs manually via arv-tool or via code. However, arv-test correctly runs.
```bash
arv-tool-0.8 control TriggerSoftware
FLIR-Blackfly S BFS-PGE-31S4C-20201067 (10.45.100.97)
TriggerSoftware execute error: [TriggerSoftware] [TriggerSoftware_Val] GigEVision write_register error (access-denied)
```
And yet dumping GENICAM XML or features list shows `Command      : [WO] 'TriggerSoftware'`

`arv-test`:
```
arv-test-0.8 
Found 1 device
Testing 'FLIR:Blackfly S BFS-PGE-31S4C'
Genicam:Load                        SUCCESS 
Genicam:Schema                      IGNORED 
Properties:SensorSizeReadout        SUCCESS 
Properties:SensorSizeCheck          IGNORED 
Properties:GainReadout              SUCCESS 
Properties:ExposureTimeReadout      SUCCESS 
MultipleAcquisitionA:BufferCheck    SUCCESS 10/10
MultipleAcquisitionA:FrameRate      SUCCESS 10.00 Hz
SoftwareTrigger:BufferCheck         SUCCESS 5/5
MultipleAcquisitionB:BufferCheck    SUCCESS 10/10
MultipleAcquisitionB:FrameRate      SUCCESS 5.00 Hz
SingleAcquisition:BufferCheck       SUCCESS 
Chunks:BufferCheck                  SUCCESS 
GigEVision:NStreamChannels          SUCCESS 
GigEVision:StreamChannel            SUCCESS
```

# Expected Result:
Running on a different camera:
```
arv-tool-0.8 --name "Point Grey Research-1E1000E6C0FE-00E6C0FE" control TriggerSoftware
TriggerSoftware executed
```

# Support documents
[arv-tool-set-triggersoftware.log](https://github.com/AravisProject/aravis/files/9296733/arv-tool-set-triggersoftware.log)
[BFS-PGE-31S4C-20201067.values.log](https://github.com/AravisProject/aravis/files/9296735/BFS-PGE-31S4C-20201067.values.log)
[BFS-PGE-31S4C.xml.txt](https://github.com/AravisProject/aravis/files/9296738/BFS-PGE-31S4C.xml.txt)

EmmanuelP, 2022-08-10:
> In interest of a possibly related bug, I am also having issues with using the `TriggerSoftware` with a `FLIR-Blackfly S BFS-PGE-31S4C` camera.

It does not seem to be the same issue. If the software trigger is successful, it probably means the TriggerSource must be set to Software before being able to emit a software trigger.

EmmanuelP, 2022-08-10:
Please attach the Genicam data of the faulty camera.

nznobody, 2022-08-12:
I believe that this _other_ `(FLIR-Blackfly S BFS-PGE-31S4C)` issue may have been a camera bug. Several camera hard power restarts later it has stopped being an issue... (But the orginal issue with `FLIR GS3-U3-23S6C-C` remains.

EmmanuelP, 2022-08-22:
Please attach the Genicam data of the `FLIR GS3-U3-23S6C-C` camera, and the debug output of `arv-tool-0.8 control TriggerSource TriggerSource=Software`.

nznobody, 2022-08-22:
Ah yes! Please find attached with `export ARV_DEBUG=genicam:4,cp:4,device:4`

[GS3-U3-23S6C-15122686.triggersource.log](https://github.com/AravisProject/aravis/files/9398073/GS3-U3-23S6C-15122686.triggersource.log)
[GS3-U3-23S6C-15122686.genicam.xml.log](https://github.com/AravisProject/aravis/files/9398076/GS3-U3-23S6C-15122686.genicam.xml.log)
[GS3-U3-23S6C-15122686.values.yml.log](https://github.com/AravisProject/aravis/files/9398079/GS3-U3-23S6C-15122686.values.yml.log)

EmmanuelP, 2022-08-23:
Please test #708

commit 540529a6 (2022-08-24): gcstructentry: default to parent access mode property

Fix #684
commit 6a8c98ec (2022-09-07): gcstructentry: default to parent access mode property

Fix #684
closed 2022-09-07 by EmmanuelP