AravisProject/aravis #685: Access violation on Windows

Jean-PhilippeGoerke, 2022-06-22:
**Describe the bug**
When handling frame packets from the camera an access violation occurs in aravis. As long as packets are blocked by the firewall nothing happens, but as soon as it is turned off the error occurs.

**To Reproduce**
I build the library using the current MSYS2 version.

Here the meson logs:

$ meson build
The Meson build system
Version: 0.62.2
Source dir: ....../aravis
Build dir: ....../aravis/build
Build type: native build
Project name: aravis
Project version: 0.8.22
C compiler for the host machine: cc (gcc 12.1.0 "cc (Rev2, Built by MSYS2 project) 12.1.0")
C linker for the host machine: cc ld.bfd 2.38
C++ compiler for the host machine: c++ (gcc 12.1.0 "c++ (Rev2, Built by MSYS2 project) 12.1.0")
C++ linker for the host machine: c++ ld.bfd 2.38
Host machine cpu family: x86_64
Host machine cpu: x86_64
Compiler for C supports arguments -Wduplicated-branches: YES
Compiler for C supports arguments -Wimplicit-fallthrough: YES
Compiler for C supports arguments -Wmisleading-indentation: YES
Compiler for C supports arguments -Wstrict-prototypes: YES
Compiler for C supports arguments -Wunused: YES
Compiler for C supports arguments -Wunused-variable: YES
Compiler for C supports arguments -Wdeclaration-after-statement: YES
Compiler for C supports arguments -Wformat=2: YES
Compiler for C supports arguments -Wimplicit-function-declaration: YES
Compiler for C supports arguments -Winit-self: YES
Compiler for C supports arguments -Wmissing-include-dirs: YES
Compiler for C supports arguments -Wmissing-prototypes: YES
Compiler for C supports arguments -Wpointer-arith: YES
Compiler for C supports arguments -Wformat-nonliteral: YES
Compiler for C supports arguments -Wenum-conversion: YES
Compiler for C supports arguments -Wmaybe-uninitialized: YES
Compiler for C supports arguments -fvisibility=hidden: YES
Found pkg-config: C:\msys64\mingw64\bin/pkg-config.EXE (1.8.0)
Run-time dependency glib-2.0 found: YES 2.72.2
Run-time dependency gobject-2.0 found: YES 2.72.2
Run-time dependency gio-2.0 found: YES 2.72.2
Run-time dependency libxml-2.0 found: YES 2.9.14
Run-time dependency zlib found: YES 1.2.12
Found CMake: C:\msys64\mingw64\bin/cmake.EXE (3.23.2)
Run-time dependency libusb-1.0 found: NO (tried pkgconfig and cmake)
Library m found: YES
Library ws2_32 found: YES
Library iphlpapi found: YES
Program glib-mkenums found: YES (C:\msys64\mingw64\bin/glib-mkenums.EXE)
Program glib-mkenums found: YES (C:\msys64\mingw64\bin/glib-mkenums.EXE)
Program glib-mkenums found: YES (C:\msys64\mingw64\bin/glib-mkenums.EXE)
Program glib-mkenums found: YES (C:\msys64\mingw64\bin/glib-mkenums.EXE)
Found pkg-config: C:\msys64\mingw64\bin/pkg-config.EXE (1.8.0)
Program glib-compile-resources found: YES (C:\msys64\mingw64\bin/glib-compile-resources.EXE)
Configuring arvapi.h using configuration
Configuring arvfeatures.h using configuration
Configuring arvparamsprivate.h using configuration
Configuring arvversion.h using configuration
Program glib-compile-resources found: YES (C:\msys64\mingw64\bin/glib-compile-resources.EXE)
Program g-ir-scanner found: NO
Run-time dependency gobject-introspection-1.0 found: NO (tried pkgconfig and cmake)
Run-time dependency gtk+-3.0 found: NO (tried pkgconfig and cmake)
Run-time dependency gstreamer-base-1.0 found: NO (tried pkgconfig and cmake)
Run-time dependency gstreamer-app-1.0 found: NO (tried pkgconfig and cmake)
Run-time dependency gstreamer-video-1.0 found: NO (tried pkgconfig and cmake)
Run-time dependency gstreamer-base-1.0 found: NO (tried pkgconfig and cmake)
Run-time dependency gstreamer-app-1.0 found: NO (tried pkgconfig and cmake)
Dependency gi-docgen skipped: feature documentation disabled
[... 122 lines cut ...]
A version I build using version 0.8.21 is working fine (~15.03.2022). But this build was also based on an older version of MSYS2 and the dependend libraries like glib.

**Camera description:**
Self build in memory camera. It is working fine with old aravis and Pleora eBus Player, so I think this is not causing the issue.

**Platform description:**
 - Aravis version: Current state of the main branch
 - OS: Windows 10 21H2
 - Hardware: Dell Laptop / Intel i9-9980HK

**Additional context**
Here the stack trace from Visual Studio:
>	libaravis-0.8-0.dll!00007fff45aff5b4()	unknown (I assume it to be _process_packet)
 	libaravis-0.8-0.dll!00007fff45b00029()	unknown (I assume it to be _loop)
 	libglib-2.0-0.dll!00007fff45866ce1()	unknown
 	libglib-2.0-0.dll!00007fff458952b7()	unknown
 	msvcrt.dll!00007ff859e6af5a()	unknown
 	msvcrt.dll!00007ff859e6b02c()	unknown
 	kernel32.dll!00007ff858d27034()	unknown
 	ntdll.dll!00007ff859fa2651()	unknown

I assume that error is caused in the _process_packet method (arvgvstream.c), but I cannot debug the code.

Side question:
Is there a way to generate a .pdb file on windows so that debugging would be possible in Visual Studio?

EmmanuelP, 2022-07-07:
> Is there a way to generate a .pdb file on windows so that debugging would be possible in Visual Studio?

Try to change the buildtype to debugoptimized:

```
meson configure -Dbuildtype=debugoptimized
```

marcovtje, 2022-08-05:
I also tried 0.8.22 last week, and it crashed in arv_camera_start_acquisition (multi example). 0.8.20 did not,.  (camera: AVT manta 125g).

Not just own code, arv-camera-test-0.8.exe also  crashes.

EmmanuelP, 2022-08-05:
If you are building aravis using the git repository, could you try to find the faulty commit using `git bissect` ?

marcovtje, 2022-08-05:
No, not yet. I'm a Delphi/Lazarus user, not MSYS.

marcovtje, 2022-08-10:
I tried my luck with msys and built aravis 8.20, 8.21 and 8.22 for x86_64-mingw64 and 21 and 22 have segmentation fault in arv-camera-test and 20 not, all built with the same msys environment.

If you have detailed instructions for the bissecting, I'll give it a go

marcovtje, 2022-08-11:
Traceback of gdb for camera test for 0.8.21:

odel name             = Manta_G-125B
device serial number   = 50-0503319299
image width            = 1292
image height           = 964
horizontal binning     = 1
vertical binning       = 1
exposure               = 15000 µs
gain                   = 0 dB
payload                = 1245488 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 0 ns
[New Thread 12560.0x3134]
gv packet size         = 8228 bytes

Thread 12 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 12560.0x3134]
_loop (thread_data=thread_data@entry=0x1cf1b6924b0) at ../src/arvgvstream.c:855
855                                     frame = _process_packet (thread_data,
(gdb) bt
#0  _loop (thread_data=thread_data@entry=0x1cf1b6924b0)
    at ../src/arvgvstream.c:855
#1  0x00007ffb647ff666 in arv_gv_stream_thread (data=0x1cf1b6924b0)
    at ../src/arvgvstream.c:1111
#2  0x00007ffb67696ce1 in ?? () from C:\msys64\mingw64\bin\libglib-2.0-0.dll
#3  0x00007ffb676c52b7 in ?? () from C:\msys64\mingw64\bin\libglib-2.0-0.dll
#4  0x00007ffba46daf5a in msvcrt!_beginthreadex ()
   from C:\Windows\System32\msvcrt.dll
#5  0x00007ffba46db02c in msvcrt!_endthreadex ()
   from C:\Windows\System32\msvcrt.dll
#6  0x00007ffba42d7034 in KERNEL32!BaseThreadInitThunk ()
   from C:\Windows\System32\kernel32.dll
#7  0x00007ffba5562651 in ntdll!RtlUserThreadStart ()
   from C:\Windows\SYSTEM32\ntdll.dll
#8  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb)

mikkellars, 2023-01-23:
Any updates regarding this?
I have the same segmentation fault on windows 11 on Aravis version 8.21 and 8.22, but not on 8.20.

EmmanuelP, 2023-01-27:
Hi @mikkellars 

I don't have a development environment on Windows, so, as I said to @marcovtje , finding the commit that introduced the issue using git bisect  would help. It is quite straightforward to use: https://git-scm.com/docs/git-bisect

EmmanuelP, 2023-05-19:
Long time, no news. I'm closing this issue. Don't hesitate to reopen when you have the requested informations.

Cheers.

closed 2023-05-19 by EmmanuelP