AravisProject/aravis #687: Got Warpped Image From ArvBuffer. Using Latest Aravis on Windows10

YuriGao, 2022-07-07:
**Describe the bug**
I got image from ArvBuffer, and it looks like this.
[image]
And when i was changed the packet size of camera from 1448 to 1044, the image i got from ArvBuffer is like this.
[image]
The real image is like this.
[image]
It seems about some data transport issues.

**Camera description:**
 - Manufacturer: Hikvision
 - Model: MV-CE013-50GM
 - Interface: GigE

**Platform description:**
 - Aravis version: Latest
 - OS: Windows10
 - Hardware: x86_64

**Additional context**
Add any other context about the problem here.

YuriGao, 2022-07-07:
Those black dots is 4 pure black pixels. When i make packet size smaller, the image was warpped bigger.

EmmanuelP, 2022-07-07:
What is the output of `arv-tool-0.8 control GevSCPSPacketSize` ?

YuriGao, 2022-07-07:
When i make ArvBuffer larger, i got more data in back. And by coincidence, the count of none-zero pixels is same as expected payload, so i can rebuild the original image by removing the zero pixels. But i think there are some wrong within Aravis source code. This rebuild step should be in the Aravis.
The warpped image is as below. The image size is 970*1280 and i can rebuild a 960*1280 image by removing the zero pixels.
[image]

YuriGao, 2022-07-07:
Do you mean the arv-tool-0.8.exe? 
This is what i got when running the arv-camera-test-0.8.exe.
[image]

EmmanuelP, 2022-07-07:
Yes, `arv-tools-0.8.exe`

EmmanuelP, 2022-07-07:
The complete command is `arv-tool-0.8.exe control GevSCPSPacketSize`.

EmmanuelP, 2022-07-07:
Please paste a text copy of the output, not images.

YuriGao, 2022-07-07:
I got something like this:
arv-tool-0.8.exe control GevSCPSPacketSize
Hikvision-MV-CE013-50GM-00C17129039 (169.254.152.5)
GevSCPSPacketSize = 1444 min:220 max:9156 inc:8

EmmanuelP, 2022-07-07:
That means the camera expects a packet size with a value being the sum of 220 and a multiple of 8. Aravis does not enforce the increment (nor the range by default).

1448 is not valid. But 1044 is, which should have worked. Did you try other packet size values ?

YuriGao, 2022-07-07:
I try 1044 and some other valid value. In that case, i can got right image in Hikvison client. But i still cannot got right image with Aravis API and arv-viewer-0.8.exe, it seems about unaligned 4 byte data. I still cannot got right result.

YuriGao, 2022-07-07:
Thank you for your patience. I would try. If there is any update, I will update in this issue.

YuriGao, 2022-07-08:
When i use arv_camera_gv_set_packet_size() to set packet size as 1444, i got 1444. That's expected. But when i check the packet size in acquiration, i got the packet size as 1448. That's 4 bytes adding, exactly the black dot width in warpped image.
This is what got.

Found camera 'MV-CE013-50GM'
Present packet size is 1500
Set the packet size as 1444
present packet size is 1444
Acquired 1280 X 960 buffer------PixelFormat Mono8------Payload 1228800
Check the present packet size 1448

And when i set the packet size as 1448, the program catch a exception.
0x00007FFC8947A1F3 (libaravis-0.8-0.dll)处(位于 AravisTest.exe 中)引发的异常: 0xC0000005: 读取位置 0x00000000000020B4 时发生访问冲突。

When i check the disassemble code in VS,  the disassemble code shows as below.
00007FFC8947A1EB  mov         qword ptr [rbp+10h],rcx  
00007FFC8947A1EF  mov         rax,qword ptr [rbp+10h]  
00007FFC8947A1F3  movzx       eax,byte ptr [rax+4]  
00007FFC8947A1F7  shr         al,7  
00007FFC8947A1FA  movzx       eax,al 

In the code, i see [rax+4]. Automatically, i assume maybe those code causes the warpped image.
Please tell me what you think.

YuriGao, 2022-07-08:
When i use the arv-fake-gv-camera-0.8.exe, i set packet size as 1444 then i got packet size as 1444. Maybe that is incompatibility problem. I will try more camera brand to test this problem.

YuriGao, 2022-07-08:
I try another camera brand (Huaray Technology-A3124MG100-7L05FA6PAK00024), and everything works fine. It is incompatiblity problem.

EmmanuelP, 2022-07-08:
> Found camera 'MV-CE013-50GM'
> Present packet size is 1500
> Set the packet size as 1444
> present packet size is 1444
> Acquired 1280 X 960 buffer------PixelFormat Mono8------Payload 1228800
> Check the present packet size 1448

How do you check the packet size ?

YuriGao, 2022-07-08:
Use the arv_camera_gv_get_packet_size() API. First i invoke the arv_camera_gv_set_packet_size() & arv_camera_gv_get_packet_size() API before arv_camera_start_acquisition(), and i got 1444 result. After that, i invoke the arv_camera_gv_get_packet_size() API in dealing ArvBuffer part, then the things happened, i got 1448 result.

EmmanuelP, 2022-07-08:
If the camera changes the packet size by itself, there is not a lot we can do in Aravis. I would report this as an issue to the camera manufacturer.

I'm closing this issue as it does not seems Aravis is at fault. Feel free to reopen if you think there is something we can do, and don't hesitate to keep us informed if you have an answer from the camera manufacturer.

Thanks.

closed 2022-07-08 by EmmanuelP

YuriGao, 2022-07-08:
Problem solved by updating camera's firmware. Thank you for your precious time.