AravisProject/aravis #688: Feature Request: CLI to configure IP settings

romainreignier, 2022-07-07:
**Is your feature request related to a problem? Please describe.**

As mentioned here: https://github.com/AravisProject/aravis/pull/681#pullrequestreview-994512851
It would be nice to have a command line tool to configure the IP settings of a camera.

So I open an issue to track this idea.

**Describe the solution you'd like**
I might try to implement it.
Should we create a new tool or add this functionality to `arv-tool`?

The base options should be to:
* Set Mode (LLA, HCP or persistent IP)
* Set IP
* Set Submask
* Set default gateway

**Describe alternatives you've considered**
I am not aware of any open-source tool allowing to change these settings on Linux from the command line.
If there are some, I will be glad to know them.

EmmanuelP, 2022-07-07:
I think it is better to add this to arv-tool.

romainreignier, 2022-07-07:
Ok, I agree with that.
About the "API", can we do something like the control mode?
```
arv-tool-0.8 network mode=PersistentIP
arv-tool-0.8 network ip=192.168.0.1
```

romainreignier, 2022-07-07:
`arv-tool` uses `ArvDevice` and not an `ArvCamera` object.
But I will need to convert IP addresses from string to `GInetAddress` like it is done in `arv_camera`:

https://github.com/AravisProject/aravis/blob/af9a095479cce8cff22a538124732eb20bb3ed5f/src/arvcamera.c#L3049-L3082

Should I duplicate this code in `arvtool.c`, use an `ArvCamera` with `arv_camera_new_with_device()` or move this helper to `arvdevice.c`?

Actually, to use the function `arv_camera_gv_set_persistent_ip_from_string()` I need to have all three settings (ip, submask and gateway) at the same time, so the API will be:
```
arv-tool-0.8 network ip=192.168.0.1,255.255.255.0,192.168.0.254
```

Is this ok, or should we prefer to set only one setting at a time?

EmmanuelP, 2022-07-07:
Please move the helper to arvdevice.

It will be better to be able to set ip mask and gateway independently.  You can modify the helper to ignore the arguments if they are NULL.

closed 2022-07-13 by EmmanuelP