AravisProject/aravis #694: Conflict in Visual Studio Conan file

ville-v, 2022-07-19:
Using file https://github.com/AravisProject/aravis/blob/main/.github/workflows/aravis-msvc.yml

Some library expected libiconv/1.16 and some other expected libiconv/1.17 -> conflict

Adding `libiconv/1.17` to the Conan file fixed the conflict.

Another issue was with the Conan command line, setting "os" was expected by some library.

This command line worked: `conan install -s os="Windows" -s compiler="Visual Studio" -s compiler.version=17 -s arch=x86_64 -s build_type=Release -b pcre -b missing -b cascade -if build .`

EmmanuelP, 2022-07-19:
Thanks.

I have fixed the msvc CI file here: #696 

The CI pipeline is failing these days, when conan is trying to install gst-plugins-base, due to an incorrect pkg-config utility bundled in the PERL distribution.

https://github.com/AravisProject/aravis/runs/7408790534?check_suite_focus=true

By chance, do you know how to fix this issue ?

ville-v, 2022-07-20:
I had no pkg issue while running Conan, but I had pkg issue while compiling Aravis.

I fixed the issue by installing Pkg-config lite and adding it to PATH: https://sourceforge.net/projects/pkgconfiglite/files/

I actually didn't use the yml file but extracted the commands from it. Here are the steps:

I used C:\tmp because the command to compile Aravis did not work when the path had spaces.

Install pkg-config in C:\tmp\pkg-config\bin

Put Aravis source to C:\tmp\aravis

Run in C:\tmp:

```
conan install -s os="Windows" -s compiler="Visual Studio" -s compiler.version=17 -s arch=x86_64 -s build_type=Release -b pcre -b missing -b cascade -if build .
```

Run in C:\tmp

```
xcopy /Y build aravis\build\
copy /Y compile-aravis-step2.ps1 aravis\compile-aravis-step2.ps1
cd aravis
powershell .\compile-aravis-step2.ps1
```

compile-aravis-step2.ps1 runs in C:\tmp\aravis

```
.\build\activate.ps1
.\build\activate_build.ps1
.\build\activate_run.ps1

$Env:PATH = "C:\tmp\pkg-config\bin;$Env:PATH"

meson --prefix "%cd%\install" --buildtype release --pkg-config-path "%cd%\build" -Ddocumentation=disabled -Dgst-plugin=enabled -Dintrospection=disabled -Dusb=enabled -Dviewer=disabled -Dgv-n-buffers=1 . .\build
meson compile -C .\build -v
meson install -C .\build
```

Finally copy the files from C:\tmp\aravis\install to my project and copy Glib 2.0 from C:\.conan to my project.

commit 34120cba (2022-07-19): ci: fix possible conflict in libiconv requirements

Fix #694
commit 803dcedf (2022-07-19): ci: fix possible conflict in libiconv requirements

Fix #694
commit c059cf1b (2022-07-21): ci: fix possible conflict in libiconv requirements

Fix #694
closed 2022-07-21 by EmmanuelP