AravisProject/aravis #700: FLIR and TIS unable to receive images

ville-v, 2022-07-25:
**Describe the bug**
I am unable to get images with two cameras I tested, one from FLIR and one from TIS. Both cameras can take images in Sapera CamExpert.

I can:

* Find all cameras (using `arv_update_device_list()` and related functions)
* Connect to cameras with `arv_camera_new()`
* Set and read features

I cannot:

* Receive images

`arv_camera_start_acquisition()` returns no error. When I tried using buffer callback with `arv_camera_create_stream()`, the callback is only called the first time (init), but is not called for any new buffers.

When I used `arv_camera_acquisition()` instead, it does not return. Replicating the function in my own code, I found that `arv_stream_pop_buffer()` does not return.

I tried both with software trigger and without.

**Camera description:**
* FLIR:Blackfly S BFS-PGE-200S6M
* The Imaging Source Europe GmbH:DMK 33GX183
* Interface: Ethernet

**Platform description:**
 - Aravis version 0.8.22
 - OS: Windows 10, Visual Studio
 - x86_64

**Additional context**
FLIR:

`arv-test-0.8.exe -d all -t Single*`: [flir results.txt](https://github.com/AravisProject/aravis/files/9178494/flir.results.txt)
`arv-tool-0.8.exe features`: [flir features.txt](https://github.com/AravisProject/aravis/files/9178497/flir.features.txt)
`arv-tool-0.8.exe genicam`: [flir genicam.txt](https://github.com/AravisProject/aravis/files/9178500/flir.genicam.txt)

TIS:

`arv-test-0.8.exe -d all -t Single*`: [tis results.txt](https://github.com/AravisProject/aravis/files/9178501/tis.results.txt)
`arv-tool-0.8.exe features`: [tis features.txt](https://github.com/AravisProject/aravis/files/9178502/tis.features.txt)
`arv-tool-0.8.exe genicam`: [tis genicam.txt](https://github.com/AravisProject/aravis/files/9178504/tis.genicam.txt)

I captured communication from Sapera CamExpert with WireShark (image size 256x100):

[tis wireshark capture.zip](https://github.com/AravisProject/aravis/files/9178525/tis.wireshark.capture.zip)
[flir wireshark capture.zip](https://github.com/AravisProject/aravis/files/9178571/flir.wireshark.capture.zip)

EmmanuelP, 2022-07-25:
Hi,

It looks like no streaming packet are ever received. Did you succeed to use Aravis with another device ?

The reason could be a unusual network setup. Are the devices directly connected to the machine ? It could also be a firewall issue.

Did you try the fake camera ?

ville-v, 2022-07-25:
I have not succeeded to use Aravis with any camera. With fake camera, the tests pass.

The computer has two network cards. The camera is connected to the network card through PoE injector. I tried to connect to another network card, and got similar output (no packets received).

I also tried with another computer, which does not have any other camera software installed. Got same result (test failed).

How can I diagnose this further?

ville-v, 2022-07-26:
Looking at the Wireshark capture in the initial post, it seems streaming packets are received by the computer. But for some reason they are not received by Aravis.

[image]

EmmanuelP, 2022-07-26:
Are the port you are seeing in the wireshark capture the same that are listed in the debug output ?

ville-v, 2022-07-26:
I run a new capture during `arv-test`, and it reported:

[GvStream::stream_new] Destination stream port = 58565
[GvStream::stream_new] Source stream port = 3957

Ports in Wireshark:

to camera, 3956, GVCP
from camera, 58563, GVCP
from camera, 58564, UDP (image stream?)

I think the test might report incorrect port numbers, because it is still able to communicate with the camera.

However, in this capture there is no such continuous streak with UDP messages. I think the UDP messages in this run are from "try packet size" section. I'm not sure what changed.

I noticed that the UDP messages in the initial Wireshark logs were 1472 bytes, and Jumbo Packets were disabled. I enabled Jumbo Packets, and now Sapera CamExpert is using 8972-byte packages when receiving images. However, arv-test is still not receiving image stream.

ville-v, 2022-07-26:
I now switched to the FLIR camera. This time, the test passed and selected packet size 1400 (was 1500 in the initial post). I'm not sure if enabling Jumbo Packets made difference, as a small packet size was still selected.

However, the TIS camera still doesn't work.

Ports used by FLIR camera:

[GvStream::stream_new] Destination stream port = 60197
[GvStream::stream_new] Source stream port = 49153

Wireshark:

to camera 3956, GVCP
from camera 60195, GVCP
from camera 60196, UDP (during package size selection)
from camera 60197, UDP (image stream)

"Source stream port" 49153 does not have any communication in the log.

Test log and Wireshark log:
[flir successful capture.zip](https://github.com/AravisProject/aravis/files/9189297/flir.successful.capture.zip)

ville-v, 2022-07-26:
I am still having communication problem with FLIR camera in my own program, which connects to camera and calls `arv_camera_acquisition()` on it:

```
[14:39:19.190] ￰ﾟﾅﾸ interface> Found 0 USB3Vision device (among 4 USB devices)
[14:39:19.191] ￰ﾟﾅﾸ interface> [GvDiscoverSocket::new] Add interface 192.168.1.119 (192.168.1.255)
[14:39:19.191] ￰ﾟﾅﾸ interface> [GvDiscoverSocket::new] Add interface 192.168.100.100 (192.168.100.255)
[14:39:19.191] ￰ﾟﾅﾸ interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.255.255.255)
[14:39:19.194] ￰ﾟﾅﾸ interface> [GvInterface::discovery] Device 'FLIR-Blackfly S BFS-PGE-200S6M-19370932' found (interface 192.168.100.100) user_id '' - MAC '2c:dd:a3:27:93:b4'
[14:39:20.203] ￰ﾟﾅﾸ interface> Found 0 USB3Vision device (among 4 USB devices)
[14:39:20.204] ￰ﾟﾅﾸ device> [GvDevice::new] Interface address = 192.168.100.100
[14:39:20.204] ￰ﾟﾅﾸ device> [GvDevice::new] Device address = 192.168.100.192
[14:39:20.211] ￰ﾟﾅﾸ device> [GvDevice::load_genicam] xml url = 'Local:BFS-PGE-200S6M_01819DDA.zip;60000000;11E0F' at 0x200
[14:39:20.211] ￰ﾟﾅﾸ device> [GvDevice::load_genicam] Xml address = 0x60000000 - size = 0x11e0f - BFS-PGE-200S6M_01819DDA.zip
[14:39:20.383] ￰ﾟﾅﾸ device> [GvDevice::load_genicam] Zipped xml data
[14:39:20.384] ￰ﾟﾅﾸ dom> [GcFeature::set_attribute] Unknown attribute 'xmlns:xi'
[14:39:20.388] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.389] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.404] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.413] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.413] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.414] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.415] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.415] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.418] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.418] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.418] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.418] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.419] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.420] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.420] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.420] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.420] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.420] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.420] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.420] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.422] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.424] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.425] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.429] ￰ﾟﾅﾸ dom> [Genicam::create_element] Unknown tag (pAlias)
[14:39:20.430] ￰ﾟﾅﾸ genicam> [Gc::set_default_node_data] Add 'GevStreamChannelCount'
[14:39:20.430] ￰ﾟﾅﾸ genicam> [Gc::set_default_node_data] Add 'GevSCPAddrCalc'
[14:39:20.437] ￰ﾟﾅﾸ device> [GvDevice::new] Device endianness = big
[14:39:20.437] ￰ﾟﾅﾸ device> [GvDevice::new] Packet resend     = yes
[14:39:20.437] ￰ﾟﾅﾸ device> [GvDevice::new] Write memory      = yes
[14:39:20.437] ￰ﾟﾅﾸ device> [GvDevice::new] Legacy endianness handling = no
[14:39:20.474] ￰ﾟﾅﾸ device> [GvDevice::create_stream] Number of stream channels = 1
[14:39:20.572] ￰ﾟﾅﾸ device> [GvDevice::auto_packet_size] Try packet size = 9000
[14:39:20.666] ￰ﾟﾅﾸ device> [GvDevice::auto_packet_size] Try packet size = 4788
[14:39:20.761] ￰ﾟﾅﾸ device> [GvDevice::auto_packet_size] Try packet size = 2682
[14:39:20.857] ￰ﾟﾅﾸ device> [GvDevice::auto_packet_size] Try packet size = 1629
[14:39:20.953] ￰ﾟﾅﾸ device> [GvDevice::auto_packet_size] Try packet size = 1102
[14:39:21.047] ￰ﾟﾅﾸ device> [GvDevice::auto_packet_size] Try packet size = 839
[14:39:21.143] ￰ﾟﾅﾸ device> [GvDevice::auto_packet_size] Try packet size = 707
[14:39:21.238] ￰ﾟﾅﾸ device> [GvDevice::auto_packet_size] Try packet size = 641
[14:39:21.334] ￰ﾟﾅﾸ device> [GvDevice::auto_packet_size] Try packet size = 608
[14:39:21.428] ￰ﾟﾅﾸ device> [GvDevice::auto_packet_size] Try packet size = 592
[14:39:21.526] ￰ﾟﾅﾸ device> [GvDevice::auto_packet_size] Packet size set to 9000 bytes
[14:39:21.534] ￰ﾟﾅﾸ device> [GvStream::stream_new] Packet size = 9000 byte(s)
[14:39:21.538] ￰ﾟﾅﾸ stream> [GvStream::stream_new] Destination stream port = 60914
[14:39:21.538] ￰ﾟﾅﾸ stream> [GvStream::stream_new] Source stream port = 49153
[14:39:21.542] ￰ﾟﾅﾸ stream> [GvStream::loop] Standard socket method
[14:39:31.558] ￰ﾟﾅﾸ stream>     bins    ;frame_retent; packet_time;inter_packet
0;           0;           0;           0
-------------
>=      2000;           0;           0;           0
<          0;           0;           0;           0
min         :         n/a;         n/a;         n/a
max         :         n/a;         n/a;         n/a
last max at :           0;           0;           0
counter     :           0:           0:           0
[... 468 lines cut ...]
                         counter     :           0:           0:           0
[14:45:12.477] ­ƒà© stream> [GvStream::finalize] n_completed_buffers    = 0
[14:45:12.477] ­ƒà© stream> [GvStream::finalize] n_failures             = 0
[... cut ...]
[14:45:12.480] ­ƒà© stream> [GvStream::finalize] n_duplicated_packets   = 0
[14:45:12.480] ­ƒà© stream> [GvStream::finalize] n_transferred_bytes    = 0
[14:45:12.480] ­ƒà© stream> [GvStream::finalize] n_ignored_bytes        = 0
[14:45:12.480] ­ƒà© stream> [Stream::finalize] Flush 1 buffer[s] in input queue
[14:45:12.480] ­ƒà© stream> [Stream::finalize] Flush 0 buffer[s] in output queue
Chunks:BufferCheck                  FAILURE
GigEVision:NStreamChannels          SUCCESS
GigEVision:StreamChannel            SUCCESS
```

EmmanuelP, 2022-07-27:
I don't know what is going wrong. There is no obvious reason why the stream packets would sometime be received by aravis, and sometime not, except may be a firewall issue, which would depend on the port used by either aravis or the camera.

When the streaming works, it works almost fine, except in only receives 3 buffers.

ville-v, 2022-08-01:
I ended up using another library (Common Vision Blox).

closed 2022-08-01 by ville-v