AravisProject/aravis #703: Handing arv_camera_[get|set}_[gain|black_level] with multiple taps

hackel6, 2022-08-10:
Some cameras have multiple sensor taps. As an example, Imperx GEV1620 has 2 taps. Each sensor tap has its own Black_Level and Analog Gain settings.
Using the current API for get and set will only return or change the value currently set via the "[Gain|BlackLevel]Selector" register which, in this case, can read either "AnalogTap1" or "AnalogTap2."
Thus, using the get/set API as it is right now only changes _half_ the image.
The other way to change both, aside from the *Selector registers, is to use each register directly: "AnalogOffset_Ch1" and "AnalogOffset_Ch2" respectively as an example (corresponds to BlackLevel).
Obviously, one possibility would be for the set() method to check for a corresponding *Selector register and set each value to the requested one at a time. However, the issue arises when a get() is performed and the taps (or channels) are _not_ all the same.

I'm more than willing to implement a solution, but I don't have a good idea how to handle it.

Another possibility would be for arv_camera_is_[gain|black_level]_available() to return false in the case where there are multiple taps, but that doesn't quite seem right either since it IS available and the proper use is to set each corresponding *Selector value to get/set the requested channel.

EmmanuelP, 2022-08-22:
I don't think it is a good idea to 'hide' the different gains in the current get/set API. And that would probably lead to strange behaviours withs camera that have overlapping gains (like All, Red, Blue, Green).

 I would add the following API instead :

* arv_camera_select_gain()
* arv_camera_dup_available_gains()

hackel6, 2022-08-24:
This makes sense to me and I may have some time soon to implement.
Just one question: i understand that just like the other *_dup_* calls already in the API, arv_camera_dup_available_gains() will return the available gain registers (probably as enumerations or strings). But just out of curiosity, what does "dup" stand for?

EmmanuelP, 2022-08-24:
`dup` is for duplicate. It is intended to indicate the returned data must be freed, as opposed to the `get` functions.

hackel6, 2023-09-08:
closed with #745

closed 2023-09-08 by hackel6