AravisProject/aravis #704: gst pipeline closes (-5) without waiting for hardware trigger

lc-spxl, 2022-08-11:
**Describe the bug**
gst pipeline fails with error

``` 
[14:43:29.379] 🅸 device> [GvDevice::new] Device endianness = big
[14:43:29.379] 🅸 device> [GvDevice::new] Packet resend     = yes
[14:43:29.379] 🅸 device> [GvDevice::new] Write memory      = yes
[14:43:29.379] 🅸 device> [GvDevice::new] Legacy endianness handling = yes
[14:43:29.604] 🅸 device> [GvDevice::create_stream] Number of stream channels = 1
[14:43:29.625] 🅸 device> [GvDevice::auto_packet_size] Current packet size check successfull (1400 bytes)
[14:43:29.633] 🅸 device> [GvStream::stream_new] Packet size = 1400 byte(s)
[14:43:29.637] 🅸 stream> [GvStream::stream_new] Destination stream port = 55890
[14:43:29.637] 🅸 stream> [GvStream::stream_new] Source stream port = 49153
[14:43:29.671] 🅸 stream> [GvStream::loop] Packet socket method
[14:43:30.379] 🅳 device> [GvDevice::Heartbeat] Ack value = 2
[14:43:31.381] 🅳 device> [GvDevice::Heartbeat] Ack value = 2

ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis0:
streaming stopped, reason error (-5)

[14:43:31.811] 🅸 stream>     bins    ;frame_retent; packet_time;inter_packet
                                    0;           0;           0;           0
                         -------------
                         >=      2000;           0;           0;           0
                         <          0;           0;           0;           0
                         min         :         n/a;         n/a;         n/a
                         max         :         n/a;         n/a;         n/a
                         last max at :           0;           0;           0
                         counter     :           0:           0:           0
```

**To Reproduce**

The pipeline is  the following (I also tested it with a Hikrobot-MV-CA032-10GC)

`aravissrc camera-name="FLIR-Blackfly S BFS-PGE-23S3C-20136026" features="TriggerSelector=FrameBurstStart TriggerSource=Line0 TriggerMode=On" ! video/x-raw, format=RGB,width=960,height=600,framerate=15/1 ! jpegenc idct-method=2 quality=100 ! multipartmux boundary="myboundary" ! tcpserversink sync=false host=0.0.0.0 port=8000`

**Expected behavior**

Pipeline should wait until the trigger is activated

**Camera description:**

Camera 1

 - HIKROBOT
 - MV-CA032-10GC
 - eth
 
Camera 2

- HIKROBOT
- MV-CA032-10GC
- eth

**Platform description:**

 - Aravis version: built from commit 97fcd7032921facc00b041373336dde536ebd0a4
 - OS: Ubuntu 22.04
 - Hardware x86_64

**Additional context**

I would point to #528 but I have no clarity on how the associated PR is solving the issue

lc-spxl, 2022-08-11:
Sorry again, this is a format issue. My fault

lc-spxl, 2022-08-11:
Sorry reopening and update the initial descritpion with a working  and not working pipeline

lc-spxl, 2022-08-12:
From the logs seems I am hitting the timeout of 2s as defined here

https://github.com/AravisProject/aravis/blob/43b3ea413b1c726d12c4541b22f096bd882f00ed/gst/gstaravis.c#L311

```
0:00:00.390509069 11896 0x55ccb7514de0 DEBUG              aravissrc gstaravis.c:287:gst_aravis_set_caps:<aravis0> Frame rate = 15 Hz
0:00:00.411558950 11896 0x55ccb7514de0 DEBUG              aravissrc gstaravis.c:298:gst_aravis_set_caps:<aravis0> Buffer timeout = 2000000 µs
0:00:00.412404905 11896 0x55ccb7514de0 DEBUG              aravissrc gstaravis.c:301:gst_aravis_set_caps:<aravis0> Actual frame rate = 15 Hz
0:00:00.412870257 11896 0x55ccb7514de0 DEBUG              aravissrc gstaravis.c:315:gst_aravis_set_caps:<aravis0> Actual gain = 0
0:00:00.413333471 11896 0x55ccb7514de0 DEBUG              aravissrc gstaravis.c:329:gst_aravis_set_caps:<aravis0> Actual exposure = 10500 µs
0:00:00.569436813 11896 0x55ccb7514de0 LOG                aravissrc gstaravis.c:373:gst_aravis_set_caps:<aravis0> Start acquisition
```

EmmanuelP, 2022-08-12:
Closing as duplicate of #528

EmmanuelP, 2022-08-12:
Thanks for the report.

closed 2022-08-12 by EmmanuelP