AravisProject/aravis #707: Using gstreamer plugin from within docker container withou network_mode=host

aldoshkind, 2022-08-21:
**Is your feature request related to a problem? Please describe.**
Our project runs in docker container and uses gstreamer to receive and decode video. We are unable to use use network-mode=host, so we would like to be able to use aravis gstreamer plugin just exposing ports.

**Describe the solution you'd like**
Specify range of ports explicitly for aravis gstreamer plugin to use. Or make aravis gstreamer plugin access cameras by their IPs using TCP.

**Describe alternatives you've considered**
Alternative is to run aravis-based gstreamer pipeline in a separate service which provides rtsp server interface, but it looks complicated. Another alternative is not to use aravis, but I like aravis :)

**Additional context**
I saw issue here https://github.com/AravisProject/aravis/issues/509 . Have something changed since then?

EmmanuelP, 2022-08-22:
Hi,

There is multiple places in Aravis where connections are handled. I guess the listening for discovery packet answers is one of the problematic case, but may be not the only one.

Are you able to connect to a camera from the docker container using its IP address ?

nznobody, 2022-08-22:
Using `arv-tool` and -arv-camera-test` I got some interesting results that I will conclude later with some logs. My setup:

- subnet A with device X and camera W
- subnet B with device Y

Subnet A and B are corporate subnets in physically different cities but linked via corporate backbone (internal network).

From device X and Y I was able to use `arv-tool` with `--name "<ip-address>"` from the host machine and also within docker containers with normal networking (not `network_mode=host`). I was able to query and control the device.

However, when attempting to stream with `arv-camera-test` I had inconclusive results. I'll check back on this later. But, the control layer at least was functioning :)

I've attached the logs with some debug logging enabled for the `arv-tool` commands. Local is device X and remote is device Y.

[camera.local.control.log](https://github.com/AravisProject/aravis/files/9398161/camera.local.control.log)
[camera.remote.control.log](https://github.com/AravisProject/aravis/files/9398163/camera.remote.control.log)

```
arv-tool-0.8 --version
0.8.21
```

aldoshkind, 2022-08-23:
Hello, @EmmanuelP 
With `--network host` I am able to successfully run `gst-launch-1.0 aravissrc camera-name="192.168.5.38" ! fakesink dump=true`. Also, `arv-tool-0.8` prints list of cameras.
Without `--network host` I have `Internal data stream error` with same pipeline. In the logs I see that connection established successfully.

So, behaviour looks similar to what @nznobody describes.

Here is the log:
```
root@e5bf910cfa9c:/code/libs/aravis# ARV_DEBUG=all gst-launch-1.0 aravissrc camera-name="192.168.5.38" ! fakesink dump=true
[11:18:18.353] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[11:18:18.354] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.17.0.2 (172.17.255.255)
[11:18:18.354] 🆆 interface> [ArvGVInterface::arv_gv_interface_camera_locate] Error: Error sending message: Invalid argument
[11:18:18.355] 🅸 device> [GvDevice::new] Interface address = 172.17.0.2
[11:18:18.355] 🅸 device> [GvDevice::new] Device address = 192.168.5.38
[11:18:18.357] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:MER2_GIGE_V2.5.4.zip;41a30000;8810' at 0x200
[11:18:18.358] 🅸 device> [GvDevice::load_genicam] Xml address = 0x41a30000 - size = 0x8810 - MER2_GIGE_V2.5.4.zip
[11:18:18.517] 🅸 device> [GvDevice::load_genicam] Zipped xml data
[11:18:18.551] 🅸 dom> [Genicam::create_element] Unknown tag (SwapEndianess)
[11:18:18.551] 🅸 genicam> [Gc::set_default_node_data] Add 'DeviceManufacturerInfo'
[11:18:18.552] 🅸 genicam> [Gc::set_default_node_data] Add 'DeviceID'
[11:18:18.552] 🅸 genicam> [Gc::set_default_node_data] Add 'GevTimestampTickFrequency'
[11:18:18.552] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPHostPort'
[11:18:18.552] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPSDoNotFragment'
[11:18:18.552] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPSBigEndian'
[11:18:18.552] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCSP'
[11:18:18.552] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPAddrCalc'
[11:18:18.554] 🅸 device> [GvDevice::new] Device endianness = big
[11:18:18.554] 🅸 device> [GvDevice::new] Packet resend     = yes
[11:18:18.554] 🅸 device> [GvDevice::new] Write memory      = yes
[11:18:18.554] 🅸 device> [GvDevice::new] Legacy endianness handling = yes
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
[11:18:18.627] 🅸 device> [GvDevice::create_stream] Number of stream channels = 1
[11:18:18.627] 🅸 device> [GvDevice::auto_packet_size] No GevSCPSFireTestPacket feature found
[11:18:18.629] 🅸 device> [GvStream::stream_new] Packet size = 1500 byte(s)
[11:18:18.631] 🅸 stream> [GvStream::stream_new] Destination stream port = 48005
[11:18:18.631] 🅸 stream> [GvStream::stream_new] Source stream port = 11220
[11:18:18.663] 🅸 stream> [GvStream::loop] Packet socket method
[11:18:18.759] 🅸 stream-thread> [GvStream::set_socket_filter] source ip = 0xc0a80526 - port = 11220 - dest ip = 0xac110002 - port 48005
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis0:
streaming stopped, reason error (-5)
Execution ended after 0:00:02.000534834
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
[11:18:20.872] 🅸 stream>     bins    ;frame_retent; packet_time;inter_packet
                                    0;           0;           0;           0
                         -------------
                         >=      2000;           0;           0;           0
                         <          0;           0;           0;           0
                         min         :         n/a;         n/a;         n/a
                         max         :         n/a;         n/a;         n/a
                         last max at :           0;           0;           0
                         counter     :           0:           0:           0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_completed_buffers    = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_failures             = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_underruns            = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_timeouts             = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_aborteds             = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_missing_frames       = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_size_mismatch_errors = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_received_packets     = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_missing_packets      = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_error_packets        = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_ignored_packets      = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_resend_requests      = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_resent_packets       = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_resend_ratio_reached = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_resend_disabled      = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_duplicated_packets   = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_transferred_bytes    = 0
[11:18:20.872] 🅸 stream> [GvStream::finalize] n_ignored_bytes        = 0
[11:18:20.873] 🅸 stream> [Stream::finalize] Flush 50 buffer[s] in input queue
[11:18:20.873] 🅸 stream> [Stream::finalize] Flush 0 buffer[s] in output queue
Setting pipeline to NULL ...
Freeing pipeline ...

```

aldoshkind, 2022-08-23:
Looks like this port here must be forwarded to allow working from docker container without `--network host`:
https://github.com/AravisProject/aravis/blob/ea09a67805920b7502d639f0eea9770c8cfa1ec9/src/arvgvstream.c#L1435

I ran container with `-p 20223:20223/udp` and replaced line above with `priv->thread_data->interface_socket_address = g_inet_socket_address_new (interface_address, 20223);` and now I can see data printed by fakesink.

EmmanuelP, 2022-08-24:
Please test #709 

In this pull request, the port range is hardcoded to [10000..10010], so `-p 10000-10010:10000-10010` should be enough.

aldoshkind, 2022-08-24:
I'll check your improvement later.
I made similar fix here: https://github.com/AravisProject/aravis/commit/fcaca69eea922de5e7a562fe0078eee1b547375a
Key difference here is that it takes port range from environment variable. It works :)

aldoshkind, 2022-08-24:
I checked #709 . It works.
**BUT!**
When running two cameras at the same time there are multiple errors like this:
```
[09:31:06.748] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 2590 (last: 2592)
[09:31:06.749] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =     2590 
                     packet_id    =     1075
                     data_size    =     1464
[09:31:06.749] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 2590 (last: 2592)
[09:31:06.749] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =     2590 
                     packet_id    =     1084
                     data_size    =     1464
[09:31:06.749] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 2590 (last: 2592)
[09:31:06.749] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =     2590 
                     packet_id    =     1102
                     data_size    =     1464

```

EmmanuelP, 2022-08-24:
It is most probably unrelated, unless destination port is the same for the 2 cameras, which should not happen.

What say the lines `🅸 stream> [GvStream::stream_new] Destination stream port` ?

aldoshkind, 2022-08-24:
These messages start to appear when I start second pipeline.
Destination ports are different:
```
[10:02:56.310] 🅸 device> [GvStream::stream_new] Packet size = 1500 byte(s)
[10:02:56.312] 🅸 stream> [GvStream::stream_new] Destination stream port = 10003
[10:02:56.312] 🅸 stream> [GvStream::stream_new] Source stream port = 11220
[10:02:56.352] 🅸 stream> [GvStream::loop] Packet socket method
[10:02:56.432] 🅸 stream-thread> [GvStream::set_socket_filter] source ip = 0xc0a80527 - port = 11220 - dest ip = 0xac110002 - port 10003
```
and 
```
[10:04:32.927] 🅸 device> [GvStream::stream_new] Packet size = 1500 byte(s)
[10:04:32.929] 🅸 stream> [GvStream::stream_new] Destination stream port = 10005
[10:04:32.929] 🅸 stream> [GvStream::stream_new] Source stream port = 11220
[10:04:32.960] 🅸 stream> [GvStream::loop] Packet socket method
[10:04:33.032] 🅸 stream-thread> [GvStream::set_socket_filter] source ip = 0xc0a80526 - port = 11220 - dest ip = 0xac110002 - port 10005
```

Also I see messages like
```
[10:05:17.307] 🆆 stream-thread> [GvStream::check_frame_completion] Timeout for frame 1221 at dt = 318347
[10:05:17.316] 🆆 stream-thread> [GvStream::check_frame_completion] Timeout for frame 1222 at dt = 288114
[10:05:17.319] 🆆 device> [GvDevice::read_register] Ack reception timeout
[10:05:17.319] 🆆 device> [GvDevice::read_register] Ack reception timeout
```
which disappear when second pipline gets shut down.

aldoshkind, 2022-08-24:
Update: if I run two cameras simulatenously application crashes in a few minutes.
Pipeline:
`ARV_DEBUG=all gst-launch-1.0 aravissrc camera-name="192.168.5.38" ! fakesink aravissrc camera-name="192.168.5.39" ! fakesink`

```
[11:36:50.804] 🅸 stream-thread> [GvStream::process_packet] Error packet at dt = 164855, packet id = 1292 frame id = 2462
[11:36:50.804] 🅸 sp> packet_type  = packet-unavailable (0x800c)
                     content_type = data-block (0x0003)
                     frame_id     =     2462 
                     packet_id    =     1292
                     data_size    =        0
[11:36:50.804] 🅸 stream-thread> [GvStream::process_packet] Error packet at dt = 164855, packet id = 1300 frame id = 2462
[11:36:50.804] 🅸 sp> packet_type  = packet-unavailable (0x800c)
                     content_type = data-block (0x0003)
                     frame_id     =     2462 
                     packet_id    =     1300
                     data_size    =        0
[11:36:50.866] 🆆 stream-thread> [GvStream::check_frame_completion] Timeout for frame 2462 at dt = 272625
[11:36:50.906] 🆆 stream-thread> [GvStream::check_frame_completion] Timeout for frame 2466 at dt = 153021
[11:36:50.920] 🆆 stream-thread> [GvStream::check_frame_completion] Timeout for frame 2462 at dt = 280730
[11:36:50.927] 🆆 stream-thread> [GvStream::check_frame_completion] Timeout for frame 2467 at dt = 132460
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis1: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis1:
streaming stopped, reason error (-5)
Execution ended after 0:01:41.273452363
Setting pipeline to PAUSED ...
Setting pipeline to READY ...

...

[11:36:52.931] 🅸 stream> [GvStream::finalize] n_completed_buffers    = 1777
[11:36:52.931] 🅸 stream> [GvStream::finalize] n_failures             = 689
[11:36:52.931] 🅸 stream> [GvStream::finalize] n_underruns            = 0
[11:36:52.931] 🅸 stream> [GvStream::finalize] n_timeouts             = 689
[11:36:52.931] 🅸 stream> [GvStream::finalize] n_aborteds             = 0
[11:36:52.931] 🅸 stream> [GvStream::finalize] n_missing_frames       = 1
[11:36:52.931] 🅸 stream> [GvStream::finalize] n_size_mismatch_errors = 0
[11:36:52.931] 🅸 stream> [GvStream::finalize] n_received_packets     = 3477608
[11:36:52.931] 🅸 stream> [GvStream::finalize] n_missing_packets      = 546253
[11:36:52.931] 🅸 stream> [GvStream::finalize] n_error_packets        = 4202
[11:36:52.931] 🅸 stream> [GvStream::finalize] n_ignored_packets      = 3
[11:36:52.931] 🅸 stream> [GvStream::finalize] n_resend_requests      = 2452851
[11:36:52.931] 🅸 stream> [GvStream::finalize] n_resent_packets       = 1294976
[11:36:52.931] 🅸 stream> [GvStream::finalize] n_resend_ratio_reached = 7
[11:36:52.932] 🅸 stream> [GvStream::finalize] n_resend_disabled      = 4202
[11:36:52.932] 🅸 stream> [GvStream::finalize] n_duplicated_packets   = 2350
[11:36:52.932] 🅸 stream> [GvStream::finalize] n_transferred_bytes    = 5104569508
[11:36:52.932] 🅸 stream> [GvStream::finalize] n_ignored_bytes        = 24
[11:36:52.932] 🅸 stream> [Stream::finalize] Flush 50 buffer[s] in input queue
[11:36:52.932] 🅸 stream> [Stream::finalize] Flush 0 buffer[s] in output queue

...

[11:36:52.964] 🅸 stream> [GvStream::finalize] n_completed_buffers    = 1804
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_failures             = 714
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_underruns            = 0
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_timeouts             = 714
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_aborteds             = 0
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_missing_frames       = 1
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_size_mismatch_errors = 0
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_received_packets     = 3547379
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_missing_packets      = 622990
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_error_packets        = 3597
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_ignored_packets      = 10
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_resend_requests      = 2418993
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_resent_packets       = 1419183
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_resend_ratio_reached = 11
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_resend_disabled      = 3597
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_duplicated_packets   = 3318
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_transferred_bytes    = 5208011788
[11:36:52.964] 🅸 stream> [GvStream::finalize] n_ignored_bytes        = 80
[11:36:52.965] 🅸 stream> [Stream::finalize] Flush 48 buffer[s] in input queue
[11:36:52.965] 🅸 stream> [Stream::finalize] Flush 2 buffer[s] in output queue

```

aldoshkind, 2022-08-24:
Update: same happens on the host, seems not related to docker. Should I open new issue?

EmmanuelP, 2022-08-24:
Yes, please.

EmmanuelP, 2022-08-24:
Back to this issue, with my branch, are you able to connect to a camera using its name instead of the IP address ?

aldoshkind, 2022-08-24:
With commit 8144377dde5d7a0c391722cc9ad7c0f5e8e09865 I am unable to connect to camera by its name (I tested same pipeline on host, it works). Accessing by IP works well.
Also arv-tool-0.8 says `No device found` if ran inside container (but it works on host).

wuhaochen, 2022-10-04:
Hi there, I'm working on a similar problem and came to this issue. I was wondering if it's possible to just reuse the same port number for different cameras. It seems that `g_socket_receive_from` function can filter by the source port in the camera and that should be fixed once the stream is created.

This way we can have aravis exposed only on a fixed number of ports while support many incoming cameras.

EmmanuelP, 2022-10-04:
@wuhaochen Did you try #709 ?

Aravis should reuse the same port if the source IP is different. At least that is what I have seen during my tests.

wuhaochen, 2022-10-04:
> @wuhaochen Did you try #709 ?
> 
> Aravis should reuse the same port if the source IP is different. At least that is what I have seen during my tests.

I haven't tried #709 yet but what I noticed from main is that aravis opens two UDP ports when I'm connected to one camera and four ports when I'm connected two cameras (with different IPs).

eudoxos, 2022-10-10:
Late to the party here, but very much interested in fixing docker without host network (i.e. under Windows). #232 is still open, could that be related? The thing there was that packets have the outer bridge IP as destination, get correctly routed by OS to the container, but Aravis discards them as it does not recognize the bridge IP as its own (it only knows the "other" side of the bridge, visible in docker). Perhaps this is orthogonal, unrelated, or no longer an issue?

wuhaochen, 2022-10-11:
@eudoxos I started working on #728 a bit and my current understanding (or at least in my use case) is that `_set_socket_filter` does not matter as it is only used with packet socket which I'm not using. However I think the `g_socket_bind` function will bind to the ip inside the container so essentially it is in some sense still filtering out packet that send to the outer bridge IP.

I'll probably see if there is a way to bind all the incoming traffic to the same IP and port that is accessible from cameras, let me know if you have other thoughts though.

eudoxos, 2022-10-11:
I was going through the genIcam Standard Features Naming Convention (it seems to be freely accessible now, did it change?), it mentions `GevPrimaryApplicationIPAddress` which should contain "our" (=primary application) IP address, the one the camera sees (outer bridge IP). Would this be useful? We currently set the stream destination to our inner IP address. 

https://github.com/AravisProject/aravis/blob/e8822c9ec9b25111ae51dc354b399f920f1b6d0e/src/arvgvstream.c#L1460-L1461

zhaotyer, 2022-11-29:
> Please test #709
> 
> In this pull request, the port range is hardcoded to [10000..10010], so `-p 10000-10010:10000-10010` should be enough.

I switch the branch to Wip/port range #709 and Rebuilt the image
then run this docker 
`docker run -it -p 10000-10010:10000-10010  172.16.1.99/aip/base/gst-transwarp:22.09-latest-arav-0.8 bash`
then run pipeline 
`ARV_DEBUG=all gst-launch-1.0 aravissrc camera-name="172.16.251.66" ! fakesink dump=true`

then i get error log:
(gst-launch-1.0:24): GStreamer-CRITICAL **: 10:07:19.576: gst_structure_fixate_field_nearest_fraction: assertion 'gst_structure_has_field (structure, field_name)' failed
New clock: GstSystemClock
[10:07:19.814] 🅸 device> [GvDevice::create_stream] Number of stream channels = 1
[10:07:19.814] 🅸 device> [GvDevice::auto_packet_size] No GevSCPSFireTestPacket feature found
[10:07:19.815] 🅸 device> [GvStream::stream_new] Packet size = 1396 byte(s)
[10:07:19.816] 🅸 stream> [GvStream::stream_new] Destination stream port = 33973
[10:07:19.816] 🅸 stream> [GvStream::stream_new] Source stream port = 3959
[10:07:19.823] 🅸 stream> [GvStream::loop] Packet socket method
[10:07:19.855] 🅸 stream-thread> [GvStream::set_socket_filter] source ip = 0xac10fb42 - port = 3959 - dest ip = 0xac110002 - port 33973
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis0:
streaming stopped, reason error (-5)
Execution ended after 0:00:02.280910718
Setting pipeline to NULL ...
[10:07:21.886] 🅸 stream>     bins    ;frame_retent; packet_time;inter_packet
                                    0;           0;           0;           0
                         -------------
                         >=      2000;           0;           0;           0
                         <          0;           0;           0;           0
                         min         :         n/a;         n/a;         n/a
                         max         :         n/a;         n/a;         n/a
                         last max at :           0;           0;           0
                         counter     :           0:           0:           0
[10:07:21.886] 🅸 stream> [GvStream::finalize] n_completed_buffers    = 0
[10:07:21.886] 🅸 stream> [GvStream::finalize] n_failures             = 0
[10:07:21.886] 🅸 stream> [GvStream::finalize] n_underruns            = 0
[10:07:21.886] 🅸 stream> [GvStream::finalize] n_timeouts             = 0
[10:07:21.887] 🅸 stream> [GvStream::finalize] n_aborted              = 0
[10:07:21.887] 🅸 stream> [GvStream::finalize] n_missing_frames       = 0
[10:07:21.887] 🅸 stream> [GvStream::finalize] n_size_mismatch_errors = 0
[10:07:21.887] 🅸 stream> [GvStream::finalize] n_received_packets     = 0
[10:07:21.887] 🅸 stream> [GvStream::finalize] n_missing_packets      = 0
[10:07:21.887] 🅸 stream> [GvStream::finalize] n_error_packets        = 0
[10:07:21.887] 🅸 stream> [GvStream::finalize] n_ignored_packets      = 0
[10:07:21.887] 🅸 stream> [GvStream::finalize] n_resend_requests      = 0
[10:07:21.887] 🅸 stream> [GvStream::finalize] n_resent_packets       = 0
[10:07:21.887] 🅸 stream> [GvStream::finalize] n_resend_ratio_reached = 0
[10:07:21.887] 🅸 stream> [GvStream::finalize] n_resend_disabled      = 0
[10:07:21.887] 🅸 stream> [GvStream::finalize] n_duplicated_packets   = 0
[10:07:21.887] 🅸 stream> [GvStream::finalize] n_transferred_bytes    = 0
[10:07:21.887] 🅸 stream> [GvStream::finalize] n_ignored_bytes        = 0
[10:07:21.887] 🅸 stream> [Stream::finalize] Flush 50 buffer[s] in input queue
[10:07:21.887] 🅸 stream> [Stream::finalize] Flush 0 buffer[s] in output queue

What should I do to avoid this problem？
Looking forward to your reply

zhaotyer, 2022-11-29:
> 

How did you run docker images and start a pipleline?, I switch the branch to Wip/port 709 and Rebuilt the image,
But it does't work

zhaotyer, 2022-11-29:
> I checked #709 . It works. **BUT!** When running two cameras at the same time there are multiple errors like this:
>
> ```
> [09:31:06.748] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame [...]

How did you run docker images and start a pipleline?, I switch the branch to Wip/port 709 and Rebuilt the image,
But it does't work

aldoshkind, 2022-11-29:
> How did you run docker images and start a pipleline?, I switch the branch to Wip/port 709 and Rebuilt the image, But it does't work

Hi! Your docker command looks correct. In yout case `Destination stream port = 33973`, which possibly indicates you use improper commit (please compare to the logs above).

zhaotyer, 2022-11-30:
> Hi! Your docker command looks correct. In yout case `Destination stream port = 33973`, which possibly indicates you use improper commit (please compare to the logs above).

Thank you for your reply，this is code commit log
[image]
In this branch,  arv_port_minimum and arv_port_maximum initial value is 0,
I can't find a place to modify their values ​​through docker port binding in the aravissrc plugin，
Please give some more detailed guidance

EmmanuelP, 2022-12-01:
@zhaotyer the port-range branch has evolved since last august. The port range is not hardcoded anymore. There is no corresponding gst-plugin option yet. But you can test the port range restriction using `arv-camera-test-0.8` with the `--gv-port-range` option.

aldoshkind, 2024-11-19:
Hi @EmmanuelP ! It's been more than two years since my first message and almost two since your last one. What's the state of port range control in gst plugin at the moment?
I am working on service which should not use network_mode=host. I use Aravis 0.8.30 from Ubuntu 24.04 repo. Should I build from source manually patched version to achieve correct operation in docker container?

EmmanuelP, 2024-11-21:
Hi,

The gst option is not implemented. Contribution (code and test) welcome !