AravisProject/aravis #711: Aravis with usb enabled, when usb ports are not available a SEGV occurs  during discovery of the devices.

ebfarias, 2022-08-25:
**Describe the bug**
Aravis with usb enabled, when usb ports are not available a SEGV occurs  during discovery of the devices. 

**To Reproduce**
Inside a docker container, with privileged off,  aravis with usb support turn on,  call  arv_update_device_list().

**Expected behavior**
Don't expect the system to crash if no usb ports are available.  Expect a warning to be printed. 

The libusb_init fails but nothing in the Aravis code ever checks for this condition. 
I added a check of the return code of libusb_init in arv_uv_interface_init and display a warning.
Since I couldn’t find a way to return an error from this function:
In _discover I added a check
 if (uv_interface->priv->usb == NULL) and if yes print warning and return.   This resolves the SEGV fault.
Not sure it’s the best way but having limited knowledge of the Aravis code this seemed the best way with limited changes. 
I've attached the file I changed.

**Camera description:**
 - Interface [USB3]

**Platform description:**
 - Aravis version 0.8.23
 - OS: Ubuntu 18.04.6 LTS, 5.4.0-120-generic
 - x86_64]

**Additional context**
Here is an example of the error we get:

ERROR: AddressSanitizer: SEGV on unknown address 0x000000000030 (pc 0x7fd0eaba2b80 bp 0x61d000004810 sp 0x7ffd064d5668 T0)
==5373==The signal is caused by a READ memory access.
==5373==Hint: address points to the zero page.
    #0 0x7fd0eaba2b7f  (/lib64/libpthread.so.0+0x9b7f)
    #1 0x7fd0eade142f in libusb_get_device_list /local/p4clients/pkgbuild-5n2yI/workspace/src/LibUsb/build/private/src/libusb/core.c:809
    #2 0x7fd0eae8962a in _discover ../../../../../../../../src/Aravis/third-party-src/src/arvuvinterface.c:283
    #3 0x7fd0eae761c0 in arv_interface_update_device_list ../../../../../../../../src/Aravis/third-party-src/src/arvinterface.c:90
    #4 0x7fd0eae52c9e in arv_update_device_list ../../../../../../../../src/Aravis/third-party-src/src/arvsystem.c:172

This occurs on our build server during unit test build.  
I was able to reproduce in a docker image when docker privileged mode is off. 

Also turning on libusb debug I found the following:
libusb: debug [libusb_init] libusb v1.0.20.11004
libusb: error [op_init] could not find usbfs
libusb: debug [libusb_get_device_list] 
Exit signal: 11 (Segmentation fault)
/usr/lib64/libpthread.so.0: +0x118e0 [0x7fdfaa8018e0]
/usr/lib64/libpthread.so.0: pthread_mutex_lock+0 [0x7fdfaa7f9c80]
libusb-1.0.so.0: libusb_get_device_list+0x160 [0x7fdfa09cf430]
[arvuvinterface.zip](https://github.com/AravisProject/aravis/files/9427848/arvuvinterface.zip)

EmmanuelP, 2022-09-02:
HI @ebfarias 

Please try #713

closed 2022-09-07 by EmmanuelP

ebfarias, 2022-09-07:
Sorry for the delayed response but I was off for a few days.  I will give this a try as soon as possible.

ebfarias, 2022-09-13:
I have tested the latest changes that were merged to mainline for this problem (issue #713) it works great.

EmmanuelP, 2022-09-13:
Thanks for the test !

Cheers.

PR #713 (2022-09-07): usb: better libusb error handling
Fix #711
commit fd942d57 (2022-09-07): usb: better libusb error handling

Fix #711