AravisProject/aravis #715: Unable to create camera stream. Incorrect size of GevSCPSPacketSize.

radimkolos, 2022-09-12:
**The Issue**
Hello, I have been using Aravis with several cameras (Flir, Basler, etc.) for a long time. Now I got my hands on a new camera (more in the camera description). The camera is visible, but it fails to create a stream. I am using eBUSPlayer to display the camera data, and to set it up. I have no problem with this software, however, I am forced to use Aravis because of the possibility of using multiple cameras from different manufacturers in one project. It seems like for some reason GevSCPSPacketSize gives value of zero and it then tries to call arv_gc_integer_get_value with this value and fails.

**To Reproduce**
The output of arv-camera-test-0.8 is as follows: 
> Looking for the first available camera
vendor name            = Pleora Technologies Inc.
model name             = iPORT-NTx-GigE-PT01-PB3IP01-128
device serial number   = 062B1909
image width            = 640
image height           = 480
payload                = 614400 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 0 ns
** (arv-camera-test-0.8:18992): CRITICAL **: 09:21:59.629: arv_gc_integer_get_value: assertion 'error == NULL || *error == NULL' failed
gv packet size         = 0 bytes
Can't create stream thread: [GevSCPSPacketSize] [GevSCPSPacketSizeReg] GigEVision write_memory error (invalid-parameter)

As you can see, the feature GevSCPSPacketSize is available and is equal to zero for some reason. 

The output of arv-tool-0.8 is as follows: 
> Pleora Technologies Inc.-iPORT-NTx-GigE-PT01-PB3IP01-128-062B1909 (192.168.110.6)

The output of arv-test-0.8 is as follows:
> Found 1 device
Testing 'Pleora Technologies Inc.:iPORT-NTx-GigE-PT01-PB3IP01-128'
Genicam:Load                        SUCCESS
Genicam:Schema                      IGNORED
Properties:SensorSizeReadout        FAILURE [SensorWidth] Not found
Properties:SensorSizeCheck          IGNORED
Properties:GainReadout              FAILURE [GainRaw] Not found
Properties:ExposureTimeReadout      FAILURE [ExposureTimeAbs] Not found
MultipleAcquisitionA:BufferCheck    FAILURE [AcquisitionFrameRateAbs] Not found
SoftwareTrigger:NoSupport           FAILURE
MultipleAcquisitionB:BufferCheck    FAILURE [AcquisitionFrameRateAbs] Not found
SingleAcquisition:BufferCheck       FAILURE [GevSCPSPacketSize] [GevSCPSPacketSizeReg] GigEVision write_memory error (invalid-parameter)
Chunks:BufferCheck                  FAILURE [ChunkSelector] 'OffsetX' not an entry
GigEVision:NStreamChannels          SUCCESS
GigEVision:StreamChannel            SUCCESS

**Expected behavior**
I expect that a stream will be created like for the other cameras I have available.

**Camera description**
 - Manufacturer: Pleora Technologies Inc.
 - Model: P-GIGE-001
 - Interface: GigE

**Platform description:**
 - Aravis version: 0.8.22 
 - OS: Arch Linux x86_64 (5.19.6-arch1-1)
 - Hardware: Intel i7-6700 (8) @ 4.000GHz,  NVIDIA GeForce GTX 1650, 15924MiB

EmmanuelP, 2022-09-12:
It is probably related to legacy endianness mechanism. Try to add your camera here:

https://github.com/AravisProject/aravis/blob/bc812389f065b9f1aa010a7132466f5fa6743889/src/arvgcport.c#L44-L46

vendor and model are those specified in the Genicam XML data, obtained using `arv-tool-0.8 genicam`.

radimkolos, 2022-09-13:
Hi Emmanuel, this did work for me, thank you so much. I manually added the camera to the array, compiled Aravis again, used it, and now no issue is present to me.

Just one additional question. Is there any way to add a camera to the array via code? If not, I should be able to patch the file right before every build, since we have an automated system to build out packages. However, this is not the preferred way.

Thank you once again for your really quick response.

EmmanuelP, 2022-09-13:
> Just one additional question. Is there any way to add a camera to the array via code? If not, I should be able to patch the file right before every build, since we have an automated system to build out packages. However, this is not the preferred way.

The list is hardcoded on purpose, this way we are aware of the devices that should be added.

Please open a merge request with your patch.

radimkolos, 2022-09-13:
Hi Emmanuel, understood. I have just created a merge request with my patch. Thank you once again for your help.

commit 191102ae (2022-09-13): gvdevice: added port legacy info for Pleora P-GIGE-001

fix #715
closed 2022-09-13 by EmmanuelP