AravisProject/aravis #716: Can detect the camera (FLIR-Blackfly S BFS-PGE-63S4M-20010137), but failed connect to it. timeout encountered.

ywWork, 2022-09-13:
**Describe the bug**
The camera can be detected using `arv-tool-0.8', but I cannot connect to it.
None of the following commands works due to disconnection. 
e.g.: `arv-viewer-0.8 -d all` or `arv-test-0.8`.

**To Reproduce**
Steps to reproduce the behavior:
1. `arv-viewer-0.8 -d all`
2. The GUI pops out and the camera is detected. And the console shows:
```
[19:14:04.835] ? interface> Found 0 USB3Vision device (among 2 USB devices)
[19:14:04.835] ? interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[19:14:04.835] ? interface> [GvDiscoverSocket::new] Add interface 10.35.241.200 (10.35.255.255)
[19:14:04.835] ? interface> [GvDiscoverSocket::new] Add interface 192.146.4.254 (192.146.4.255)
[19:14:04.835] ? interface> [GvDiscoverSocket::new] Add interface 192.168.172.0 (192.168.172.15)
[19:14:04.835] ? interface> [GvDiscoverSocket::new] Add interface 192.168.172.1 (192.168.172.255)
[19:14:04.838] ? interface> [GvInterface::discovery] Device 'FLIR-Blackfly S BFS-PGE-63S4M-20010137' found (interface 192.168.172.1) user_id '' - MAC 'blahblah'
```
3. Once I click the camera in the GUI for further steps, following printed out:
```
[19:17:03.667] ? interface> Found 0 USB3Vision device (among 2 USB devices)
[19:17:03.667] ? device> [GvDevice::new] Interface address = 192.168.172.1
[19:17:03.667] ? device> [GvDevice::new] Device address = blahblah
[19:17:04.167] ? device> [GvDevice::read_memory] Ack reception timeout
[19:17:04.167] ? device> [GvDevice::read_memory] Ack reception timeout
[19:17:04.668] ? device> [GvDevice::read_memory] Ack reception timeout
[19:17:04.668] ? device> [GvDevice::read_memory] Ack reception timeout
[19:17:05.169] ? device> [GvDevice::read_memory] Ack reception timeout
[19:17:05.169] ? device> [GvDevice::read_memory] Ack reception timeout
[19:17:05.669] ? device> [GvDevice::read_memory] Ack reception timeout
[19:17:05.669] ? device> [GvDevice::read_memory] Ack reception timeout
[19:17:06.170] ? device> [GvDevice::read_memory] Ack reception timeout
[19:17:06.170] ? device> [GvDevice::read_memory] Ack reception timeout
```

**Expected behavior**
I should be able to interact with the camera. 

**Camera description:**
 - Manufacturer: FLIR
 - Model: BFS-PGE-63S4M-20010137
 - Interface: GigE

**Platform description:**
 - Aravis version: 0.8.22
 - OS: Almalinux 8
 - Hardware: x86_64

**Additional context**
Add any other context about the problem here.

EmmanuelP, 2022-09-13:
Hi,

How the camera is connected to your machine ?
Are you able to connect to the fake camera ? (run arv-fake-camera-0.8 in another console)
Are you able to connect to another camera ?

Could you attach the output of `arv-tool-0.8 -d all:3 features` ?

ywWork, 2022-09-13:
Hello:
To answer your questions:
1. The camera is connected to a server. 
2. I cannot, it is hanging.
3. No. I just have one camera.
4. For the printout of `arv-tool-0.8 -d all:3 features`
```
[09:55:44.151] 🅸 interface> Found 0 USB3Vision device (among 2 USB devices)
[09:55:44.152] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[09:55:44.152] 🅸 interface> [GvDiscoverSocket::new] Add interface blahblah (10.35.255.255)
[09:55:44.152] 🅸 interface> [GvDiscoverSocket::new] Add interface blahblah (192.146.4.255)
[09:55:44.152] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.172.0 (192.168.172.15)
[09:55:44.152] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.172.1 (192.168.172.255)
[09:55:44.155] 🅳 cp> packet_type  = ack
                     command      = discovery-ack
                     size         = 248
                     id           = 65535
                     manufacturer = FLIR
                     name         =
                     model        = Blackfly S BFS-PGE-63S4M
                     address      = blahblah
                     00000000 00 00 00 03 00 f8 ff ff 00 01 00 02 80 00 00 01  ................
                     00000010 00 00 2c dd a3 31 54 99 00 00 00 07 00 00 00 07  ..,..1T.........
                     00000020 00 00 00 00 00 00 00 00 00 00 00 00 a9 fe 00 01  ................
                     00000030 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00  ................
                     00000040 00 00 00 00 00 00 00 00 00 00 00 00 a9 fe 00 02  ................
                     00000050 46 4c 49 52 00 00 00 00 00 00 00 00 00 00 00 00  FLIR............
                     00000060 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
                     00000070 42 6c 61 63 6b 66 6c 79 20 53 20 42 46 53 2d 50  Blackfly S BFS-P
                     00000080 47 45 2d 36 33 53 34 4d 00 00 00 00 00 00 00 00  GE-63S4M........
                     00000090 31 38 31 30 2e 34 2e 34 37 2e 30 00 00 00 00 00  1810.4.47.0.....
                     000000a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
                     000000b0 44 65 63 20 20 36 20 32 30 31 38 20 30 38 3a 33  Dec  6 2018 08:3
                     000000c0 39 3a 34 36 00 00 00 00 00 00 00 00 00 00 00 00  9:46............
                     000000d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
                     000000e0 32 30 30 31 30 31 33 37 00 00 00 00 00 00 00 00  20010137........
                     000000f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[09:55:44.155] 🅸 interface> [GvInterface::discovery] Device 'FLIR-Blackfly S BFS-PGE-63S4M-20010137' found (interface 192.168.172.1) user_id '' - MAC 'blahblah'
[09:55:45.156] 🅳 misc> Regex '^.*$' created from glob '*'
FLIR-Blackfly S BFS-PGE-63S4M-20010137 (blahblah)
[09:55:45.157] 🅸 interface> Found 0 USB3Vision device (among 2 USB devices)
[09:55:45.157] 🅸 device> [GvDevice::new] Interface address = 192.168.172.1
[09:55:45.157] 🅸 device> [GvDevice::new] Device address = blahblah
[09:55:45.658] 🆆 device> [GvDevice::read_memory] Ack reception timeout
[09:55:45.658] 🆆 device> [GvDevice::read_memory] Ack reception timeout
[09:55:46.158] 🆆 device> [GvDevice::read_memory] Ack reception timeout
[09:55:46.158] 🆆 device> [GvDevice::read_memory] Ack reception timeout
[09:55:46.659] 🆆 device> [GvDevice::read_memory] Ack reception timeout
[09:55:46.659] 🆆 device> [GvDevice::read_memory] Ack reception timeout
[09:55:47.160] 🆆 device> [GvDevice::read_memory] Ack reception timeout
[09:55:47.160] 🆆 device> [GvDevice::read_memory] Ack reception timeout
[09:55:47.661] 🆆 device> [GvDevice::read_memory] Ack reception timeout
[09:55:47.661] 🆆 device> [GvDevice::read_memory] Ack reception timeout
Failed to open device 'FLIR-Blackfly S BFS-PGE-63S4M-20010137': GigEVision read_memory timeout
```

ywWork, 2022-09-13:
Addtional information: The detected ip address for the camera is not under the server subnet 192.163.172.1.
Could that be a problem?

EmmanuelP, 2022-09-13:
> Addtional information: The detected ip address for the camera is not under the server subnet 192.163.172.1/24. Could that be a problem?

Yes. Either you set a fixed IP address that fits in the subnet, or you configure the camera to use a dhcp configuration (but you also need a dhcp server that serves IP addresses for this subnet).

You may try to use the network command of arv-tool-0.8, but you will have to configure an interface with a subnet that includes the current IP address of your camera.

ywWork, 2022-09-13:
Thank you for the reply.
After I reconfigure the IP of the server, the camera is working.
However, I need the camera to follow the server's IP not the other way around.
Can you tell me how to configure the IP for the camera using Aravis?
The IP config tools are most like designed for windows, macOS or ubuntu. I am using Almalinux...
Any way of configure the IP using command line prompt?

EmmanuelP, 2022-09-14:
> Any way of configure the IP using command line prompt?

Try to use `arv-tool-0.8 network`. For example:

```
arv-tool-0.8 network mode=PersistentIP ip=192.168.0.1 mask=255.255.255.0 gateway=192.168.0.254
```

This command relies on some standard features which may not be available on your device.

EmmanuelP, 2022-09-16:
I'm closing this issue, as it is a network configuration issue. Don't hesitate to open another issue if you have troubles using `arv-tool network` command.

Cheers.

closed 2023-05-19 by EmmanuelP