AravisProject/aravis #718: Auto negotiation of packet size doesn't seem to working all the time

ebfarias, 2022-09-14:
**Describe the bug**
Aravis seems to use a wrong size for IP-layer headers, this lets any reception of test-packets fail.
The auto negotiation never test for packet sizes above the starting value.

**Expected behavior**
Expect auto negotiation to always pick the greatest packet size available. Unless of course it's turned off by the user. 

**Camera description:**
 - Manufacturer  SICK
 - Model Visionary-S CX V3S102-1x
 - Interface:  GigE

**Platform description:**
 - Aravis version 0.8.23
 - OS: Ubuntu 18.04
 - Hardware x86_64

**Additional context**
This also happens on other cameras like the Omron SENTECH-STC-MCS500POE-21HM319 and Basler-acA2040-25gc-22094718.
I've attached the changes made to allow this to work.
[aravischanges.txt](https://github.com/AravisProject/aravis/files/9566049/aravischanges.txt)

EmmanuelP, 2022-09-16:
Hi,

Thanks for your report. Could you open a merge request with your changes ? It makes the review easier, and you will properly credited once the change will be merged.

ebfarias, 2022-09-16:
It's not possible for me to open a merge request because of work constraints.

EmmanuelP, 2022-09-21:
Please try #723

commit 7ce1f55a (2022-09-21): gvdevice: improve auto packet size

The computation of the expected test packet was wrong on some platform, so don't
use sizeof.

Also, GevSCPSPacketSize increment was not correctly take into account.

And finally, don't limit the precision to 16 but to increment instead. I fear
forcing the precision may conflict with cameras with increment not a divisor of
16 (which would be surprising, but who knows (yes, the GigEVision documentation,
but we can not use it :( )).

Fix #718
commit 80f8801d (2022-09-30): gvdevice: improve auto packet size

The computation of the expected test packet was wrong on some platform, so don't
use sizeof.

Also, GevSCPSPacketSize increment was not correctly take into account.

And finally, don't limit the precision to 16 but to increment instead. I fear
forcing the precision may conflict with cameras with increment not a divisor of
16 (which would be surprising, but who knows (yes, the GigEVision documentation,
but we can not use it :( )).

Fix #718
closed 2022-09-30 by EmmanuelP