AravisProject/aravis #720: USB Image Capture / Buffer Management

Dennis10231, 2022-09-20:
**Describe the bug**
When setting up image capture like in the example: 02-multiple-acquisition.c
the user allocates buffers with size "payload" as reported by the camera prior to calling start_acquision().
Let's assume that we grab a few images and we decide to adjust the image size.
So we call stop_acquisition() but we cannot pull the images from the stream_grabber and register new (bigger?)
buffers. In GenTL, DSFlushQueue() would flush the input queue to the output queue which would allow 
the user to do this. The only function that flushes the queue seems to be arv_stream_finalize(),
which implies that I would need to shut down the stream_grabber entirely and restart from the beginning.

**To Reproduce**
Use example 02-multiple-acquisition.c and modify accordingly.

**Expected behavior**
There are several solutions:
(1) Aravis provides a new function, arv_stream_flush_queue(), similar to the GenTL function.
Streaming needs to be stopped before calling, all buffers from the input queue are flushed to the
output queue and marked accordingly.
(2) stop_acquisition() does this automagically.

**Camera description:**
Basler a2A1920-160ucBAS

**Platform description:**
 - Aravis version: 0.8.22
 - OS: Ubuntu 22.04
 - Hardware: x86_64

**Additional Discussion**
Another solution would have been to create buffers with the maximum size,
e.g. 02-multiple-acquisition.c, but in line 69 instead of the payload one would use the sensor size.
(or just use payload*2 for a first try) - This creates a new problem:
When you trigger the first image, the stream_callback is called with ARV_STREAM_CALLBACK_TYPE_START_BUFFER 
but not with ARV_STREAM_CALLBACK_TYPE_BUFFER_DONE although the trailer was received.
Only when you trigger the second image, the stream_callback is called with ARV_STREAM_CALLBACK_TYPE_BUFFER_DONE 
for the first image and with ARV_STREAM_CALLBACK_TYPE_START_BUFFER for the second image.
Basically it appears as if Aravis is lagging one image behind.

It seems that Aravis is looking at the buffer size to determine when to deliver an image.
But in theory Aravis, upon receiving the trailer, could determine if all packets are complete and the image can be handed to the user.

EmmanuelP, 2022-09-20:
Hi,

Thanks for the report.

It is already possible to change the payload size without destroying the stream object. Have a look at tests/arvroitest.c in the aravis repository. This sample should be ported to the aravis-c-examples repository. Please let us now if the solution suits your need.

Regarding the detection of the frame completion, that is something we are already aware of, though there is no corresponding issue report.

EmmanuelP, 2022-09-20:
> It is already possible to change the payload size without destroying the stream object. Have a look at tests/arvroitest.c in the aravis repository. This sample should be ported to the aravis-c-examples repository. Please let us now if the solution suits your need.

https://github.com/AravisProject/aravis-c-examples/issues/1

> Regarding the detection of the frame completion, that is something we are already aware of, though there is no corresponding issue report.

#721
#722

closed 2022-09-20 by EmmanuelP