AravisProject/aravis #724: ara-viewer-0.8 not finding GStreamer plugins

eddie3ruff, 2022-09-29:
**Describe the bug**
When trying to run ara-viewer-0.8 from the terminal, I receive the following output.

Gstreamer plugin 'appsrc' is missing.
Gstreamer plugin 'videoconvert' is missing.
Gstreamer plugin 'videoflip' is missing.
Missing GStreamer video output plugin (autovideosink, gtksink or gtkglsink)
Check your gstreamer installation.

**To Reproduce**
Steps to reproduce the behavior:
0. Fresh install of homebrew (/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)")
1. brew install aravis
2. cd to /opt/homebrew/bin
3. sudo ./ara-viewer-0.8
4. Terminal output is the following
Gstreamer plugin 'appsrc' is missing.
Gstreamer plugin 'videoconvert' is missing.
Gstreamer plugin 'videoflip' is missing.
Missing GStreamer video output plugin (autovideosink, gtksink or gtkglsink)
Check your gstreamer installation.

**Expected behavior**
Aravis Viewer opens

**Camera description:**
no camera connected yet

**Platform description:**
I've tried this on 3 different macs.
0. 2022 M1 iMac with Ventura beta
1. 2015 MBP with Catalina (intel)
2. M1 MacBook Air

All three have the exact same terminal message via GStreamer.

**Additional context**
None.

EmmanuelP, 2022-09-29:
Thanks for your report.

You are missing gst-plugins-base package, at least. Probably also gst-plugins-good and gst-plugins-bad. This issue should be reported to homebrew, `brew install aravis` should have installed these gstreamer plugins.

closed 2022-09-29 by EmmanuelP

eddie3ruff, 2022-09-29:
it did install those. When I run brew list, all three of them are listed (base, bad, and good).

EmmanuelP, 2022-09-29:
So it is not a dependency issue. Please report it to homebrew-core: https://github.com/Homebrew/homebrew-core

Cheers.

EmmanuelP, 2022-09-29:
`sudo ./ara-viewer-0.8`

Why sudo ? Did you try without ?

eddie3ruff, 2022-09-29:
without sudo results in the same error. When I built from source (without using brew install aravis) I ran into the same issue. There was a similar ticket that had been closed too, issue 660.

EmmanuelP, 2022-09-29:
Same answer than for #660, it is not an Aravis issue. For some reason the required plugins are not found at runtime.
Are you able to find them using `gst-inspect-1.0` ? Like:

```
gst-inspect-1.0 videoconvert
 ```

eddie3ruff, 2022-09-29:
Yes, I get outputs for all three plugins that the viewer claim to be missing. Looks like 'appsrc' 'videoflip' and 'videoconvert' are located in /usr/local/lib/gstreamer-1.0/

The other Aravis binaries run just fine on all machines previously listed. My FLIR Grasshopper camera is detected correctly. However, I would like to check out the viewer application. Perhaps the viewer is not looking in the correct path for the missing plugins?

Thanks for all of the help thus far!

EmmanuelP, 2022-09-29:
Try `GST_PLUGIN_PATH=/usr/local/lib/gstreamer-1.0/ arv-viewer-0.8`

Things seem seriously broken in gstreamer brew packages. It used to work.

eddie3ruff, 2022-09-29:
Success! thank you so much, working great.