AravisProject/aravis #726: Imaging Source GigE cameras Current IP/Mode/Mask/Gateway settings are not read

xyk2, 2022-10-01:
**Describe the bug**
On Imaging Source GigE cameras, trying to read the current network settings (Mode, Current IP, Current Mask, and Current Gateway) always returns Mode: LLA and 0.0.0.0 for IP, mask, and gateway. Using `tcam-gigetool` (Imaging Source provided software) reads back the correct IP configurations. This happens both for an older model camera (DMK 39GX265-Z20) and also a new DFK 33GX183.

**To Reproduce**
Run arv-tool-0.8 with an Imaging Source GigE camera connected

**Expected behavior**
arv-tool-0.8 should be reading back the correct Mode, Current IP, Current Mask, and Current Gateway from the camera.

**Camera description:**
 - The Imaging Source / DMK 39GX265-Z20 / GigEVision
 - The Imaging Source / DFK 33GX183 / GigEVision

**Additional context**
With `arv-tool-0.8`:
```
~$ arv-tool-0.8 -n "The Imaging Source Europe GmbH-DMK 39GX265-Z20-15910175" network
Mode: LLA
Current IP: 0.0.0.0
Current Mask: 0.0.0.0
Current Gateway: 0.0.0.0
Persistent IP: 172.20.3.3
Persistent Mask: 255.255.255.0
Persistent Gateway: 172.20.3.1
```

With `tcam-gigetool list`:
```
~$ tcam-gigetool list --format msuingINGfdSMr
Model Name           | Serial Number    | User Defined Name    | Current IP        | Current Netmask   | Current Gateway   | Persistent IP     | Persistent Netmask   | Persistent Gateway   | Interface    | DHCP       | Static IP  | MAC Address        | Reachable
DMK 39GX265-Z20        15910175                                  172.20.3.3          255.255.255.0       172.20.3.1          172.20.3.3          255.255.255.0          172.20.3.1             eth3           No           Yes          00:07:48:41:9d:8f    Yes
```

Current IP/Netmask/Gateway is returned correctly with `tcam-gigetool` but not with `arv-tool-0.8`

--- 

From a cursory glance at the tcam-gigetool source code, they have [hard-coded the register addresses](https://github.com/TheImagingSource/tiscamera/blob/2225d34d2f6be6134965b504d56fc7388c1d67f7/src/tcam-network/gigevision.h#L104) for CurrentIP configs, IP, subnet, and gateway. These registers are not present in the device genicam XML.

EmmanuelP, 2022-10-06:
> Should GevCurrentIPAddress 0x24, GevCurrentSubnetMask 0x34, and GevCurrentDefaultGateway 0x44 also be part of the default node data?

Possibly. But I'm not sure these registers are mandatory. The wireshark gvcp dissector has probably this information.

EmmanuelP, 2022-10-07:
From wireshark gvcp dissector:

```
	case GVCP_SUPPORTED_IP_CONFIGURATION_0:
	case GVCP_SUPPORTED_IP_CONFIGURATION_1:
	case GVCP_SUPPORTED_IP_CONFIGURATION_2:
	case GVCP_SUPPORTED_IP_CONFIGURATION_3:
		proto_tree_add_item(branch, hf_gvcp_ip_config_can_handle_pause_frames_v2_0, tvb, offset, 4, ENC_BIG_ENDIAN);
		proto_tree_add_item(branch, hf_gvcp_ip_config_can_generate_pause_frames_v2_0, tvb, offset, 4, ENC_BIG_ENDIAN);
		proto_tree_add_item(branch, hf_gvcp_ip_config_lla, tvb, offset, 4, ENC_BIG_ENDIAN);
		proto_tree_add_item(branch, hf_gvcp_ip_config_dhcp, tvb, offset, 4, ENC_BIG_ENDIAN);
		proto_tree_add_item(branch, hf_gvcp_ip_config_persistent_ip, tvb, offset, 4, ENC_BIG_ENDIAN);
		break;
```

```
		{ &hf_gvcp_ip_config_can_handle_pause_frames_v2_0,
		{ "IP Config Can Handle Pause Frames", "gvcp.bootstrap.ipconfig.canhandlepauseframes",
		FT_BOOLEAN, 32, NULL, 0x80000000,
		NULL, HFILL }},

		{ &hf_gvcp_ip_config_can_generate_pause_frames_v2_0,
		{ "Can Generate Pause Frames", "gvcp.bootstrap.ipconfig.cangeneratepauseframes",
		FT_BOOLEAN, 32, NULL, 0x40000000,
		NULL, HFILL }},

		{ &hf_gvcp_ip_config_lla,
		{ "LLA", "gvcp.bootstrap.ipconfig.lla",
		FT_BOOLEAN, 32, NULL, 0x00000004,
		NULL, HFILL }},

		{ &hf_gvcp_ip_config_dhcp,
		{ "DHCP", "gvcp.bootstrap.ipconfig.dhcp",
		FT_BOOLEAN, 32, NULL, 0x00000002,
		NULL, HFILL }},

		{ &hf_gvcp_ip_config_persistent_ip,
		{ "Persistent IP", "gvcp.bootstrap.ipconfig.persistentip",
		FT_BOOLEAN, 32, NULL, 0x00000001,
		NULL, HFILL }},
```

EmmanuelP, 2022-10-07:
Also:

```

	case GVCP_CURIPCFG_0:
	case GVCP_CURIPCFG_1:
	case GVCP_CURIPCFG_2:
	case GVCP_CURIPCFG_3:
		proto_tree_add_item(branch, hf_gvcp_ip_config_can_handle_pause_frames_v2_0, tvb, offset, 4, ENC_BIG_ENDIAN);
		proto_tree_add_item(branch, hf_gvcp_ip_config_can_generate_pause_frames_v2_0, tvb, offset, 4, ENC_BIG_ENDIAN);
		proto_tree_add_item(branch, hf_gvcp_ip_config_lla, tvb, offset, 4, ENC_BIG_ENDIAN);
		proto_tree_add_item(branch, hf_gvcp_ip_config_dhcp, tvb, offset, 4, ENC_BIG_ENDIAN);
		proto_tree_add_item(branch, hf_gvcp_ip_config_persistent_ip, tvb, offset, 4, ENC_BIG_ENDIAN);
		break;

	case GVCP_CURRENT_IP_ADDRESS_0:
	case GVCP_CURRENT_IP_ADDRESS_1:
	case GVCP_CURRENT_IP_ADDRESS_2:
	case GVCP_CURRENT_IP_ADDRESS_3:
		proto_tree_add_item(branch, hf_gvcp_current_IP, tvb, offset, 4, ENC_BIG_ENDIAN);
		break;

	case GVCP_CURRENT_SUBNET_MASK_0:
	case GVCP_CURRENT_SUBNET_MASK_1:
	case GVCP_CURRENT_SUBNET_MASK_2:
	case GVCP_CURRENT_SUBNET_MASK_3:
		proto_tree_add_item(branch, hf_gvcp_current_subnet_mask, tvb, offset, 4, ENC_BIG_ENDIAN);
		break;

	case GVCP_CURRENT_DEFAULT_GATEWAY_0:
	case GVCP_CURRENT_DEFAULT_GATEWAY_1:
	case GVCP_CURRENT_DEFAULT_GATEWAY_2:
	case GVCP_CURRENT_DEFAULT_GATEWAY_3:
		proto_tree_add_item(branch, hf_gvcp_current_default_gateway, tvb, offset, 4, ENC_BIG_ENDIAN);
		break;
```

EmmanuelP, 2022-10-07:
So yes, the current ip configuration features should probably always be available.

xyk2, 2022-10-07:
For what it's worth, I added the three registers as defaults with`arv_gc_set_default_node_data` in `src/arvgvdevice.c`, recompiled, and it is reading the current IP, subnet, and mask fine now. However the mode is still stuck on LLA (`Mode: LLA`). This seems like it is because the bit index for LLA, DHCP, and Static is different for Imaging Source cameras: 

https://github.com/TheImagingSource/tiscamera/blob/dd20eed75267a6bf5e80b93fb5da572e29bc35a2/src/tcam-network/Camera.cpp#L283-L295 

```
        const int static_ip_bit = 0x00;
        const int dhcpBit = 0x01;
        const int llaBit = 0x02;
```

I'd like to help contribute here but I am not intimately familiar with the codebase yet to know what needs to change. Is it common to have vendor-specific changes? What would you suggest doing next?

EmmanuelP, 2022-10-07:
> This seems like it is because the bit index for LLA, DHCP, and Static is different for Imaging Source cameras

What makes you think it is different than what is in the wireshark dissector ? In the dissector, the bit is defined by its mask, not its position.

EmmanuelP, 2023-02-03:
Fixed by f23f130c30eccb22465d17c34b972ddf188e2b76

closed 2023-02-03 by EmmanuelP