AravisProject/aravis #729: Basler daA3840-45um: Bandwidth Control

ngenser, 2022-10-12:
The Basler daA3840-45um camera module (Pylon SDK) provides two flags for controlling the bandwidth:
- "Device Link Throughput Limit" => expected value: integer => implemented via `arv_camera_uv_set_bandwidth()`
- "Device Link Throughput Limit Mode" => expected value: bool (on/off) => not implemented so far

**Bug/Goal:**
 "Device Link Throughput Limit Mode" needs to be implemented for turning on bandwidth control.
It would be great to add this feature :-)

P.S.: Thanks for your great work!

**Camera description:**
 - Basler
 - daA3840-45um
 - USB3

**Platform description:**
 - Aravis version 0.8.21
 - OS: Kubuntu 22.04
 - Hardware x86_64

ngenser, 2022-10-17:
@EmmanuelP Can you imagine this feature being implemented in the near future?

EmmanuelP, 2022-10-17:
Yes sure. But it will happen sooner if someone creates a merge request :)

It should be pretty straightforward to add.

EmmanuelP, 2022-10-17:
I have just read the code, and Device Link Throughput Limit Mod should be already handled by Aravis. Why does it not work for you ?

EmmanuelP, 2023-05-19:
Long time no news. I'm closing this issue. Don't hesitate to reopen if you have the requested information.

Cheers.

closed 2023-05-19 by EmmanuelP