AravisProject/aravis #732: Spectral Instruments SI-1700 (RVT100) failing to take images with arv-camera-test

hackel6, 2022-10-20:
**Describe the bug**
We are trying to fold a new camera into our generic control architecture. We're finding that even with the simplest of test programs (arv-camera-tool), it seems to get the first frame and all others timeout. I have tried tweaking some of the GEV parameters listed [here](https://aravisproject.github.io/docs/aravis-0.8/ArvGvStream.html), but nothing seems to help with the timeouts. The same camera executes properly using Pleora eBUS SDK on Windows.

**Camera description:**
 - Spectral Instruments, Inc.-1700 (also known as RVT100)
 - Interface GigE

**Platform description:**
 - Aravis version: 0.8.19
 - OS: `Linux dvid-int-17 4.14.178-gentoo #17 SMP Thu May 27 17:04:21 PDT 2021 x86_64 Intel(R) Xeon(R) CPU E3-1220 V2 @ 3.10GHz GenuineIntel GNU/Linux`

**Additional context**
A short excerpt from the debug log:
`[11:30:27.650] 🅳 genicam> [GcCommand::execute] AcquisitionStart (0x2)
[11:30:27.650] 🅳 stream-thread> [GvStream::find_frame_data] Start frame 1
[11:30:27.650] 🅳 misc> [PixelFormat::to_gst_caps_string] 0x01100005 -> video/x-raw, format=(string)GRAY16_LE
[11:30:27.650] 🅳 sp> packet_type  =       ok (0x0000)
                     content_type = data-leader (0x0001)
                     frame_id     =        1
                     packet_id    =        0
                     data_size    =       36
                     payload_type = image
                     pixel format = video/x-raw, format=(string)GRAY16_LE
                     width        = 200
                     height       = 200
                     x_offset     = 0
                     y_offset     = 0
[11:30:27.671] 🅳 sp> packet_type  =       ok (0x0000)
                     content_type = data-trailer (0x0002)
                     frame_id     =        1
                     packet_id    =       56
                     data_size    =        8
[11:30:27.671] 🅳 stream-thread> [GvStream::check_frame_completion] Completed frame 1
[11:30:27.671] 🅳 stream-thread> [GvStream::close_frame] Close frame 1
[11:30:27.671] 🅳 stream-thread> [GvStream::find_frame_data] Start frame 2
[11:30:27.671] 🅳 misc> [PixelFormat::to_gst_caps_string] 0x01100005 -> video/x-raw, format=(string)GRAY16_LE
[11:30:27.671] 🅳 sp> packet_type  =       ok (0x0000)
                     content_type = data-leader (0x0001)
                     frame_id     =        2
                     packet_id    =        0
                     data_size    =       36
                     payload_type = image
                     pixel format = video/x-raw, format=(string)GRAY16_LE
                     width        = 200
                     height       = 200
                     x_offset     = 0
                     y_offset     = 0
[11:30:27.771] 🆆 stream-thread> [GvStream::check_frame_completion] Timeout for frame 2 at dt = 100401
[11:30:27.771] 🅳 stream-thread> [GvStream::close_frame] Close frame 2
[11:30:27.868] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 2 (last: 2)
[11:30:27.868] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =        2
                     packet_id    =        1
                     data_size    =     1464
[11:30:27.868] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 2 (last: 2)
[11:30:27.868] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =        2
                     packet_id    =        2
                     data_size    =     1464
[11:30:27.869] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 2 (last: 2)
[11:30:27.869] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =        2
                     packet_id    =        3
                     data_size    =     1464
[11:30:27.869] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 2 (last: 2)
[11:30:27.869] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
[... 304 lines cut ...]
[11:30:27.884] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =        2
                     packet_id    =       55
                     data_size    =      944
[11:30:27.886] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 2 (last: 2)
[11:30:27.887] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-trailer (0x0002)
                     frame_id     =        2
                     packet_id    =       56
                     data_size    =        8
[11:30:27.887] 🅳 stream-thread> [GvStream::find_frame_data] Start frame 3
[11:30:27.887] 🅳 misc> [PixelFormat::to_gst_caps_string] 0x01100005 -> video/x-raw, format=(string)GRAY16_LE
[11:30:27.887] 🅳 sp> packet_type  =       ok (0x0000)
                     content_type = data-leader (0x0001)
                     frame_id     =        3
                     packet_id    =        0
                     data_size    =       36
                     payload_type = image
                     pixel format = video/x-raw, format=(string)GRAY16_LE
                     width        = 200
                     height       = 200
                     x_offset     = 0
                     y_offset     = 0
[11:30:27.987] 🆆 stream-thread> [GvStream::check_frame_completion] Timeout for frame 3 at dt = 100416`

EmmanuelP, 2022-10-20:
Hi,

For frame1, it takes 20ms to receive all the data after the leader packet.
For frame2, for some reason, the missing data packets (all ?) seem to arrive 200ms after the leader packet, after the 100ms timeout. It takes 20ms to receive them, the same duration as for frame 1.

It is surprising the leader packet for frame 2 is received just after the trailer packet of frame1 (same millisecond).

 Did you try to increase `frame-retention` to 300000µs ?

hackel6, 2022-10-20:
Making that change reduced the failures/timeouts to zero in arv-camera-test. Thank you!
We will try adding that to our generic camera controller code.
What are the consequences (if any) of making that change to control of existing cameras (we control many with the same software)? I see from the documentation that it might increase maximum frame latency. Currently, we set it to 25,000 for streaming operations and leave it as default (100,000, i guess) for single image ops.

Thanks so much!

EmmanuelP, 2022-10-21:
> What are the consequences (if any) of making that change to control of existing cameras (we control many with the same software)? I see from the documentation that it might increase maximum frame latency.

The consequence is an increased latency of frame[n+1] if frame[n] is incomplete, as it will not be put in the output buffer pool until the reception of frame[n] times out.

I guess you should check the packet timing using wireshark, and report the behaviour to the camera manufacturer if it is confirmed the leader packet  is emitted just after the trailer packet of the previous frame. I think the camera should emit the leader packet just before the first image packet. But I can not say that for sure, as we are not allowed to check in the GigEVision documentation.

closed 2022-10-21 by EmmanuelP

hackel6, 2022-10-31:
We reached out the vendor, and their response was: 

> There is no mention of a timing requirement between the leader packet and the first data packet in the GigE Vision specification

EmmanuelP, 2022-11-03:
Thanks Brian.