AravisProject/aravis #734: Segfault on arv_stream_pop_buffer()

gotnone, 2022-10-26:
**Describe the bug**
Segfault on windows (msys2) when attempting to retrieve a buffer.

**To Reproduce**
Steps to reproduce the behavior:
Set up camera for start acquisition with open stream and a buffer attached to the stream.
Attempt to retrieve a buffer with arv_stream_pop_buffer, or arv_stream_try_pop_buffer.

**Expected behavior**
A clear and concise description of what you expected to happen.
Buffer should be successfully popped from the stream.

**Camera description:**
 - Manufacturer Keyence
 - Model VJ-3002
 - Interface [e.g. USB3] Ethernet

**Platform description:**
 - Aravis version 0.8.23 git 0df6d6aa9eb54454ac821c8db5e58a02a5e5f2a4  also occurs in the msys aravis-0.8 version 0.8.22
 - OS: [e.g. Fedora 20] Windows 10
 - Hardware [e.g. x86_64] x86_64

**Additional context**
the value of n_msgs in arv_gvstream.c is -1 at line 861.  This results in a problem with  the loop at line 868 where _process_packet attempts to read past the end of the frame.

n_msgs = g_socket_receive_messages (thread_data->socket,
		 					    packet_im,
		 					    ARV_GV_STREAM_NUM_BUFFERS,
		 					    G_SOCKET_MSG_NONE,
		 					    NULL,
		 					    NULL);

ARV_DEBUG=all:3

[17:33:00.429] ≡ƒà│ genicam> [GcEnumeration::set_string_value] value = 1 - string = SetupMode
[17:33:00.430] ≡ƒà│ genicam> [GcRegisterNode::_set_integer_value] address = 0xa300, value = 0x1
[17:33:00.931] ≡ƒåå device> [GvDevice::write_memory] Ack reception timeout
[17:33:00.932] ≡ƒåå device> [GvDevice::write_memory] Ack reception timeout
[17:33:00.937] ≡ƒà│ genicam> [GcEnumeration::set_string_value] value = 1 - string = Line
[17:33:00.937] ≡ƒà│ genicam> [GcRegisterNode::_set_integer_value] address = 0xa218, value = 0x1
[17:33:00.938] ≡ƒà╕ device> [GvDevice::write_memory] Unexpected answer (0x00)
[17:33:01.366] ≡ƒà│ genicam> [17:33:01.366] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[GcEnumeration::set_string_value] value = 0 - string = RunMode
[17:33:01.367] ≡ƒà│ genicam> [GcRegisterNode::_set_integer_value] address = 0xa300, value = 0x0
[17:33:01.848] ≡ƒà│ genicam> [GcRegisterNode::_set_integer_value] address = 0xd440, value = 0x1
[17:33:01.861] ≡ƒà│ genicam> [GcEnumeration::set_string_value] value = 0 - string = SingleFrame
[17:33:01.862] ≡ƒà│ genicam> [GcRegisterNode::_set_integer_value] address = 0xa200, value = 0x0
[17:33:01.864] ≡ƒà│ genicam> [GcRegisterNode::_set_integer_value] address = 0xa22c, value = 0x1
[17:33:01.865] ≡ƒà│ genicam> [GcEnumeration::set_string_value] value = 300 - string = FilteredImage1
[17:33:01.865] ≡ƒà│ genicam> [GcRegisterNode::_set_integer_value] address = 0xa230, value = 0x12c
[17:33:01.867] ≡ƒà│ genicam> [GcRegisterNode::_get_integer_value] address = 0xa090, value = 0x4bb400
[17:33:01.869] ≡ƒà│ genicam> [GcRegisterNode::_set_integer_value] address = 0xa204, value = 0x1
[17:33:01.888] ≡ƒà│ genicam> [GcCommand::execute] AcquisitionStart (0x1)
before arv_stream_pop_buffer
[17:33:02.371] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[17:33:03.374] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[17:33:04.381] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[17:33:04.598] ≡ƒà│ stream-thread> [GvStream::find_frame_data] Start frame 1
[17:33:04.599] ≡ƒà│ misc> [PixelFormat::to_gst_caps_string] 0x01080001 -> video/x-raw, format=(strin
g)GRAY8
[17:33:04.600] ≡ƒà│ sp> packet_type  =       ok (0x0000)
                     content_type = data-leader (0x0001)
                     frame_id     =        1
                     packet_id    =        0
                     data_size    =       36
                     payload_type = image
                     pixel format = video/x-raw, format=(string)GRAY8
                     width        = 2432
                     height       = 2040
                     x_offset     = 0
[... 91 lines cut ...]
[17:33:04.655] ≡ƒà│ stream-thread> [GvStream::process_packet] Duplicated packet 68 for frame 1
[17:33:04.656] ≡ƒà│ sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =        1
                     packet_id    =       68
                     data_size    =     1024

Thread 12 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 30020.0x47d4]
0x00007ffbca97cdb3 in arv_gvsp_packet_has_extended_ids (packet=0x9b4) at ../src/arvgvspprivate.h:211
211             return (packet->header[2] & ARV_GVSP_PACKET_EXTENDED_ID_MODE_MASK) != 0;
(gdb) bt
#0  0x00007ffbca97cdb3 in arv_gvsp_packet_has_extended_ids (packet=0x9b4)
    at ../src/arvgvspprivate.h:211
#1  0x00007ffbca97ea3d in _process_packet (thread_data=0x1dd8aeec750, packet=0x9b4,
    packet_size=1032, time_us=1137931222547) at ../src/arvgvstream.c:711
#2  0x00007ffbca97f20d in _loop (thread_data=0x1dd8aeec750) at ../src/arvgvstream.c:869
#3  0x00007ffbca97f354 in arv_gv_stream_thread (data=0x1dd8aeec750) at ../src/arvgvstream.c:1192
#4  0x00007ffbccac79d1 in ?? () from C:\msys64\ucrt64\bin\libglib-2.0-0.dll
#5  0x00007ffbccaf63a7 in ?? () from C:\msys64\ucrt64\bin\libglib-2.0-0.dll
#6  0x00007ffc3f2d1bb2 in ucrtbase!_configthreadlocale () from C:\WINDOWS\System32\ucrtbase.dll
#7  0x00007ffc3fdb7034 in KERNEL32!BaseThreadInitThunk () from C:\WINDOWS\System32\kernel32.dll
#8  0x00007ffc41c226a1 in ntdll!RtlUserThreadStart () from C:\WINDOWS\SYSTEM32\ntdll.dll
#9  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)

EmmanuelP, 2022-10-27:
Does the crash happen every time ?

If so, I would be interested in the error message you will get in the log using the changes in #736

EmmanuelP, 2022-10-27:
Thanks.

I'm still interested in any informations on why the receive_messages call fails, even though we are checking the number of revents before. The reported error is: `A non-blocking socket operation could not be completed immediately`. In the log you have posted, it seems to happen a lot at the start of the streaming, then it sill happen, but rarely.

commit 7505b5de (2022-10-27): gvstream: handle receive_message error

I though an error here should not happen, as we check revents before calling
receive_messages.

Fix #734
commit 46125f94 (2022-10-27): gvstream: handle receive_message error

I though an error here should not happen, as we check revents before calling
receive_messages.

It seems to only happen on Windows and msys2. When the call fails, the error is:

```
A non-blocking socket operation could not be completed immediately
```

Fix #734
commit a994c5fa (2022-10-27): gvstream: handle receive_message error

I though an error here should not happen, as we check revents before calling
receive_messages.

It seems to only happen on Windows and msys2. When the call fails, the error is:

```
A non-blocking socket operation could not be completed immediately
```

Fix #734
closed 2022-10-27 by EmmanuelP