AravisProject/aravis #737: Repeating U3V Leader Packet

Dennis10231, 2022-11-08:
**Describe the bug**
Using ARV_DEBUG=all:3 I can observe the following with a Point Grey Cameleon3 CM3-U3-31S4C:

[07:02:10.109]  sp> USB transfer error: LIBUSB_ERROR_TIMEOUT
[07:02:10.109]  sp> Asking for 1048576 bytes
[07:02:10.342]  sp> Received 52 bytes
[07:02:10.342]  misc> [PixelFormat::to_gst_caps_string] 0x02180014 -> video/x-raw, format=(string)RGB
[07:02:10.342]  sp> **packet_type  = leader**
                     size         = 52
                     **frame id     = 0**
                     payload_type = image
                     pixel format = video/x-raw, format=(string)RGB
                     width        = 768
                     height       = 768
                     x_offset     = 0
                     y_offset     = 0
[07:02:10.342]  sp> Asking for 1048576 bytes
[07:02:10.357]  sp> Received 1048576 bytes
[07:02:10.357]  sp> packet_type  = image
[07:02:10.357]  sp> Asking for 720896 bytes
[07:02:10.368]  sp> Received 720896 bytes
[07:02:10.368]  sp> packet_type  = image
[07:02:10.368]  sp> Asking for 48 bytes
[07:02:10.368]  sp> Received 0 bytes
[07:02:10.368]  misc> [PixelFormat::to_gst_caps_string] 0x02180014 -> video/x-raw, format=(string)RGB
[07:02:10.368]  sp> **packet_type  = leader**
                     size         = 52
                     **frame id     = 0**
                     payload_type = image
                     pixel format = video/x-raw, format=(string)RGB
                     width        = 768
                     height       = 768
                     x_offset     = 0
                     y_offset     = 0
[07:02:10.368]  stream-thread> New leader received while a buffer is still open
[07:02:10.368]  sp> Asking for 1048576 bytes
FAIL(3)  8.11.2022 07:02:10 12171
[07:02:10.368]  sp> Received 0 bytes
[07:02:10.368]  sp> packet_type  = image
[07:02:10.368]  sp> Asking for 1048576 bytes
[07:02:10.369]  sp> Received 32 bytes
[07:02:10.369]  sp> **packet_type  = trailer**
                     size         = 32
                     **frame id     = 0**
                     payload_size = 1769472
[07:02:10.369]  stream-thread> Received 0 bytes - expected 1769472
[07:02:10.369]  stream-thread> Incomplete image received, dropping (received 0 / expected 1769472)
[07:02:10.369]  sp> Asking for 1048576 bytes
FAIL(5)  8.11.2022 07:02:10 0
[07:02:10.537]  sp> Received 52 bytes
[07:02:10.537]  misc> [PixelFormat::to_gst_caps_string] 0x02180014 -> video/x-raw, format=(string)RGB
[07:02:10.537]  sp> **packet_type  = leader**
                     size         = 52
                     **frame id     = 1**
                     payload_type = image
                     pixel format = video/x-raw, format=(string)RGB
                     width        = 768
                     height       = 768
                     x_offset     = 0
                     y_offset     = 0
[07:02:10.537]  sp> Asking for 1048576 bytes
[07:02:10.552]  sp> Received 1048576 bytes
[07:02:10.552]  sp> packet_type  = image
[07:02:10.552]  sp> Asking for 720896 bytes
[07:02:10.563]  sp> Received 720896 bytes
[07:02:10.563]  sp> packet_type  = image
[07:02:10.563]  sp> Asking for 48 bytes
[07:02:10.563]  sp> Received 0 bytes
[07:02:10.563]  misc> [PixelFormat::to_gst_caps_string] 0x02180014 -> video/x-raw, format=(string)RGB
[07:02:10.563]  sp> **packet_type  = leader**
                     size         = 52
                     **frame id     = 1**
                     payload_type = image
                     pixel format = video/x-raw, format=(string)RGB
                     width        = 768
                     height       = 768
                     x_offset     = 0
                     y_offset     = 0
[07:02:10.563]  stream-thread> New leader received while a buffer is still open
[07:02:10.563]  sp> Asking for 1048576 bytes
FAIL(3)  8.11.2022 07:02:10 194
[07:02:10.563]  sp> Received 0 bytes
[07:02:10.563]  sp> packet_type  = image
[07:02:10.563]  sp> Asking for 1048576 bytes
[07:02:10.563]  sp> Received 32 bytes
[07:02:10.563]  sp> **packet_type  = trailer**
                     size         = 32
                     **frame id     = 1**
                     payload_size = 1769472
[07:02:10.563]  stream-thread> Received 0 bytes - expected 1769472
[07:02:10.563]  stream-thread> Incomplete image received, dropping (received 0 / expected 1769472)
[07:02:10.563]  sp> Asking for 1048576 bytes
FAIL(5)  8.11.2022 07:02:10 0
[07:02:11.043]  sp> USB transfer error: LIBUSB_ERROR_TIMEOUT

**Expected behavior**
What I see is a pattern
Leader / 0
Leader / 0
-> Aravis: Error
Trailer / 0
-> But at this time, Image 0 is complete!

The leader gets repeated, leading to an error.
But the second leader is ignored (= not processed), so when the trailer is received, no image is delivered.

I would expect the image to be delivered.

Interestingly, this only happens after connecting the camera to the PC.
If I restart any camera application, this problem goes away.

**Camera description:**
 - Manufacturer: Point Grey / FLIR
 - Model:  Cameleon3 CM3-U3-31S4C
 - Interface: USB3

**Platform description:**
 - Aravis version: 0.8.22
 - OS: Ubuntu 22.04
 - Hardware:  x86_64

Dennis10231, 2022-11-08:
Using Wireshark with usbcap, I get this:
[image]
No repeated leader?

Dennis10231, 2022-11-08:
I have captured another sample within the same container as my application. 
As previously mentioned, wireshark / tcpdump only see one leader packet, not two.

[usbmon.zip](https://github.com/AravisProject/aravis/files/9960262/usbmon.zip)

Dennis10231, 2022-11-08:
Here are two wireshark dumps next to each other.
The one on the left produces the error, the one on the right does not.

[image]

EmmanuelP, 2022-11-08:
The second leader event happens because there was no transferred data, even if the bulk transfer succeed. In the absence of a transfer size check, aravis uses the old content of the transfer buffer, which contains a leader packet.

Please try #738

Dennis10231, 2022-11-09:
I can confirm that #738 solves this issue.
Great Work! Thank you very much!

PR #738 (2022-11-09): uvstream: check transferred size before using the result
It happens arv_uv_device_bulk_transfer succeed, but with no data transferred. Check the number of transferred bytes before trying to interpret the buffer content.

Fixes #737
commit 7a18ea64 (2022-11-09): uvstream: check transferred size before using the result

It happens arv_uv_device_bulk_transfer succeed, but with no data transferred.
Check the number of transferred bytes before trying to interpret the buffer
content.

Fixes #737
closed 2022-11-09 by EmmanuelP