AravisProject/aravis #741: arv_camera_get_device_id return null

mlorgan, 2022-11-23:
**Describe the bug**
arv_camera_get_device_id() return null

**To Reproduce**
run arv-roi-test
but no KO reported in /fake/camera-api test

**Expected behavior**
arv_camera_get_device_id() should return same as arv_get_device_id()

**Camera description:**
 - Basler
 - a2A1920-160ucBAS
 - USB3

**Platform description:**
 - Aravis version 0.8.22
 - OS: Ubuntu 18.04.6 LTS / kernel 4.9.253-tegra
 - aarch64

EmmanuelP, 2022-12-27:
Hi,

It is a bit confusing, but `arv_camera_get_device_id()` is no supposed to return the same thing as `arv_device_get_id()`.

`arv_camera_get_device_id()` returns the content of the `DeviceID` feature string, while `arv_device_get_id()` returns a unique identifier built from the manufacturer, model and serial number of the device.

I guess we could add a `arv_camera_get_id()` that would return the Aravis built Id.

eudoxos, 2023-10-15:
I am just in the need of this, but fail to see `arv_device_get_id` — tehere `arv_get_device_id` (by index). Can this information be retrieved from within `ArvDevice*`?