AravisProject/aravis #742: When the image size is large, the streaming error is reported

zhaotyer, 2022-11-29:
**Describe the bug**
the device is 8K camera,When I don't make the camera size smaller，it report:
Aravis-CRITICAL **: 11:30:31.089: arv_stream_push_buffer: assertion 'ARV_IS_STREAM (stream)' failed

**To Reproduce**
when i run this pipeline, the width is 8240 height is 5628
`gst-launch-1.0 aravissrc  gain=10 ! video/x-raw,format=GRAY8,framerate=4/1 ! videoconvert ! ximagesink sync="false" --gst-debug=1,aravissrc:7`
the error log is:
:00:00.046148292 20749 0x5598b032e550 LOG                aravissrc gstaravis.c:138:gst_aravis_get_caps:<aravis0> Available caps = ANY
0:00:00.046186340 20749 0x5598b032e550 LOG                aravissrc gstaravis.c:138:gst_aravis_get_caps:<aravis0> Available caps = ANY
0:00:00.046999490 20749 0x5598b032e550 LOG                aravissrc gstaravis.c:138:gst_aravis_get_caps:<aravis0> Available caps = ANY
0:00:00.047806486 20749 0x5598b032e550 LOG                aravissrc gstaravis.c:138:gst_aravis_get_caps:<aravis0> Available caps = ANY
0:00:00.047889709 20749 0x5598b032e550 LOG                aravissrc gstaravis.c:138:gst_aravis_get_caps:<aravis0> Available caps = ANY
Setting pipeline to PAUSED ...
0:00:00.048965981 20749 0x5598b032e550 LOG                aravissrc gstaravis.c:297:gst_aravis_start:<aravis0> Open camera '(null)'
0:00:00.159354638 20749 0x5598b032e550 LOG                aravissrc gstaravis.c:86:gst_aravis_get_all_camera_caps:<aravis0> Get all camera caps
Pipeline is live and does not need PREROLL ...
0:00:00.166303516 20749 0x5598b038d000 LOG                aravissrc gstaravis.c:138:gst_aravis_get_caps:<aravis0> Available caps = video/x-raw, format=(string)GRAY8, width=(int)[ 1, 8240 ], height=(int)[ 1, 5628 ], framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, format=(string)GRAY16_LE, width=(int)[ 1, 8240 ], height=(int)[ 1, 5628 ], framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, format=(string)GRAY16_LE, width=(int)[ 1, 8240 ], height=(int)[ 1, 5628 ], framerate=(fraction)[ 0/1, 2147483647/1 ]
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:00.170883789 20749 0x5598b038d000 LOG                aravissrc gstaravis.c:432:gst_aravis_fixate_caps:<aravis0> Fixate caps
0:00:00.170973850 20749 0x5598b038d000 LOG                aravissrc gstaravis.c:156:gst_aravis_set_caps:<aravis0> Requested caps = video/x-raw, format=(string)GRAY8, width=(int)8240, height=(int)5628, framerate=(fraction)4/1, pixel-aspect-ratio=(fraction)1/1
0:00:00.291216308 20749 0x5598b038d000 DEBUG              aravissrc gstaravis.c:186:gst_aravis_set_caps:<aravis0> Frame rate = 4 Hz
0:00:00.304832592 20749 0x5598b038d000 DEBUG              aravissrc gstaravis.c:197:gst_aravis_set_caps:<aravis0> Buffer timeout = 2000000 µs
0:00:00.304908547 20749 0x5598b038d000 DEBUG              aravissrc gstaravis.c:199:gst_aravis_set_caps:<aravis0> Actual frame rate = 0 Hz
0:00:00.304948314 20749 0x5598b038d000 DEBUG              aravissrc gstaravis.c:203:gst_aravis_set_caps:<aravis0> Auto Gain = continuous
0:00:00.836419196 20749 0x5598b038d000 DEBUG              aravissrc gstaravis.c:215:gst_aravis_set_caps:<aravis0> Auto Exposure = continuous
(gst-launch-1.0:20749): GLib-GObject-CRITICAL **: 11:30:31.087: g_object_set: assertion 'G_IS_OBJECT (object)' failed
(gst-launch-1.0:20749): Aravis-CRITICAL **: 11:30:31.088: arv_stream_push_buffer: assertion 'ARV_IS_STREAM (stream)' failed
(gst-launch-1.0:20749): Aravis-CRITICAL **: 11:30:31.088: arv_stream_push_buffer: assertion 'ARV_IS_STREAM (stream)' failed
(gst-launch-1.0:20749): Aravis-CRITICAL **: 11:30:31.090: arv_stream_push_buffer: assertion 'ARV_IS_STREAM (stream)' failed
0:00:00.849625367 20749 0x5598b038d000 LOG                aravissrc gstaravis.c:261:gst_aravis_set_caps:<aravis0> Start acquisition
(gst-launch-1.0:20749): Aravis-CRITICAL **: 11:30:31.118: arv_stream_timeout_pop_buffer: assertion 'ARV_IS_STREAM (stream)' failed
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis0:
streaming stopped, reason error (-5)
Execution ended after 0:00:00.711755474
Setting pipeline to NULL ...
0:00:00.889358698 20749 0x5598b032e550 DEBUG              aravissrc gstaravis.c:324:gst_aravis_stop:<aravis0> Stop acquisition
Freeing pipeline ...

when i set width 4096 height 2160, it work well
`gst-launch-1.0 aravissrc  gain=10 ! video/x-raw,format=GRAY8,width=4096,height=2160,framerate=4/1 ! videoconvert ! ximagesink sync="false" --gst-debug=1,aravissrc:7`

**Expected behavior**
it's work well

**Camera description:**
 - MindVision-XG4701M-052112020002
 - 8K

**Platform description:**
 - Aravis version 0.6.4
 - ubuntu 20.04
 - x86_64

**Additional context**

EmmanuelP, 2022-12-27:
Could you test your camera using `arv-camera-test-0.8`, and report the output here ?

EmmanuelP, 2023-05-19:
No news, I'm closing this report. Don't hesitate to reopen it if you have the requested informations.

Cheers.

closed 2023-05-19 by EmmanuelP