AravisProject/aravis #752: A lot of timeout errors with 4K camera on Windows10

wintersui, 2022-12-27:
**Describe the bug**
I have used MSys2 to compile the source on Win10.
Net Adapter of my PC is 1000Mb.
When used arv-camera-test command to test the camera, a lot of timeout errors were showed on console.
My camera's resolution is 4000 x 3000, frames are almost timeout error images.
But, I changed the resolution to smaller size, it works well.

I have tried to modify the source code, find the problem.
In arvgvstream.c:_loop() method, I removed all source, only left the g_socket_receive_messages().
The average speed was still 20MiB/s.

I have used arv-viewer.exe to test the camera, GUI Page showed the same timeout errors and frames speed,
camera's image update rate was very very slow.

Can you fixed the timeout problem ?

**Expected behavior**
On my Ubuntu20 PC, Fps was ok, camera's image was well too.
 frames/s - 90MiB/s

**Camera description:**
 - iDS
 - GV-5890CP-C-HQ
 - Interface [GigE]

**Platform description:**
 - Aravis 0.8.22
 - OS: [Win10]
 - Hardware [x86_64]

**Additional context**
I use the same camera on Ubuntu20, it works well.

EmmanuelP, 2022-12-27:
Some questions:

- Did you try to monitor the CPU use ? I'm not used to the Windows platform, but it would help if we know if the issue is in Aravis itself, or in Windows network code.
- Did you use Aravis on Ubuntu ?
- Using arv-camera-test-0.8, did you try to increase the socket buffer size ?

wintersui, 2022-12-28:
Q: Did you try to monitor the CPU use?
A: CPU usage was 50%. I do not think it's CPU's problem.

Q: Did you use Aravis on Ubuntu ?
A: I also use aravis on Ubuntu, it works well. But, Windows is not good.

Q: Using arv-camera-test-0.8, did you try to increase the socket buffer size ?
A: Yes, I have tried to increase the socket buffer size from 16 to 1600, the time out still happened.

EmmanuelP, 2022-12-28:
Please attach the output of: `arv-camera-test-0.8 --duration=10 -d all`

wintersui, 2022-12-29:
$ arv-camera-test-0.8 --duration=10 -d all
Looking for the first available camera
[14:10:43.298] 🅸 interface> [GvDiscoverSocket::new] Add interface 10.194.222.174 (10.194.223.255)
[14:10:43.449] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.0.11 (192.168.0.255)
[14:10:43.449] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.255.255.255)
[14:10:43.453] 🅸 interface> [GvInterface::discovery] Device 'GEV-MV-CU120-10GC-K71604143' found (interface 192.168.0.11) user_id '' - MAC '34:bd:20:10:61:4c'
[14:10:43.454] 🅸 device> [GvDevice::new] Interface address = 192.168.0.11
[14:10:43.454] 🅸 device> [GvDevice::new] Device address = 192.168.0.199
[14:10:43.457] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:GigE_Vision_1_V4_0_4.zip;100000;17b04' at 0x200
[14:10:43.457] 🅸 device> [GvDevice::load_genicam] Xml address = 0x100000 - size = 0x17b04 - GigE_Vision_1_V4_0_4.zip
[14:10:43.712] 🅸 device> [GvDevice::load_genicam] Zipped xml data
[14:10:43.715] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'xmlns:mstns'
[14:10:43.949] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPAddrCalc'
[14:10:43.954] 🅸 device> [GvDevice::new] Device endianness = big
[14:10:43.954] 🅸 device> [GvDevice::new] Packet resend     = yes
[14:10:43.954] 🅸 device> [GvDevice::new] Write memory      = yes
[14:10:43.954] 🅸 device> [GvDevice::new] Legacy endianness handling = yes
vendor name            = GEV
model name             = MV-CU120-10GC
device serial number   = K71604143
image width            = 4024
image height           = 3036
horizontal binning     = 1
vertical binning       = 1
exposure               = 5000 µs
gain                   = 0 dB
payload                = 12216864 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 4000 ns
[14:10:44.002] 🅸 device> [GvDevice::create_stream] Number of stream channels = 1
[14:10:44.121] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 1500
[14:10:44.212] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 860
[14:10:44.290] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 540
[14:10:44.386] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 380
[14:10:44.482] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 300
[14:10:44.593] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 260
[14:10:44.704] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 240
[14:10:44.828] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 232
[14:10:44.912] 🅸 device> [GvDevice::auto_packet_size] Packet size set to 1500 bytes
[14:10:44.924] 🅸 device> [GvStream::stream_new] Packet size = 1500 byte(s)
[14:10:44.930] 🅸 stream> [GvStream::stream_new] Destination stream port = 61075
[14:10:44.930] 🅸 stream> [GvStream::stream_new] Source stream port = 9001
[14:10:44.930] 🅸 stream> [GvStream::loop] Standard socket method
gv packet size         = 1500 bytes
  0 frame/s  -       0 MiB/s
[14:10:46.273] 🆆 stream-thread> [GvStream::check_frame_completion] Timeout for frame 210 at dt = 1197893
[14:10:46.495] 🆆 device> [GvDevice::read_register] Ack reception timeout
[14:10:46.495] 🆆 device> [GvDevice::read_register] Ack reception timeout
[14:10:46.944] 🆆 stream-thread> [GvStream::check_frame_completion] Timeout for frame 211 at dt = 1550230
  0 frame/s  -       0 MiB/s - 2 errors
[14:10:47.045] 🅸 stream-thread> [GvStream::missing_packet_check] Maximum number of requests reached at dt = 57792, n_packet_requests = 7977 (8347 packets/frame), frame_id = 213
[14:10:47.822] 🆆 stream-thread> [GvStream::check_frame_completion] Timeout for frame 212 at dt = 1506722
[14:10:47.823] 🆆 stream-thread> [GvStream::check_frame_completion] Timeout for frame 213 at dt = 833253
[14:10:47.833] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 212 (last: 214)
[14:10:47.833] 🅸 sp> packet_type  =   resend (0x0100)
                     content_type = data-block (0x0003)
                     frame_id     =      212
                     packet_id    =     4366
                     data_size    =     1464
[14:10:47.834] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 212 (last: 214)
[14:10:47.834] 🅸 sp> packet_type  =   resend (0x0100)
                     content_type = data-block (0x0003)
                     frame_id     =      212
                     packet_id    =     4479
                     data_size    =     1464
[14:10:47.834] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 212 (last: 214)
[14:10:47.834] 🅸 sp> packet_type  =   resend (0x0100)
                     content_type = data-block (0x0003)
                     frame_id     =      212
[... 2922 lines cut ...]
                         <          0;           0;           0;           0
                         min         :      279192;           0;           0
                         max         : 1.57648e+06; 1.44789e+06;      117582
                         last max at :          13;       95625;      113213
                         counter     :          21:      113429:      113408
[14:10:55.360] 🅸 stream> [GvStream::finalize] n_completed_buffers    = 1
[14:10:55.361] 🅸 stream> [GvStream::finalize] n_failures             = 18
[14:10:55.361] 🅸 stream> [GvStream::finalize] n_underruns            = 0
[14:10:55.361] 🅸 stream> [GvStream::finalize] n_timeouts             = 18
[14:10:55.361] 🅸 stream> [GvStream::finalize] n_aborteds             = 2
[14:10:55.361] 🅸 stream> [GvStream::finalize] n_missing_frames       = 0
[... cut ...]
[14:10:55.363] 🅸 stream> [GvStream::finalize] n_resend_ratio_reached = 10
[14:10:55.363] 🅸 stream> [GvStream::finalize] n_resend_disabled      = 0
[14:10:55.363] 🅸 stream> [GvStream::finalize] n_duplicated_packets   = 8861
[14:10:55.363] 🅸 stream> [GvStream::finalize] n_transferred_bytes    = 166934404
[14:10:55.364] 🅸 stream> [GvStream::finalize] n_ignored_bytes        = 681338
[14:10:55.364] 🅸 stream> [Stream::finalize] Flush 48 buffer[s] in input queue
[14:10:55.364] 🅸 stream> [Stream::finalize] Flush 2 buffer[s] in output queue

gotnone, 2023-01-27:
I see your packet size is very small.  Have you enabled jumbo frames on your network adapter?  Is there a chance you have the [known issue](https://answers.microsoft.com/en-us/windows/forum/all/how-i-can-change-my-mtu-over-1500/beb54ab7-0334-43f7-83b3-7e21625aa2fd#:~:text=The%20January%202022%20update%20for%20Windows%2010%20%28KB5009543%29,above%201500.%207%20people%20found%20this%20reply%20helpful) with particular windows updates installed which break changing the MTU on a network interface?

EmmanuelP, 2023-05-19:
@wintersui , did you try what @gotnone suggested ?