AravisProject/aravis #753: Determine GigE vision version or adapt to older versions.

durnezj, 2023-01-17:
Hi,
I have developed a GigE camera management and capture system using the Aravis library which is working flawlessly. A big thanks for developing this library so that I can support multiple vendors without resorting to their proprietary software!
Up until now I have been using Basler and NET Gmbh type cameras which both support GigE version 2.0.
Now I have to add support for FLIR BlackFly cameras, which only support GigE version 1.2. This out of necessity due to component shortages on the to-develop product.

However, this standard does not seem to operate as I'm used to. The camera settings are the same as the other ones and I'm able to acquire images, but its BufferPayloadType is reported as ARV_BUFFER_PAYLOAD_TYPE_CHUNK_DATA (where I would expect IMAGE or IMAGE_EXTENDED_CHUNK), so when I try and obtain the dimensions it gives 0x0 because it is not treated as a valid image buffer in my code using the ARV_* buffer checks.

The payload size is however valid and equals the real width x height. When I save this data to file, using the dimensions of the full resolution or current ROI settings, it gives a perfect image in all pixel formats...
All the other functionality (setting/getting properties, auto modes,..) work perfectly as well. Are you aware of changes between those versions with regards to mainly image buffer acquisition/processing?

Now coming back to my main question: can I determine the GigE version (2.0, 1.2,...) using Aravis? Otherwise I will just stick to the method mentioned above.

Regards,
Durneztj

EmmanuelP, 2023-02-10:
Hi,

Thanks for the report.

When you receive buffers with a `ARV_BUFFER_PAYLOAD_TYPE_CHUNK_DATA` payload type, you probably have some chunk data enabled. Unfortunately, the leader packet of this type of payload does not have informations about the image properties. You may be able transfer these informations as chunk data, but still the different image property getters of the ArvBuffer API will not return meaningful values. That is because the streaming code of Aravis has no knowledge of which chunk data are enabled, and also knows nothing about the genicam data, which are necessary to retrieve data from the chunk data.

I don't think all of this is related to the version of GigEVision supported by the device.

I'm closing this issue as I think there is not a lot we can do in Aravis. Feel free to reopen if you don't think so :) . 

Cheers.

closed 2023-02-10 by EmmanuelP

durnezj, 2023-02-10:
Hi,
Thanks for the clarification! I indeed have chunk data enabled because we need the effective exposure time chunk data amongst others.
When I get the image buffer I check the dimensions, which are 0 width and 0 height for these 'problematic' image buffers. There my code went wrong because I pass these dimensions on to another part of the code who saves the image to file, and it thus crashed there.

As a fix I changed my application code to check if the dimensions are 0x0 and if they are, I grab the correct dimensions from the camera's current ROI settings. This has been working ever since in my application for different types of cameras.