AravisProject/aravis #755: Calling arv_camera_gv_auto_packet_size causes infinite loop

TIS-Edgar, 2023-01-24:
**Describe the bug**
Hi Emmanuel,
I hope you are doing well!
When opening a device and manually calling arv_camera_gv_auto_packet_size the program enters an infinite loop.
The current_size variable is repeatedly set to the same value.

Excerpt from running with ARV_DEBUG=all:3
> [12:11:25.872] 🅳 genicam> [GcRegisterNode::_set_integer_value] mask  = 0x80000000
[12:11:25.872] 🅳 genicam> [GcRegisterNode::_set_integer_value] address = 0xd04, value = 0xc0002328
[12:11:25.875] 🅳 genicam> [GcCommand::execute] GevSCPSFireTestPacket (0x1)
[12:11:25.875] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 9003
[12:11:25.875] 🅳 genicam> [GcRegisterNode::_set_integer_value] reglsb = 31, regmsb, 16, lsb = 0, msb = 15
[12:11:25.875] 🅳 genicam> [GcRegisterNode::_set_integer_value] value = 0x0000232b
[12:11:25.875] 🅳 genicam> [GcRegisterNode::_set_integer_value] mask  = 0x0000ffff
[12:11:25.875] 🅳 genicam> [GcRegisterNode::_set_integer_value] address = 0xd04, value = 0x4000232b
[12:11:25.876] 🅳 genicam> [GcRegisterNode::_get_integer_value] reglsb = 31, regmsb, 16, lsb = 0, msb = 15
[12:11:25.876] 🅳 genicam> [GcRegisterNode::_get_integer_value] value = 0x40002328
[12:11:25.876] 🅳 genicam> [GcRegisterNode::_get_integer_value] mask  = 0x0000ffff
[12:11:25.876] 🅳 genicam> [GcRegisterNode::_get_integer_value] address = 0xd04, value = 0x2328
[12:11:25.876] 🅳 genicam> [GcRegisterNode::_set_integer_value] reglsb = 0, regmsb, 0, lsb = 31, msb = 31
[12:11:25.876] 🅳 genicam> [GcRegisterNode::_set_integer_value] value = 0x00000001
[12:11:25.876] 🅳 genicam> [GcRegisterNode::_set_integer_value] mask  = 0x80000000
[12:11:25.876] 🅳 genicam> [GcRegisterNode::_set_integer_value] address = 0xd04, value = 0xc0002328
[12:11:25.879] 🅳 genicam> [GcCommand::execute] GevSCPSFireTestPacket (0x1)
[12:11:25.879] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 9003
[12:11:25.879] 🅳 genicam> [GcRegisterNode::_set_integer_value] reglsb = 31, regmsb, 16, lsb = 0, msb = 15
[12:11:25.879] 🅳 genicam> [GcRegisterNode::_set_integer_value] value = 0x0000232b
[12:11:25.879] 🅳 genicam> [GcRegisterNode::_set_integer_value] mask  = 0x0000ffff
[12:11:25.879] 🅳 genicam> [GcRegisterNode::_set_integer_value] address = 0xd04, value = 0x4000232b
[12:11:25.880] 🅳 genicam> [GcRegisterNode::_get_integer_value] reglsb = 31, regmsb, 16, lsb = 0, msb = 15
[12:11:25.880] 🅳 genicam> [GcRegisterNode::_get_integer_value] value = 0x40002328
[12:11:25.880] 🅳 genicam> [GcRegisterNode::_get_integer_value] mask  = 0x0000ffff
[12:11:25.880] 🅳 genicam> [GcRegisterNode::_get_integer_value] address = 0xd04, value = 0x2328
[12:11:25.880] 🅳 genicam> [GcRegisterNode::_set_integer_value] reglsb = 0, regmsb, 0, lsb = 31, msb = 31
[12:11:25.880] 🅳 genicam> [GcRegisterNode::_set_integer_value] value = 0x00000001
[12:11:25.880] 🅳 genicam> [GcRegisterNode::_set_integer_value] mask  = 0x80000000
[12:11:25.880] 🅳 genicam> [GcRegisterNode::_set_integer_value] address = 0xd04, value = 0xc0002328
[12:11:25.883] 🅳 genicam> [GcCommand::execute] GevSCPSFireTestPacket (0x1)
[12:11:25.883] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 9003

**To Reproduce**
Open a device and call arv_camera_gv_auto_packet_size

    ArvCamera* cam = arv_camera_new("ident", NULL);
    guint packet_size = arv_camera_gv_auto_packet_size(cam, NULL);

**Expected behavior**
Correct auto size negotiation.

**Camera description:**
 - Manufacturer: TIS
 - Model: Tested with DFK 33GX265e and DFK 33GP1300
 - Interface: Gigabit interface with mtu 9000

**Platform description:**
 - Aravis version: Current main (f06f3660)
 - OS: Debian Testing, Debian 10, Ubuntu 20.04
 - Hardware x86_64

**Additional context**
The genicam xml of the dfk33gx365e can be found here: https://gist.github.com/TIS-Edgar/0115974cb9fc82513c2fc255e205da05

If you require any additional information please let me know.
And once again thank you for your work!

EmmanuelP, 2023-01-24:
Hi Edgar,

Thanks for the report. Please try #756 

In the device xml data, Inc is not set for GevSCPSPacketSize node, which was triggering the infinite loop. The aravis code should be more robust now.

TIS-Edgar, 2023-01-24:
Hi Emmanuel,

#756 seems to fix the issue. I am not able to reproduce it with that commit.

Thank you very much!

PR #756 (2023-01-24): gvdevice: fix infinite loop in auto packet size
When the increment of GevSCPSPacketSize is not correctly returned by the device, the difference between the computed packet size and the actual packet size can lead to an infinite loop.

Fix #755
commit 5602abb0 (2023-01-24): gvdevice: fix infinite loop in auto packet size

When the increment of GevSCPSPacketSize is not correctly returned by the device,
the difference between the computed packet size and the actual packet size can
lead to an infinite loop.

Fix #755
closed 2023-01-24 by EmmanuelP