AravisProject/aravis #758: Building aravis on macOS with apple silicon failed

BZandi, 2023-01-28:
**Describe the bug**
I cannot build aravis on macOS with apple silicon. After using the commands stated in **To Reproduce**, the following error message appears:

```shell
Compiler for C supports arguments -Wduplicated-branches: NO 
Compiler for C supports arguments -Wimplicit-fallthrough: YES 
Compiler for C supports arguments -Wmisleading-indentation: YES 
Compiler for C supports arguments -Wstrict-prototypes: YES 
Compiler for C supports arguments -Wunused: YES 
Compiler for C supports arguments -Wunused-variable: YES 
Compiler for C supports arguments -Wdeclaration-after-statement: YES 
Compiler for C supports arguments -Wformat=2: YES 
Compiler for C supports arguments -Wimplicit-function-declaration: YES 
Compiler for C supports arguments -Winit-self: YES 
Compiler for C supports arguments -Wmissing-include-dirs: YES 
Compiler for C supports arguments -Wmissing-prototypes: YES 
Compiler for C supports arguments -Wpointer-arith: YES 
Compiler for C supports arguments -Wformat-nonliteral: YES 
Compiler for C supports arguments -Wenum-conversion: YES 
Compiler for C supports arguments -Wmaybe-uninitialized: NO 
Compiler for C supports arguments -fvisibility=hidden: YES 
Found pkg-config: /opt/homebrew/bin/pkg-config (0.29.2)
Run-time dependency glib-2.0 found: YES 2.72.3
Run-time dependency gobject-2.0 found: YES 2.72.3
Run-time dependency gio-2.0 found: YES 2.72.3
Run-time dependency libxml-2.0 found: YES 2.9.4
Run-time dependency zlib found: YES 1.2.11
Run-time dependency libusb-1.0 found: YES 1.0.26
Library m found: YES
Program glib-mkenums mkenums found: NO

src/meson.build:219:0: ERROR: Program 'glib-mkenums mkenums' not found or not executable
```

**To Reproduce**
I followed the tutorial on the webpage and used the following commands in the terminal
```bash
brew install gettext intltool gtk-doc libxml2 meson libusb
brew install gtk+3 gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad libnotify gnome-icon-theme
meson configure -Dviewer=enabled build
meson setup build
```

**Expected behavior**
My goal is to start the aravis viewer.

**Platform description:**
 - OS: MacOS Ventura (13.2)
 - Hardware Macbook Pro (M1)/ Apple Silicon

Any suggestions on how to proceed?

EmmanuelP, 2023-01-28:
`brew install glib` ?

BZandi, 2023-01-28:
Glib is already installed via Homebrew. Reinstall did not help. I also tried to put the lib into the path and pkgconfig using the following commands, but it did not work:
```shell
export PKG_CONFIG_PATH="/opt/homebrew/Cellar/glib/2.72.3/lib/pkgconfig:$PKG_CONFIG_PATH"
export DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:/opt/homebrew/Cellar/glib/2.72.3/lib/
```

EmmanuelP, 2023-01-28:
glib-mkenums should be in the glib package. What gives `brew list glib` ?

BZandi, 2023-01-28:
``glib-mkenums`` is not listed in the output when calling ``brew list glib``. Could it be possible that ``glib-mkenums``is removed in glib v2.72.3?

```shell
/opt/homebrew/Cellar/glib/2.72.3/bin/gdbus
/opt/homebrew/Cellar/glib/2.72.3/bin/gio
/opt/homebrew/Cellar/glib/2.72.3/bin/gio-querymodules
/opt/homebrew/Cellar/glib/2.72.3/bin/glib-compile-resources
/opt/homebrew/Cellar/glib/2.72.3/bin/glib-compile-schemas
/opt/homebrew/Cellar/glib/2.72.3/bin/glib-gettextize
/opt/homebrew/Cellar/glib/2.72.3/bin/gobject-query
/opt/homebrew/Cellar/glib/2.72.3/bin/gresource
/opt/homebrew/Cellar/glib/2.72.3/bin/gsettings
/opt/homebrew/Cellar/glib/2.72.3/bin/gtester
/opt/homebrew/Cellar/glib/2.72.3/etc/bash_completion.d/ (5 files)
/opt/homebrew/Cellar/glib/2.72.3/include/gio-unix-2.0/ (6 files)
/opt/homebrew/Cellar/glib/2.72.3/include/glib-2.0/ (269 files)
/opt/homebrew/Cellar/glib/2.72.3/lib/libgio-2.0.0.dylib
/opt/homebrew/Cellar/glib/2.72.3/lib/libglib-2.0.0.dylib
/opt/homebrew/Cellar/glib/2.72.3/lib/libgmodule-2.0.0.dylib
/opt/homebrew/Cellar/glib/2.72.3/lib/libgobject-2.0.0.dylib
/opt/homebrew/Cellar/glib/2.72.3/lib/libgthread-2.0.0.dylib
/opt/homebrew/Cellar/glib/2.72.3/lib/glib-2.0/ (2 files)
/opt/homebrew/Cellar/glib/2.72.3/lib/pkgconfig/ (8 files)
/opt/homebrew/Cellar/glib/2.72.3/lib/ (10 other files)
/opt/homebrew/Cellar/glib/2.72.3/share/aclocal/ (3 files)
/opt/homebrew/Cellar/glib/2.72.3/share/bash-completion/ (5 files)
/opt/homebrew/Cellar/glib/2.72.3/share/gettext/ (2 files)
/opt/homebrew/Cellar/glib/2.72.3/share/glib-2.0/ (3 files)
/opt/homebrew/Cellar/glib/2.72.3/share/locale/ (98 files)
```

EmmanuelP, 2023-01-28:
glib-mkenums is still present in glib 2.75.2. Please open an homebrew bug.

I'm closing this issue as there is not much we can do in Aravis.

Thanks for the report.

eli-schwartz, 2023-01-29:
For a short month, homebrew split the python scripts such as `glib-mkenums` into a separate bottle called `glib-utils`, before restoring sanity.

This was wonky in 2.72.2_1 and 2.72.3, but fixed in 2.72.3_1, so from the look of things you managed to get stuck on basically the only version of homebrew glib that is broken. Did you try pulling updates? Otherwise, just install glib-utils.

BZandi, 2023-01-29:
Thank you very much! The build process works now. For the sake of completeness, I updated brew as suggested and built aravis using

```shell
brew update
brew upgrade
meson --buildtype=plain -Ddocumentation=disabled -Dgst-plugin=enabled -Dintrospection=disabled -Dusb=enabled -Dviewer=enabled . ./build
ninja -C ./build
```

However, when starting the aravis viewer, there is the following output:

```shell
Gstreamer plugin 'appsrc' is missing.
Gstreamer plugin 'videoconvert' is missing.
Gstreamer plugin 'videoflip' is missing.
Missing GStreamer video output plugin (autovideosink, gtksink or gtkglsink)
Check your gstreamer installation.
```

Any suggestions?

EmmanuelP, 2023-01-29:
You have to install the gstreamer packages:

https://aravisproject.github.io/aravis/building.html#building-on-macos

BZandi, 2023-02-07:
The ``gstreamer`` packages are installed, and I followed the building instructions on the Aravis webpage, but unfortunately, the Aravis viewer does not start as intended.

```shell
Gstreamer plugin 'appsrc' is missing.
Gstreamer plugin 'videoconvert' is missing.
Gstreamer plugin 'videoflip' is missing.
Missing GStreamer video output plugin (autovideosink, gtksink or gtkglsink)
Check your gstreamer installation.
```

EmmanuelP, 2023-02-07:
What gives `gst-inspect-1.0 videoflip` ?

BZandi, 2023-02-07:
```shell
Factory Details:
  Rank                     none (0)
  Long-name                Video flipper
  Klass                    Filter/Effect/Video
  Description              Flips and rotates video
  Author                   David Schleef <ds@schleef.org>

Plugin Details:
  Name                     videofilter
  Description              Video filters plugin
  Filename                 /opt/homebrew/lib/gstreamer-1.0/libgstvideofilter.dylib
  Version                  1.20.5
  License                  LGPL
  Source module            gst-plugins-good
  Source release date      2022-12-19
  Binary package           GStreamer Good Plug-ins source release
  Origin URL               Unknown package origin

GObject
 +----GInitiallyUnowned
       +----GstObject
             +----GstElement
Pads:
  SINK: 'sink'
    Pad Template: 'sink'
  SRC: 'src'
    Pad Template: 'src'

Element Properties:
  method              : method (deprecated, use video-direction instead)
                        flags: lesbar, schreibbar, regelbar, änderbar in den Status NULL, BEREIT, PAUSIERT oder ABSPIELEN
                        Enum "GstVideoFlipMethod" Default: 0, "none"
                           (0): none             - Identity (no rotation)
                           (1): clockwise        - Rotate clockwise 90 degrees
                           (2): rotate-180       - Rotate 180 degrees
                           (3): counterclockwise - Rotate counter-clockwise 90 degrees
                           (4): horizontal-flip  - Flip horizontally
                           (5): vertical-flip    - Flip vertically
                           (6): upper-left-diagonal - Flip across upper left/lower right diagonal
                           (7): upper-right-diagonal - Flip across upper right/lower left diagonal
                           (8): automatic        - Select flip method based on image-orientation tag
  name                : The name of the object
                        flags: lesbar, schreibbar, 0x2000
                        String. Default: "videoflip0"
  parent              : The parent of the object
                        flags: lesbar, schreibbar, 0x2000
                        Object of type "GstObject"
  qos                 : Handle Quality-of-Service events
                        flags: lesbar, schreibbar
                        Boolean. Default: true
  video-direction     : Video direction: rotation and flipping
                        flags: lesbar, schreibbar, regelbar, änderbar in den Status NULL, BEREIT, PAUSIERT oder ABSPIELEN
                        Enum "GstVideoOrientationMethod" Default: 0, "identity"
                           (0): identity         - GST_VIDEO_ORIENTATION_IDENTITY
                           (1): 90r              - GST_VIDEO_ORIENTATION_90R
                           (2): 180              - GST_VIDEO_ORIENTATION_180
                           (3): 90l              - GST_VIDEO_ORIENTATION_90L
                           (4): horiz            - GST_VIDEO_ORIENTATION_HORIZ
                           (5): vert             - GST_VIDEO_ORIENTATION_VERT
                           (6): ul-lr            - GST_VIDEO_ORIENTATION_UL_LR
                           (7): ur-ll            - GST_VIDEO_ORIENTATION_UR_LL
                           (8): auto             - GST_VIDEO_ORIENTATION_AUTO
                           (9): custom           - GST_VIDEO_ORIENTATION_CUSTOM
```

EmmanuelP, 2023-02-07:
And `otool -l viewer/arv-viewer-0.8` ?

BZandi, 2023-02-07:
```shell
Load command 0
      cmd LC_SEGMENT_64
  cmdsize 72
  segname __PAGEZERO
   vmaddr 0x0000000000000000
   vmsize 0x0000000100000000
  fileoff 0
 filesize 0
  maxprot 0x00000000
 initprot 0x00000000
   nsects 0
    flags 0x0
Load command 1
      cmd LC_SEGMENT_64
  cmdsize 552
  segname __TEXT
   vmaddr 0x0000000100000000
   vmsize 0x0000000000018000
  fileoff 0
 filesize 98304
  maxprot 0x00000005
 initprot 0x00000005
   nsects 6
    flags 0x0
Section
  sectname __text
   segname __TEXT
      addr 0x00000001000054b4
      size 0x0000000000006918
    offset 21684
     align 2^2 (4)
    reloff 0
    nreloc 0
     flags 0x80000400
 reserved1 0
 reserved2 0
Section
  sectname __stubs
   segname __TEXT
      addr 0x000000010000bdcc
      size 0x0000000000000d74
    offset 48588
     align 2^2 (4)
    reloff 0
    nreloc 0
     flags 0x80000408
 reserved1 0 (index into indirect symbol table)
 reserved2 12 (size of stubs)
Section
  sectname __init_offsets
   segname __TEXT
      addr 0x000000010000cb40
      size 0x0000000000000008
    offset 52032
     align 2^2 (4)
    reloff 0
    nreloc 0
     flags 0x00000016
 reserved1 0
 reserved2 0
Section
  sectname __const
   segname __TEXT
      addr 0x000000010000cb48
      size 0x000000000000a878
    offset 52040
     align 2^3 (8)
    reloff 0
    nreloc 0
[... 299 lines cut ...]
         path /opt/homebrew/Cellar/gstreamer/1.20.5/lib (offset 12)
Load command 37
          cmd LC_RPATH
      cmdsize 56
         path /opt/homebrew/Cellar/gstreamer/1.20.5/lib (offset 12)
Load command 38
          cmd LC_RPATH
      cmdsize 56
         path /opt/homebrew/Cellar/gstreamer/1.20.5/lib (offset 12)
Load command 39
      cmd LC_FUNCTION_STARTS
  cmdsize 16
  dataoff 139560
 datasize 136
Load command 40
      cmd LC_DATA_IN_CODE
  cmdsize 16
  dataoff 139696
 datasize 0
Load command 41
      cmd LC_CODE_SIGNATURE
  cmdsize 16
  dataoff 158096
 datasize 1371
```

EmmanuelP, 2023-02-08:
I fail to understand how a GStreamer based application is supposed to run on macOS, at least when GStreamer comes from homebrew. For some reason the default plugin search path is not correct. `gst-inspect-1.0` and `gst-launch-1.0` are shell scripts that set `GST_PLUGIN_PATH` to `/usr/local/lib/gstreamer-1.0`. The problem is I don't know how to find this path during compile time.

So for now, the workaround is to use the `gst-plugin-path` option of arv-viewer:

```
arv-viewer-0.8 --gst-plugin-path=/usr/usr/local/lib/gstreamer-1.0
```

EmmanuelP, 2023-02-08:
Alternatively, you can set `GST_PLUGIN_PATH` in your shell configuration file.

BZandi, 2023-02-08:
Thank you very much! ``arv-viewer-0.8 --gst-plugin-path=/opt/homebrew/lib/gstreamer-1.0/``worked for me.