AravisProject/aravis #759: Problem popping buffer in windows

gotnone, 2023-01-29:
**Describe the bug**
arv-acquisiton-test fails to pop a buffer on several windows machines.  This behavior can be generalized to any of the arv_stream_*_pop_buffer functions.  Strangely, it works on another windows machine.  All machines are running Windows 10, but there may be differences in which updates are installed.  

**To Reproduce**
Build Aravis from source on windows.  Then ensure that the correct dlls are in the same directory with arv-acquisition-test.  Run arv-acquisition-test or any Aravis program that attempts to pop a buffer.

**Expected behavior**
Expect the buffer to be popped and control returned if using the blocking arv_stream_pop_buffer.

**Camera description:**
 - Manufacturer: Keyence
 - Model: VJ-3002
 - Interface: Ethernet

**Platform description:**
 - Aravis version: 0.8.23 git, tested versions include 7a18ea6, a994c5f, and  5e80a79.
 - OS:  Windows 10
 - Hardware: x86_64

**Additional context**
The camera works consistently with a Linux test machine for all of the mentioned library versions.  It also works well with one of the Windows computers that I tested with.  I have used the [DepenedenciesGui](https://github.com/lucasg/Dependencies) and [ProcessExplorer](https://learn.microsoft.com/en-us/sysinternals/downloads/process-explorer) to verify that the correct dlls are being found by the test program.

On the machines where the buffer fails to transmit, there doesn't seem to be any obvious indication of an error.  The camera does correctly trigger on the AcquisitionStart command and will activate the flash if configured to do so.  If the camera is configured for external trigger, the AcquisitonStart command appears to properly prepare the camera and a rising edge on the trigger line causes the flash to activate.  However, instead of transmitting the buffer, the library and camera remain in a perpetual Heartbeat Ack loop.

Working and broken machines have had any Mingw64 / Msys2 Aravis libraries removed.  Only the built-from-source Aravis libraries are present on the machines.  The problem is present when building with gcc (Mingw64 and the Mingw64 UCRT versions), as well as the Clang 15.1 installed with the Visual Studio Build Tools.

All Windows machines will work with vendor supplied library.

Typical debug output for non-working Windows machine.
````
[18:50:46.798] ≡ƒà╕ stream> [GvStream::stream_new] Destination stream port = 61934
[18:50:46.798] ≡ƒà╕ stream> [GvStream::stream_new] Source stream port = 61934
[18:50:46.798] ≡ƒà╕ stream> [GvStream::loop] Standard socket method
[18:50:46.798] ≡ƒà│ genicam> [GcRegisterNode::_get_integer_value] address = 0xa090, valu
e = 0x57e400
[18:50:46.798] ≡ƒà│ genicam> [GcEnumeration::set_string_value] value = 0 - string = Sing
leFrame
[18:50:46.798] ≡ƒà│ genicam> [GcRegisterNode::_set_integer_value] address = 0xa200, valu
e = 0x0
[18:50:46.813] ≡ƒà│ genicam> [GcRegisterNode::_set_integer_value] address = 0xa204, valu
e = 0x1
[18:50:46.813] ≡ƒà│ genicam> [GcCommand::execute] AcquisitionStart (0x1)
[18:50:47.657] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[18:50:48.672] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[18:50:49.687] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[18:50:50.703] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[18:50:51.716] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[18:50:52.726] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
````

The working Windows machine prints the following
````
[21:10:48.018] 🅸 stream> [GvStream::stream_new] Destination stream port = 61927
[21:10:48.018] 🅸 stream> [GvStream::stream_new] Source stream port = 61927
[21:10:48.019] 🅸 stream> [GvStream::loop] Standard socket method
[21:10:48.024] 🅳 genicam> [GcRegisterNode::_get_integer_value] address = 0xa090, value = 0x1d4c00
[21:10:48.024] 🅳 genicam> [GcEnumeration::set_string_value] value = 0 - string = SingleFrame
[21:10:48.024] 🅳 genicam> [GcRegisterNode::_set_integer_value] address = 0xa200, value = 0x0
[21:10:48.029] 🅳 genicam> [GcRegisterNode::_set_integer_value] address = 0xa204, value = 0x1
[21:10:48.031] 🅳 genicam> [GcCommand::execute] AcquisitionStart (0x1)
[21:10:48.152] 🅳 stream-thread> [GvStream::find_frame_data] Start frame 1
[21:10:48.152] 🅳 misc> [PixelFormat::to_gst_caps_string] 0x01080001 -> video/x-raw, format=(string)G
RAY8
[21:10:48.152] 🅳 sp> packet_type  =       ok (0x0000)
                     content_type = data-leader (0x0001)
                     frame_id     =        1
                     packet_id    =        0
                     data_size    =       36
                     payload_type = image
                     pixel format = video/x-raw, format=(string)GRAY8
                     width        = 1600
                     height       = 1200
                     x_offset     = 0
                     y_offset     = 0
````

EmmanuelP, 2023-01-29:
Hi,

Could you attach the packet statistics at the end of the log ?

gotnone, 2023-01-30:
Here is the result from the working windows computer.  It is an older laptop and the network card has seen better days.

````
[21:10:49.201] 🅸 stream>     bins    ;frame_retent; packet_time;inter_packet
                                    0;           0;          60;        3784
                                 2000;           0;          40;          64
                                 4000;           0;          60;          32
                                 6000;           0;          60;          14
                                 8000;           0;          60;          11
                                10000;           0;          60;           5
                                12000;           0;          32;           2
                                14000;           0;          48;           3
                                16000;           0;          48;           1
                                18000;           0;          32;           0
                                20000;           0;          48;           2
                                22000;           0;          32;           1
                                24000;           0;          48;           0
                                26000;           0;          16;           3
                                28000;           0;          16;           1
                                30000;           0;          16;           0
                                32000;           0;          16;           0
                                34000;           0;          16;           0
                                36000;           0;           0;           0
                                38000;           0;          16;           2
                                40000;           0;           0;           0
                                42000;           0;          16;           0
                                44000;           0;           0;           0
                                46000;           0;           0;           0
                                48000;           0;           0;           0
                                50000;           0;          16;           0
                                52000;           0;           0;           0
                                54000;           0;          16;           0
                                56000;           0;           0;           0
                                58000;           0;           0;           0
                                60000;           0;          16;           0
                                62000;           0;           0;           0
                                64000;           0;          16;           0
                                66000;           0;           0;           0
                                68000;           0;           0;           0
                                70000;           0;           0;           0
                                72000;           0;          16;           0
                                74000;           0;           0;           0
                                76000;           0;           0;           0
                                78000;           0;           0;           0
                                80000;           0;          16;           0
                                82000;           0;           0;           0
                                84000;           0;          16;           0
                                86000;           0;           0;           0
                                88000;           0;           0;           0
                                90000;           0;           0;           0
                                92000;           0;           0;           0
                                94000;           0;           0;           0
                                96000;           0;           0;           0
                                98000;           0;           0;           0
                               100000;           0;           0;           0
                               102000;           0;           0;           0
                               104000;           0;           0;           0
                               106000;           0;           0;           0
                               108000;           0;           0;           0
                               110000;           0;           0;           0
                               112000;           0;          16;           0
                               114000;           0;           0;           0
                               116000;           0;           0;           0
                               118000;           0;           0;           0
                               120000;           0;           0;           0
                               122000;           0;           0;           0
                               124000;           0;           0;           0
                               126000;           0;           0;           0
                               128000;           0;           0;           0
                               130000;           0;           0;           0
[... 44 lines cut ...]
[21:10:49.203] 🅸 stream> [GvStream::finalize] n_received_packets     = 3926
[21:10:49.203] 🅸 stream> [GvStream::finalize] n_missing_packets      = 0
[... cut ...]
I can not capture any statistics from the broken windows machines because arv-acquisition-test.exe uses a blocking call to pop the buffer.  The program just sits in a loop printing the heartbeat ack line.

````
[18:50:52.726] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
````

If i try to interrupt the program when it is stuck in this state, no statistics are printed.

I will try to modify the program to use a timeout version of pop buffer to see if there are any packet statistics printed in that case.

gotnone, 2023-01-30:
Here is a log from a non-working windows machine.  I modified the timeout in arv_camera_acquisition to 100 seconds (100 * 1000000).  Trigger seems to have been configured correctly, and camera flash operated, but no buffer is transmitted before the timeout expired.
````
[20:20:36.078] ≡ƒà│ genicam> [GcCommand::execute] AcquisitionStart (0x1)
[20:20:36.922] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:37.937] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:38.953] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:39.968] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:40.983] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:41.998] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:43.014] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:44.029] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:45.044] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:46.060] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:47.075] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:48.090] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:49.105] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:50.121] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:51.136] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:52.142] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:53.158] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:54.173] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:55.188] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:56.204] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:57.219] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:58.235] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:20:59.250] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:00.265] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:01.280] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:02.295] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:03.311] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:04.326] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:05.342] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:06.357] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:07.372] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:08.387] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:09.403] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:10.418] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:11.433] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:12.449] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:13.464] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:14.479] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:15.495] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:16.510] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:17.525] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:18.540] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:19.556] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:20.571] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:21.586] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:22.602] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:23.612] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:24.628] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:25.643] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:26.658] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:27.674] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:28.689] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:29.704] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:30.719] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:31.735] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:32.750] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:33.765] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:34.780] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:35.796] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:36.811] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:37.827] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:38.842] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:39.857] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:40.872] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:41.888] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:42.903] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[20:21:43.918] ≡ƒà│ device> [GvDevice::Heartbeat] Ack value = 2
[... 40 lines cut ...]
                         max         :         n/a;         n/a;         n/a
                         last max at :           0;           0;           0
                         counter     :           0:           0:           0
[20:22:16.080] ≡ƒà╕ stream> [GvStream::finalize] n_completed_buffers    = 0
[20:22:16.080] ≡ƒà╕ stream> [GvStream::finalize] n_failures             = 0
[20:22:16.080] ≡ƒà╕ stream> [GvStream::finalize] n_underruns            = 0
[20:22:16.080] ≡ƒà╕ stream> [GvStream::finalize] n_timeouts             = 0
[20:22:16.096] ≡ƒà╕ stream> [GvStream::finalize] n_aborted              = 0
[20:22:16.096] ≡ƒà╕ stream> [GvStream::finalize] n_missing_frames       = 0
[20:22:16.096] ≡ƒà╕ stream> [GvStream::finalize] n_size_mismatch_errors = 0
[20:22:16.096] ≡ƒà╕ stream> [GvStream::finalize] n_received_packets     = 0
[... cut ...]
[20:22:16.096] ≡ƒà╕ stream> [GvStream::finalize] n_resend_disabled      = 0
[20:22:16.096] ≡ƒà╕ stream> [GvStream::finalize] n_duplicated_packets   = 0
[20:22:16.111] ≡ƒà╕ stream> [GvStream::finalize] n_transferred_bytes    = 0
[20:22:16.111] ≡ƒà╕ stream> [GvStream::finalize] n_ignored_bytes        = 0
[20:22:16.111] ≡ƒà╕ stream> [Stream::finalize] Flush 1 buffer[s] in input queue
[20:22:16.111] ≡ƒà╕ stream> [Stream::finalize] Flush 0 buffer[s] in output queue
Failed to acquire a single image
````

EmmanuelP, 2023-01-30:
So on the broken setup you never receive any stream packets. The next step consists in checking if the camera is actually sending packets to aravis, using wireshark for example.

Did you check the firewall ? Is the camera directly connected to the machine ?

gotnone, 2023-01-30:
Firewall appears to be correctly configured and problem still exists when firewall is disabled.

Camera is directly connected to machine being tested.

I have wireshark captures, please let me know what to look for, or where I can send them for review.

It appears that there are two long transmissions, one for the XML at the beginning, with appropriate READMEM_ACKs from the computer, and another long transmission of "UDP" packets without any ACKs from the computer, which may be the buffer.  On the working windows machine, there are occasional PACKETRESEND_CMD packets from the computer to the camera interrupting this transmission.  These are absent on the broken machine.

EmmanuelP, 2023-02-06:
> I have wireshark captures, please let me know what to look for, or where I can send them for review.
>
> It appears that there are two long transmissions, one for the XML at the beginning, with [...]

Please attach a wireshark capture using arv-camera-test-0.8, taking care to minimize the capture size by setting the image size to the minimum possible, and setting the framerate to 1Hz and the test duration to 2s.

gotnone, 2023-02-07:
This packet dump is using aravis 0.8.25 commit cd6ef301caedeb132d7e18e27a3dc2914b88d62e

I only have access to the camera on a non-working machine.  The buffer is still not properly transmitted in the newest commit.

Unfortunately, the camera does not support a smaller native image size than 1600x1200, so apologies for the large capture size.

```
$ ./arv-camera-test-0.8.exe -o 1 --duration 2
Looking for the first available camera
vendor name            = KEYENCE CORPORATION
model name             = CA-H200MX
device serial number   = 1A910122
image width            = 1600
image height           = 1200
gain                   = 3 dB
payload                = 1920000 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 0 ns
gv packet size         = 1060 bytes
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
n_completed_buffers    = 0
n_failures             = 0
n_underruns            = 0
n_timeouts             = 0
n_aborted              = 0
n_missing_frames       = 0
n_size_mismatch_errors = 0
n_received_packets     = 0
n_missing_packets      = 0
n_error_packets        = 0
n_ignored_packets      = 0
n_resend_requests      = 0
n_resent_packets       = 0
n_resend_ratio_reached = 0
n_resend_disabled      = 0
n_duplicated_packets   = 0
n_transferred_bytes    = 0
n_ignored_bytes        = 0
```

[arv_camera_test_dump.zip](https://github.com/AravisProject/aravis/files/10671713/arv_camera_test_dump.zip)

EmmanuelP, 2023-02-07:
So it looks like for some reason the g_poll ( https://github.com/AravisProject/aravis/blob/cd6ef301caedeb132d7e18e27a3dc2914b88d62e/src/arvgvstream.c#L1027 ) call never sees the incoming packets. source/destination port and address seem correctly configured.

@eudoxos , may be you have an idea on what is going wrong ?

EmmanuelP, 2023-02-07:
Something to check: do you have a GigEVision driver installed on this machine ?

EmmanuelP, 2023-02-07:
Another test: can you replicate the issue using `arv-fake-gv-camera-0.8.exe` ?

gotnone, 2023-02-07:
Yes, "eBUS Universal Pro For Ethernet Driver"

This same filter driver is installed on the working windows machine.

I tried uninstalling and rebooting, but the transferred buffer still fails to be detected by Aravis.

gotnone, 2023-02-07:
Using arv-fake-gv-camera-0.8.exe on the loopback device appears to work. 

```
$ ./arv-camera-test-0.8.exe -o 1 --duration 2
Looking for the first available camera
vendor name            = Aravis
model name             = Fake
device serial number   = GV01
image width            = 512
image height           = 512
horizontal binning     = 1
vertical binning       = 1
exposure               = 10000 ┬╡s
gain                   = 0 dB
payload                = 262144 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 0 ns
gv packet size         = 1400 bytes
  1 frame/s  -   0.262 MiB/s
  1 frame/s  -   0.262 MiB/s
n_completed_buffers    = 2
n_failures             = 0
n_underruns            = 0
n_timeouts             = 0
n_aborted              = 0
n_missing_frames       = 0
n_size_mismatch_errors = 0
n_received_packets     = 390
n_missing_packets      = 0
n_error_packets        = 0
n_ignored_packets      = 0
n_resend_requests      = 0
n_resent_packets       = 0
n_resend_ratio_reached = 0
n_resend_disabled      = 0
n_duplicated_packets   = 0
n_transferred_bytes    = 527496
n_ignored_bytes        = 0
```

[loopback_dump.zip](https://github.com/AravisProject/aravis/files/10672388/loopback_dump.zip)

KNSd2, 2023-09-14:
Did you find the issue? I think I am experiencing the same problems. Connecting to the camera (ethernet) and setting the parameters works fine. However, the buffer stays empty.

gotnone, 2023-09-15:
Unfortunately, no.  I had to use the vendor library because I could not find the cause of the buffer transfer issue.  The strangest part was the fact that it did work on at least one Windows machine, despite failing to transfer on a couple of others.  I couldn't narrow down any difference between firewalls, network settings, or shared libraries that might be the cause.

At the present time I don't have access to the camera for further testing as it is on a working line performing captures for inspection.

gotnone, 2023-09-15:
Try capturing the communication with wireshark.  It would be interesting to see if the buffer is actually transferred and then g_poll is not picking up the incoming packet stream.  It seemed like that was the issue identified in my captures.

KNSd2, 2023-09-18:
Thanks for the info. In my case, it was a firewall issue and an old ebus driver installed on the test system.

venusspace, 2024-05-08:
Did you find out what the problem is? I'm experiencing the same problem, the camera is connected and I can set the parameters, but no buffer available.

Fran-ATRIA, 2024-11-05:
Hi @EmmanuelP @KNSd2 @venusspace @gotnone did you ever find a solution for this issue?

gotnone, 2024-11-05:
@Fran-ATRIA,

Not that I know of.  We were starting to investigate the wireshark dumps, and seem to have isolated the problem to an issue with the g_poll ( function.  Specifically, the g_poll function was not ever seeing the buffer packets from the camera.  It appeared that the packets were sent to the correct IP address and port, but somehow they were being lost between the network card and g_poll.  

Unfortunately without access to the camera hardware I can not investigate further.

If I recall during my investigation I found some discussion of certain windows updates causing issues with jumbo frame reception, but that may be a red herring.

Another potential issue is a conflicting filter driver may be intercepting / blocking the packets.  I had tried uninstalling the  `eBUS Universal Pro For Ethernet Driver`, but there is a chance that there was something lingering from that which was causing my problem.

Do you have any filter drivers installed in your non-working setup?

Fran-ATRIA, 2024-11-12:
@gotnone,

As far as I know, no, I have no filter drivers. At least none that I've installed or that I can find on my system. Another developer told me [here](https://github.com/AravisProject/aravis/issues/956#issuecomment-2468628210) that this is a known issue. May it be related to what you said?

VisionShilin, 2025-02-19:
Did you find a solution for this issue?
I met the same problem too.

Fran-ATRIA, 2025-02-20:
@VisionShilin Sadly, no. I made it work including this line to my Dockerfile:
 
`RUN ln -s /usr/local/lib/x86_64-linux-gnu/girepository-1.0/Aravis-0.8.typelib /usr/lib/x86_64-linux-gnu/girepository-1.0/Aravis-0.8.typelib
`
But then I delivered the code and camera to some colleagues, they probably changed something in the camera configuration and it never worked again, neither did I discover why it's not working.

It seems like it's not a Windows problem, because it works when I run it on Ubuntu but it gives the same error on other Linux systems. I don't have the time or knowledge to get deeper into this, so I got stuck here.

Sorry if that's not helpful.

Fran-ATRIA, 2025-02-20:
@VisionShilin Funnily enough, I had the chance to work on this today again. For other Linux systems, it seems like the issue had to do with the network permissions. Launching my Docker with the `--net=host` flag solved the issue.

It's weird because I was able to ping the camera and connect to it without this, but it seems like something was blocking packages on their way back to the computer. On Windows it still does not work, but it may be a driver blocking them or some firewall process, although I disabled all firewalls and it still didn't work.

Hope this helps :)

VisionShilin, 2025-02-20:
@Fran-ATRIA Thanks. It works fine after I disable the firewall or add the program into the exception list. The firewall of Windows did cause the problem I think.

eudoxos, 2025-02-21:
> Makes sense to think that the same is happening somehow in Windows, right?

With containers, this is a [long-standing issue](https://github.com/AravisProject/aravis/issues/232), see also #707 .

When running without containerization (as @VisionShilin noted), disabling the firewall will help (see #709 for an option to limit port numbers so that not everything needs to be opened — did not test that, though). Firewalls don't understand GigEVision protocol so don't know the inbound traffic is related and will normally block it.

TIS-Edgar, 2025-02-21:
If I understand this correctly you need port knocking, right?
#375 at least documents this. Basically aravis needs to "knock" the correct stream port  so that inbound packets are passed through.

eudoxos, 2025-02-24:
> If I understand this correctly you need port knocking, right?

I don't see how that could possibly work: how could Aravis knock on the NAT port on the other side it has no access to (the camera does)? Reading the discussion linked in #375, it seems to be just a vague idea (which IMHO will not work for technical reasons) rather than something implemented and functional — but I might be overlooking something. This scenario is the very reason that dual-channel protocols (like FTP with control and data channel) had to have layer-4 firewalls.