AravisProject/aravis #761: Set USB response timeout

hgaiser, 2023-02-03:
**Is your feature request related to a problem? Please describe.**

I am using a USB3Vision camera in a special mode, where it acts as a linescanner, aimed at a rotating object. The camera buffers every line, creating an image. After a full rotation this buffer is sent using the USB3Vision protocol. After the first few successfully received frames I am getting a `LIBUSB_ERROR_TIMEOUT`.

The camera reports that its `ARV_ABRM_MAX_DEVICE_RESPONSE_TIME` is 300ms, but that seems to be too tight for the packets. Attached is a Wireshark capture when the timeout occurs:

[libusb.pcapng.gz](https://github.com/AravisProject/aravis/files/10580465/libusb.pcapng.gz)

**Describe the solution you'd like**
I would like a function to increase the timeout manually to deal with camera's reporting too tight timeouts.

**Describe alternatives you've considered**
Writing to the register of the camera to change its timeout value, but the register is write protected.

hgaiser, 2023-02-09:
It might not be the timeout that is to blame, but the size of the packets being sent. I took the following steps:

1. I modified [this](https://github.com/AravisProject/aravis/blob/main/src/arvuvcpprivate.h#L37) line to set the timeout to 5000msec, to avoid triggering timeout errors.
2. I captured a single frame using Aravis:
   [aravis.pcapng.gz](https://github.com/AravisProject/aravis/files/10699167/aravis.pcapng.gz)
4. I captured a single frame using ids-peak, the viewer provided by the manufacturer IDS:
   [ids_peak.pcapng.gz](https://github.com/AravisProject/aravis/files/10699053/ids_peak.pcapng.gz)

Comparing these two captures I noticed the following:

1. Using Aravis there's 6 `Stream Payload` frames being sent with the following sizes: `1048576 * 5 + 665792 = 5908672` bytes. Using ids-peak I see 11 `Stream Payload` frames being sent: `524288 * 11 + 141504 = 5908672`. The same number of bytes are being transmitted, however using ids-peak they are sent over more frames.
2. The time between the `Stream Payload` frames for Aravis is approximately 530msec, except for the last frame which is smaller and is sent in approximately 330msec. With ids-peak the time between frames is approximately 270msec.
3. Interestingly ids-peak seems to send all the requests for the frames in one go (as can be seen in frames nr 1133-1152 in the ids-peak capture). This means that the last frame, the `Stream Trailer`, reports `[Time from request: 3.083197000 seconds]`. I'm not sure why this is the case? Perhaps they're doing something weird with setting the timeout to `nth_request * 300msec` instead of requesting a frame and waiting for the response before requesting another? Not sure if this is relevant, but thought it's worth noting.

My conclusion is that the (negotiated?) frame size in Aravis is too large for the timeout as reported by the camera. If the frame size is reduced, the timeout wouldn't trigger.

Next step for me is to figure out how the frame size is determined and to see if that can be changed. Figured I'd update this issue as I go in case there is some input :).

EmmanuelP, 2023-02-09:
> 3\. Interestingly ids-peak seems to send all the requests for the frames in one go (as can be seen in frames nr 1133-1152 in the ids-peak capture). This means that the last frame, the `Stream [...]

In aravis there is 2 USB modes, synchronous and asynchronous, synchronous being the default. ids-peak probably uses asynchronous requests by default. You can select asynchronous USB requests in arv-camera-test-0.8 with ` --usb-mode=async` option.

hgaiser, 2023-02-10:
> In aravis there is 2 USB modes, synchronous and asynchronous, synchronous being the default. ids-peak probably uses asynchronous requests by default. You can select asynchronous USB requests in arv-camera-test-0.8 with ` --usb-mode=async` option.

Interesting, with `--usb-mode=async` it indeed works, but should it? Shouldn't the max device response time (which is reported as 300msec in my situation) still be respected? I see in Aravis that there is a [timeout of 0 being selected](https://github.com/AravisProject/aravis/blob/main/src/arvuvstream.c#L317) (which means no timeout) when it's using async mode. In my case I would expect a timeout to be triggered if a payload is received after 300msec of the previous payload. According to Wireshark I am receiving a payload roughly every 530msec.

It seems to me that there's two things that need to be done:

1. Allow changing the device response time and the payload size through the Aravis API.
2. Use the max device response time to trigger timeout errors in asynchronous mode.

I'm curious to hear your thoughts on this @EmmanuelP . I'm fine with using asynchronous mode for now, but it doesn't feel like the "correct" solution.

EmmanuelP, 2023-02-10:
I'm not sure the maximum response time given by the device applies to the streaming. It is may be an error in aravis synchronous mode to use this value as a timeout. If I'm not mistaken, maximum response time is the maximum time the device will take to answer to a control packet, and will issue a pending ack packet if the response takes longer.

So the right fix would be to use an infinite timeout in the arv_uv_device_bulk_transfer function call in synchronous mode, while still calling libusb_bulk_transfer with a non zero timeout multiple times until the transfer is done or an error occur. That will allow us to cancel the streaming.

In asynchronous mode, using 0 as the timeout value is not an issue, as the call to libusb being asynchronous, we are still able to cancel the streaming.

hgaiser, 2023-02-10:
> I'm not sure the maximum response time given by the device applies to the streaming. It is may be an error in aravis synchronous mode to use this value as a timeout. If I'm not mistaken, maximum [...]

Ahh interesting, hadn't considered that. I'm going to see if I can slow down the data transfer to see if I get timeout errors from ids-peak. If they keep the same payload size, then it should trigger a timeout (assuming they use the max device response time as timeout for payload frames).

> So the right fix would be to use an infinite timeout in the arv_uv_device_bulk_transfer function call in synchronous mode, while still calling libusb_bulk_transfer with a non zero timeout multiple times until the transfer is done or an error occur. That will allow us to cancel the streaming.

How would the timeout be determined though? Would that be selected by the user?

EmmanuelP, 2023-02-17:
> How would the timeout be determined though? Would that be selected by the user?

That would be a non configurable small timeout of about 1s. I'm not completely sure yet it will work, I have to experiment a bit. Not having a test device does not help, I will have to add the possibility to reproduce the behaviour of your device using aravis fake camera.