AravisProject/aravis #762: Reset camera's frame id to 1 on start_acquisition() [flir/pointgrey gige cameras]

watsonmw, 2023-02-05:
If I do multiple cycles of:

- create stream
- start_acquisition() 
- end_acquisition() 
- free stream

Without freeing the camera, the frame id doesn't get reset to 1 with Aravis between cycles.  But it does get reset to 1 each cycle with the camera manufacturer's SDK. 

Changing some configuration values on the camera also cause this frame id to get bumped by varying amounts (e.g. trigger params or calling the software trigger).

This is a problem for me since I use the frame id to detect missing frames and match up frame across several cameras.  My application logs lead me to believe I was (rarely) getting spurious frames triggered & delivered as well (not just the id being bumped).

My workaround is that on end_acquisition() / free'ing the stream I've added the following, to cause the stream frame id to be reset. This resets the frame id AND stops extra frames coming in that I didn't trigger when I start the new acquisition.

      // Set the stream port to zero, this is needed for FLIR cameras as otherwise the frame id doesn't get reset to 1
      // on the next start_acquisition().
      arv_device_set_integer_feature_value(arv_camera_get_device(_camera), "GevSCPHostPort", 0, &error);
      if (error) {
          logerror("{}:\tError arv_device_set_integer_feature_value GevSCPHostPort: {}", _camId, error->message);
          g_clear_error(&error);
      }

watsonmw, 2023-02-05:
pcaps for aravis and spinnaker SDK:

[aravis_port_set0.tar.gz](https://github.com/AravisProject/aravis/files/10610051/aravis_port_set0.tar.gz)

EmmanuelP, 2023-02-05:
Hi Mark,

This is fixed since f8f2adc4a1105ff041215a2ca38fa9a71d7bb28c

closed 2023-02-05 by EmmanuelP

watsonmw, 2023-02-05:
Wow, that was fast :p 

Tried it out, it's working great now.