AravisProject/aravis #774: Aravis cannot automatically find the highest possible value for the packet size

JMaravalhasSilva, 2023-03-10:
**Describe the bug**
The function arv_camera_gv_set_packet_size_adjustment(), when called with ARV_GV_PACKET_SIZE_ADJUSTMENT_ONCE or with ARV_GV_PACKET_SIZE_ADJUSTMENT_ALWAYS does not find the highest possible value for the packet size at stream creation. arv_camera_gv_auto_packet_size() also has the same behavior.

**To Reproduce**
1) (Optional) Use arv_camera_gv_set_packet_size() to set the packet size to the minimum supported by the camera

2) Call arv_camera_gv_set_packet_size_adjustment() with the ARV_GV_PACKET_SIZE_ADJUSTMENT_ONCE or ARV_GV_PACKET_SIZE_ADJUSTMENT_ALWAYS options. Alternatively, call arv_camera_gv_auto_packet_size().

3) Create the stream

4) Call arv_camera_gv_get_packet_size(). With certain cameras, this will be some value lower than the maximum accepted by the camera.

**Expected behavior**
Using a specim camera which accepts a maximum value of 8228 for the GevSCPSPacketSize parameter, I expected arv_camera_gv_set_packet_size_adjustment() or arv_camera_gv_auto_packet_size() to set the packet size to 8228, but instead it sets it to 7204.

I expected these functions to read the SCPSMin, SCPSMax and SCPSInc parameters and try all possible values.

Note that if the packet size is previously set to 8228, both arv_camera_gv_set_packet_size_adjustment() and arv_camera_gv_auto_packet_size() leave it at 8228. Otherwise, with any other initial values, these functions always set it to 7204.

**Camera description:**
 - Specim
 - FX10e
 - Interface: ethernet

**Platform description:**
 - Aravis version 0.8.26
 - OS: Ubuntu 22.04 (jammy)
 - Hardware x86_64

**Additional context**
Relevant parameters of tested camera:
 - SCPSMin: 1060
 - SCPSMax: 8228
 - SCPSInc: 1024

EmmanuelP, 2023-03-10:
Hi,

Please attach the output of `ARV_DEBUG=all ./tests/arv-auto-packet-size-test <your-camera>`.

JMaravalhasSilva, 2023-03-10:
```
[14:54:33.171] 🅸 interface> Found 0 USB3Vision device (among 9 USB devices)
[14:54:33.171] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[14:54:33.171] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.10.1 (192.168.10.255)
[14:54:33.171] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.220.61 (192.168.255.255)
[14:54:33.177] 🅸 interface> [GvInterface::discovery] Device 'Specim-FX10e-077000038337' found (interface 192.168.10.1) user_id ' ' - MAC '70:f8:e7:b0:02:6e'
[14:54:33.177] 🅸 device> [GvDevice::new] Interface address = 192.168.10.1
[14:54:33.177] 🅸 device> [GvDevice::new] Device address = 192.168.10.3
[14:54:33.178] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:FX10e.zip;400000;bbe9' at 0x200
[14:54:33.178] 🅸 device> [GvDevice::load_genicam] Xml address = 0x400000 - size = 0xbbe9 - FX10e.zip
[14:54:33.732] 🅸 device> [GvDevice::load_genicam] Zipped xml data
[14:54:33.848] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationLLA'
[14:54:33.849] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationDHCP'
[14:54:33.849] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationPersistentIP'
[14:54:33.849] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPAddress'
[14:54:33.849] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentSubnetMask'
[14:54:33.849] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentDefaultGateway'
[14:54:33.849] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevNumberOfNetworkInterfaces'
[14:54:33.849] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentIPAddress'
[14:54:33.849] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentSubnetMask'
[14:54:33.849] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentDefaultGateway'
[14:54:33.849] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelCount'
[14:54:33.849] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelSelector'
[14:54:33.849] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPAddrCalc'
[14:54:33.849] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevTimestampTickFrequency'
[14:54:33.849] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPHostPort'
[14:54:33.850] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSFireTestPacket'
[14:54:33.850] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSDoNotFragment'
[14:54:33.850] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSBigEndian'
[14:54:33.850] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSPacketSize'
[14:54:33.850] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPD'
[14:54:33.850] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCDA'
[14:54:33.850] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCSP'
[14:54:33.850] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCCFGMultipart'
[14:54:33.851] 🅸 device> [GvDevice::new] Device endianness = big
[14:54:33.851] 🅸 device> [GvDevice::new] Packet resend     = yes
[14:54:33.851] 🅸 device> [GvDevice::new] Write memory      = yes
[14:54:33.851] 🅸 device> [GvDevice::new] Legacy endianness handling = no
[14:54:33.853] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 1060 (min: 1060 - max: 8228 - inc: 1024)
[14:54:33.854] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 4132 (min: 1060 - max: 8228 - inc: 1024)
[14:54:33.855] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 6180 (min: 4132 - max: 8228 - inc: 1024)
[14:54:33.856] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 7204 (min: 6180 - max: 8228 - inc: 1024)
[14:54:33.857] 🅸 device> [GvDevice::auto_packet_size] Packet size set to 7204 bytes
Packet size set to 7204 bytes on camera Specim-077000038337
```

commit 2ee47340 (2023-03-10): gvdevice: allow auto packet_size to be equal to max size

Fix #774
commit 4293ebc2 (2023-05-18): gvdevice: allow auto packet_size to be equal to max size

Fix #774
PR #775 (2023-05-23): gvdevice: allow auto packet_size to be equal to max size
Fix #774
commit 61c4af49 (2023-05-23): gvdevice: allow auto packet_size to be equal to max size

Fix #774
closed 2023-05-23 by EmmanuelP