AravisProject/aravis #778: AcquisitionFrameRateEnable write permission error.

osiler, 2023-03-27:
**Describe the bug**
Unable to rerun camera config without power cycle

**To Reproduce**

```C
#include <arv.h>
#include <stdio.h>
#include <sys/types.h>

int main(int argc, char **argv) {
  if (argc != 1) {
    printf("%s takes no arguments.\n", argv[0]);
    return 1;
  }

  ArvCamera *camera;
  GError *error = NULL;
  const char feature[] = "AcquisitionFrameRateEnable";

  camera = arv_camera_new(NULL, &error);

  if (ARV_IS_CAMERA(camera)) {

    if (arv_camera_is_feature_available(camera, feature, &error)) {
      printf("Feature exisits\n");
      if (error == NULL) {
        arv_camera_set_boolean(camera, feature, TRUE, &error);
        if (error == NULL) {
          printf("Can enable Aqusition Frame Rate\n");
        } else {
          printf("Error: %s\n", error->message);
          return EXIT_FAILURE;
        }
        arv_camera_set_boolean(camera, feature, FALSE, &error);
        if (error == NULL) {
          printf("Can disable Aqusition Frame Rate\n");
        } else {
          printf("Error: %s\n", error->message);
          return EXIT_FAILURE;
        }
        arv_camera_set_trigger(camera, "Software", &error);
        if (error == NULL) {
          printf("Can set software trigger\n");
        } else {
          printf("Error: %s\n", error->message);
          return EXIT_FAILURE;
        }
      }
    } else {
      printf("Feature is unavailable");
    }
  } else {
    printf("Error: %s\n", error->message);
    return EXIT_FAILURE;
  }
  arv_shutdown();
  return 0;
}
```

**Expected behavior**
Allow multiple reruns of camera setup, for use case such as unit testing.

**Platform description:**
 - Aravis version 0.8.22, 0.8.26 
 - OS: Ubuntu 20.04
 - Hardware x86

**Additional context**
This is a follow up from an [issue](https://github.com/robohouse-delft/aravis-rs/issues/13#issuecomment-1479784200) posted in the aravis rust crate bindings where running the provided [capture CLI](https://github.com/robohouse-delft/aravis-rs/blob/main/aravis-utils/src/bin/aravis-capture.rs) works once per power cycle. Subsequent runs without power cycling the camera leads to a AqusitionFrameRateEnable write error, I've reproduced this error in 0.8.22 and 0.8.26 with the above code snippet.

```bash
./config_test
Feature exisits
Can enable Aqusition Frame Rate
Can disable Aqusition Frame Rate
Can set software trigger
./config_test
Error: [AcquisitionFrameRateEnable] [RegAcquisitionFrameRateEnable] GigEVision write_memory error (access-denied)
```

osiler, 2023-05-19:
Apologies I will be looping back this weekend to provide more information.

osiler, 2023-06-04:
After trialing several different camera manufactures I found that it was indeed limiited to a specific manufacturer. What I had to do to was reset the camera after every acquisition event as the memory location for the the software trigger would become unavailable. I have been using the aravis-rs crate and ended up implementing something along the lines of.

``` rust
fn initialize(settings: CameraSettings) -> Result<Camera, BoxedError> {
        {
            let _c = Camera::new(Some(&settings.address))?;
            _c.execute_command(SNFCCommand::DeviceReset.as_str())?;
            drop(_c);
        }

        let camera: aravis::Camera = loop {
            match Camera::new(Some(&settings.address)) {
                Ok(camera) => {
                    break camera;
                }
                Err(_) => {
                    event!(Level::INFO, "Camera connecting after DeviceRest");
                    std::thread::sleep(Duration::from_secs(1));
                    continue;
                }
            };
        };
        
	camera.set_trigger("Software")?;
        camera.set_acquisition_mode(aravis::AcquisitionMode::Continuous)?;
        camera.set_pixel_format(aravis::PixelFormat::RGB_8_PACKED)?;
        Ok(camera)
        
```

EmmanuelP, 2023-06-04:
So probably not an Aravis issue. Could you attach the Genicam data of the faulty camera ?

osiler, 2023-06-04:
Unfortunately I do not have the camera in my possession (returned as we decided to go with another), the only other reason I could think of this happening is if it required similar additional logic as specified [here](https://github.com/AravisProject/aravis/pull/769).

EmmanuelP, 2023-06-05:
Ok. Thanks.

EmmanuelP, 2023-06-05:
If you could at least leave here the manufacturer and model name, it will help others finding this informations if they experience the same issue.

closed 2023-06-05 by EmmanuelP