AravisProject/aravis #787: U3V camera FPS are stuck at 30 while using python

WeaponXI, 2023-05-10:
**Issue faced**
I am using python APIs to open a camera and streaming it. Idea is to run multiple cameras (Max 3) simultaneously. While opening multiple cameras with Aravis Viewer, It is coming full 60 FPS. But with python it stuck at 30, even if I am streaming only single camera.

**To Reproduce**

import gi

gi.require_version ('Aravis', '0.8')
from gi.repository import Aravis

import numpy as np
import time

# Get U3V device names
Aravis.update_device_list()
n = Aravis.get_n_devices()
devices = [Aravis.get_device_id(i) for i in range(0, n)]
print(f"No of devices: {n}")
print(f"Device List: {devices}")

# Start any of the camera by default; or use from device list
# Aravis.enable_interface("Fake") # using arv-fake-gv-camera-0.6
camera = Aravis.Camera.new(None)

[x,y,width,height] = camera.get_region ()
print ("camera vendor : %s" %(camera.get_vendor_name ()))
print ("camera model  : %s" %(camera.get_model_name ()))
print ("ROI           : %dx%d at %d,%d" %(width, height, x, y))
print ("Pixel format str : %s" %(camera.get_pixel_format_as_string ()))
print ("frame rate  : %s" %(camera.get_frame_rate ()))

stream = camera.create_stream (None, None)
payload = camera.get_payload ()

# push 50 payload to buffer, 
for i in range(0,50):
    stream.push_buffer (Aravis.Buffer.new_allocate (payload))

print ("Start acquisition")
camera.start_acquisition()
pt = time.perf_counter()
fps = 0
counter = 0
while True:
    ct = time.perf_counter()
    buffer = stream.pop_buffer()
    if buffer:
        fps += 1
        addr = buffer.get_data()
        stream.push_buffer(buffer) #push buffer back into stream
        # img_data = np.ndarray(buffer=addr, dtype=np.uint8,shape=(height, width, 2))
        # some more code to display, I tested without display as well but getting constant 30 FPS.
    if ct - pt >= 1:
        print(f"FPS: {fps}, Running for [{counter}] sec")
        counter +=1
        fps = 0
        pt = ct
        if counter > 50:
            break
# delete stream and close camera
del stream
camera.stop_acquisition()

**Expected behavior**
Ideally it should give 60 FPS which is the FPS I am getting using Aravis Viewer

**Camera description:**
 - Manufacturer: Cypress
 - Model: NA
 - Interface : USB3

**Platform description:**
 - Aravis version: 0.8.26
 - OS: Ubuntu 20.04
 - Hardware: HP Omen Ryzen 9 6000 series (x86-64) & Nvidia Jetson AGX Orin (ARM64)

**Additional context**
I found that if I do modifiation similar to below (Push buffer before get_data):
while True:
    ct = time.perf_counter()
    buffer = stream.pop_buffer()
    if buffer:
        fps += 1
        stream.push_buffer(buffer) #push buffer back into stream
        addr = buffer.get_data()

Then I get more that 30 FPS but code crashed.

Also, I build the Aravis using MSYS2 in windows os as well (Dell XPS i7 12xx series (x86_64)) there also same issue. With Aravis viewer it is giving 60 FPS for all 3 cameras but stuck at 30 when use python code.

EmmanuelP, 2023-05-17:
At least on linux, the python overhead is relatively small. I'm able to stream 166 images/s (approx. 100MB/s) with your sample application:

```python
Sources/aravis/build on  main [$] via 🐍 v3.11.3 took 6s ✦ ❯ ARV_DEBUG=stream ./tests/pylaunch test.py 
No of devices: 1
Device List: ['Basler-2676014EA111-21930257']
camera vendor : Basler
camera model  : acA1920-155um
ROI           : 960x600 at 0,0
Pixel format str : Mono8
frame rate  : 100.0
[08:42:08.015] 🅸 stream> SIRM_INFO             = 0x02000000
[08:42:08.015] 🅸 stream> SIRM_REQ_PAYLOAD_SIZE = 0x000000000008ca00
[08:42:08.015] 🅸 stream> SIRM_REQ_LEADER_SIZE  = 0x00000400
[08:42:08.015] 🅸 stream> SIRM_REQ_TRAILER_SIZE = 0x00000400
[08:42:08.015] 🅸 stream> Required alignment    = 4
[08:42:08.017] 🅸 stream> SIRM_PAYLOAD_SIZE     = 0x0008ca00
[08:42:08.017] 🅸 stream> SIRM_PAYLOAD_COUNT    = 0x00000001
[08:42:08.017] 🅸 stream> SIRM_TRANSFER1_SIZE   = 0x00000000
[08:42:08.017] 🅸 stream> SIRM_TRANSFER2_SIZE   = 0x00000000
[08:42:08.017] 🅸 stream> SIRM_MAX_LEADER_SIZE  = 0x00000400
[08:42:08.017] 🅸 stream> SIRM_MAX_TRAILER_SIZE = 0x00000400
Start acquisition
FPS: 167, Running for [0] sec
FPS: 167, Running for [1] sec
FPS: 167, Running for [2] sec
FPS: 167, Running for [3] sec
FPS: 167, Running for [4] sec
FPS: 167, Running for [5] sec
^CTraceback (most recent call last):
  File "/home/pacaud/Sources/aravis/build/test.py", line 43, in <module>
    buffer = stream.pop_buffer()
             ^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt
[08:42:14.529] 🅸 stream> [UvStream::finalize] n_completed_buffers    = 1081
[08:42:14.529] 🅸 stream> [UvStream::finalize] n_failures             = 0
[08:42:14.529] 🅸 stream> [UvStream::finalize] n_underruns            = 0
[08:42:14.530] 🅸 stream> [UvStream::finalize] n_aborted              = 0
[08:42:14.530] 🅸 stream> [UvStream::finalize] n_transferred_bytes    = 622746804
[08:42:14.530] 🅸 stream> [UvStream::finalize] n_ignored_bytes        = 0
[08:42:14.530] 🅸 stream> [Stream::finalize] Flush 46 buffer[s] in input queue
[08:42:14.530] 🅸 stream> [Stream::finalize] Flush 3 buffer[s] in output queue
```

The CPU use is 16/18% of one core of a i5-8365U.

Did you check the CPU use on your machine ?

Please open another issue for the multi camera issue, with a sample application.

WeaponXI, 2023-05-26:
Hi @EmmanuelP ,
Thanks for looking in to this.

Now I am able to get up tp 57-58 FPS now in python as well.

I earlier used Processes (multiprocessing) for multiple cameras and displaying in a separate process.
Now I changes processes to Thread and also displaying in to main process. Somehow I am able to get full FPS. But what happed after changing process to Thread still don't know. 
May be data sharing overhead using sharedmemory is more with Process compared to Thread.

We can close this issue now.
Thanks.

EmmanuelP, 2023-05-26:
Ok. Thanks.

closed 2023-05-26 by EmmanuelP