AravisProject/aravis #788: Only receiving a few frames with software trigger and stream callback

rgov, 2023-05-12:
**Describe the bug**
I have my camera set in continuous capture mode, with a software trigger. My stream has *n* buffers attached. A stream callback is configured.

The first *n* times I fire the software trigger, I get a`ARV_STREAM_CALLBACK_TYPE_BUFFER_DONE` callback. Afterwards, the software trigger does not cause a callback to fire.

**Camera description:**
 - Manufacturer: 
 - Model
 - Interface [e.g. USB3] 

**Platform description:**
 - Aravis version: 0.8.20
 - OS: Ubuntu 22.04
 - Hardware: Raspberry Pi

**Other note:**
The `ArvStreamCallback` type is not documented, especially with regard to how the recipient of the ArvBuffer is expected to manage it.

EmmanuelP, 2023-05-17:
Hi,

I have created a callback example here:

https://github.com/AravisProject/aravis-c-examples/blob/e56b3feba261823ac0018489fa10724b7d4f3798/02-multiple-acquisition-callback.c

The buffers must be pushed back to the stream input FIFO to be reused.

closed 2023-05-17 by EmmanuelP

rgov, 2023-05-17:
Cool, thank you.