AravisProject/aravis #795: Current and cached value mismatch

matwey, 2023-05-31:
**Describe the bug**
I am facing "Current and cached value mismatch" error when I run the code sample below with `ARV_DEBUG=genicam`:
```
[22:38:41.276] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationLLA'
[22:38:41.276] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationDHCP'
[22:38:41.276] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationPersistentIP'
[22:38:41.276] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPAddress'
[22:38:41.276] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentSubnetMask'
[22:38:41.277] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentDefaultGateway'
[22:38:41.277] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevNumberOfNetworkInterfaces'
[22:38:41.277] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentIPAddress'
[22:38:41.277] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentSubnetMask'
[22:38:41.277] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentDefaultGateway'
[22:38:41.277] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelCount'
[22:38:41.278] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelSelector'
[22:38:41.278] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPAddrCalc'
[22:38:41.278] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevTimestampTickFrequency'
[22:38:41.278] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPHostPort'
[22:38:41.278] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSFireTestPacket'
[22:38:41.278] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSDoNotFragment'
[22:38:41.278] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSBigEndian'
[22:38:41.278] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSPacketSize'
[22:38:41.278] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPD'
[22:38:41.278] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCDA'
[22:38:41.279] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCSP'
[22:38:41.279] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCCFGMultipart'
[22:38:41.303] 🆆 policies> Current and cached value mismatch for 'LineMode_Output_Avl'
[22:38:41.304] 🆆 policies> Current and cached value mismatch for 'LineMode_Input_Avl'
```

If I use `ARV_REGISTER_CACHE_POLICY_ENABLE`, then I see the following issue:
```
Error: [LineMode] [LineMode] No available entry found
```

**To Reproduce**
```C
#include <stdlib.h>
#include <stdio.h>

#include <arv.h>

int main (int argc, char **argv) {
	ArvCamera *camera;
	GError *error = NULL;
	const char* ret;

	camera = arv_camera_new (NULL, &error);
	if (error != NULL) {
		printf ("Error: %s\n", error->message);
		return 1;
	}

	arv_camera_set_register_cache_policy(camera, ARV_REGISTER_CACHE_POLICY_DEBUG);

	arv_camera_set_string (camera, "LineSelector", "Line0", &error);
	if (error != NULL) {
		printf ("Error: %s\n", error->message);
		return 1;
	}

	arv_camera_set_boolean (camera, "LineInverter", 1, &error);
	if (error != NULL) {
		printf ("Error: %s\n", error->message);
		return 1;
	}

	arv_camera_set_string (camera, "LineMode", "Input", &error);
	if (error != NULL) {
		printf ("Error: %s\n", error->message);
		return 1;
	}

	arv_camera_set_string (camera, "LineSelector", "Line1", &error);
	if (error != NULL) {
		printf ("Error: %s\n", error->message);
		return 1;
	}

	arv_camera_set_boolean (camera, "LineInverter", 1, &error);
	if (error != NULL) {
		printf ("Error: %s\n", error->message);
		return 1;
	}

	arv_camera_set_string (camera, "LineMode", "Output", &error);
	if (error != NULL) {
		printf ("Error: %s\n", error->message);
		return 1;
	}

	printf ("Ok\n");

	return 0;
}
```

**Expected behavior**
The program is expected to run successfully with `ARV_REGISTER_CACHE_POLICY_ENABLE`:
```
Ok
```

**Camera description:**
 - Flir
 - BFS-PGE-16S2M
 - GigE

**Platform description:**
 - Aravis version 0.8.26
 - OS: openSUSE Leap 15.4
 - Hardware x86_64

matwey, 2023-05-31:
genicam dump is here: [flir.xml.gz](https://github.com/AravisProject/aravis/files/11618366/flir.xml.gz)

EmmanuelP, 2023-06-01:
Hi,

Thanks for the report.

It looks like the Genicam data are incorrect. Default cachable for a MaskedInt is WriteThrough. LineMode_Output_Avl and LineMode_Input_Avl are read only, and the cache is never invalidated by a pSelected, so Aravis is always using what is in the cache after once it is filled after a first read. And it seems the register value is actually changing.

The cache feature is not mandatory, which means its support is varying between manufacturers. This is why it is not enabled by default in Aravis.

But my understanding is may be wrong.  An input from FLIR would be helpful.