AravisProject/aravis #796: Failed to claim USB interface to '...' : LIBUSB_ERROR_NOT_SUPPORTED

mackel90, 2023-06-07:
**Describe the bug**
Having an issue opening USB3Vision Cameras in Windows 10. Aravis build with recent sys2,mingw.
Have used Zadig with all possible driver installations (libusbK, libusb-win32 and even winusb). 
Have also used a U3V Camera and versions from another supplier. Under Mac they are working properly.

Checked out GIT (release tag  0.8.26) , installed all dependencies and built. No changes.

Giving following output:
```$ arv-tool-0.8.exe features
IDS Imaging Development Systems GmbH-<stripped> (USB3)
[14:47:27.091] ­ƒà© device> [UvDevice::new] GUID    = 1409f4a5ac74
[14:47:27.101] ­ƒåå device> Failed to set auto kernel detach feature for USB device '(NULL)-(NULL)-(NULL)': LIBUSB_ERROR_NOT_SUPPORTED
[14:47:27.102] ­ƒà© device> [UvDevice::new] Using control endpoint 1, interface 0
[14:47:27.102] ­ƒà© device> [UvDevice::new] Using data endpoint 2, interface 1

(arv-tool-0.8.exe:7712): GLib-CRITICAL **: 14:47:27.102: g_thread_join: assertion 'thread' failed
[14:47:27.112] ­ƒà© device> [UvDevice::new] GUID    = 1409f4a5ac74
[14:47:27.121] ­ƒåå device> Failed to set auto kernel detach feature for USB device '(NULL)-(NULL)-(NULL)': LIBUSB_ERROR_NOT_SUPPORTED
[14:47:27.121] ­ƒà© device> [UvDevice::new] Using control endpoint 1, interface 0
[14:47:27.122] ­ƒà© device> [UvDevice::new] Using data endpoint 2, interface 1

(arv-tool-0.8.exe:7712): GLib-CRITICAL **: 14:47:27.122: g_thread_join: assertion 'thread' failed
Failed to open device 'IDS Imaging Development Systems GmbH-<stripped>': Failed to claim USB interface to '(NULL)-(NULL)-(NULL)-1409f4a5ac74': LIBUSB_ERROR_NOT_SUPPORTED
```

**To Reproduce**
$ arv-tool-0.8.exe features

**Expected behavior**
Opens the camera, prints the features.

**Camera description:**
 - Multiple: TheImagingSource / IDS
 - Multiple
 - USB3

**Platform description:**
- Version: 0.8.26
- gcc version 13.1.0 (Rev6, Built by MSYS2 project)
- OS: Windows 10 22H2 64 Bit

**Additional context**
Add any other context about the problem here.

EmmanuelP, 2023-07-05:
Hi,

Would you mind to explain how you solved this issue ?

mackel90, 2023-08-08:
Sorry for replying so late. I have solved this issue installing the libusbK driver to the Composite Parent in Zadig. These gets listed whan you switch the default setting "Ignore Hubs or Composite Parents". To get an idea what the right composite parent might be the right one right I used "USB Device Tree Viewer".