AravisProject/aravis #801: Use same network interface to connect to detected camera

rgov, 2023-06-24:
I think I have the same issue as #716 with Aravis 0.8.27: The network auto discovery of a GigE camera works, but subsequently Aravis cannot connect to it:

```
$ arv-tool-0.8 features
Lucid Vision Labs-PHX051S-C-222701823 (169.254.37.213)
Failed to open device 'Lucid Vision Labs-PHX051S-C-222701823': GigEVision read_memory timeout
```

If we check the routing tables, the networking stack is trying to route the connection over Wi-Fi, but the camera is really accessible over `eth0`:

```
$ ip route get 169.254.37.213
169.254.37.213 via 192.168.1.1 dev wlan0 src 192.168.1.239 uid 1000 
    cache 

$ ping -I eth0 169.254.37.213
PING 169.254.37.213 (169.254.37.213) from 192.168.99.4 eth0: 56(84) bytes of data.
64 bytes from 169.254.37.213: icmp_seq=1 ttl=255 time=0.864 ms
64 bytes from 169.254.37.213: icmp_seq=2 ttl=255 time=0.869 ms
```

If we manually add an entry in the routing table, it works:

```
$ sudo ip route add 169.254.37.213/32 dev eth0

$ arv-tool-0.8 features
Lucid Vision Labs-PHX051S-C-222701823 (169.254.37.213)
Category    : 'Root'
    Category    : 'DeviceControl'
        ...
```

Could Aravis use the same network interface for communication as the one that the camera was discovered on?

rgov, 2023-06-25:
Maybe this would be implemented by #779.

EmmanuelP, 2023-07-05:
It makes sense.