AravisProject/aravis #803: Namespace not found error, fresh install on fedora 37

drmcnelson, 2023-07-10:
The listing inserted below shows output from a fresh install of Aravis 0.8.27, on a recently upgraded Fedora 37, cinnamon desktop, I7 with 32 GB RAM.

The following errors is produced on attempting to use Aravis from python.  The error is not resolved by ldconfig, not by adding a file to /etc/ld.so.conf.d/ to specify the path to the libary, and not by rebooting after all else.

Python 3.11.2 (main, Feb  8 2023, 00:00:00) [GCC 12.2.1 20221121 (Red Hat 12.2.1-4)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gi
>>> gi.require_version( 'Aravis', '0.8' )\
>>> gi.require_version( 'Aravis', '0.8' )
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib64/python3.11/site-packages/gi/__init__.py", line 126, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace Aravis not available

======================================================

[nelson@LAB02 build]$ sudo ninja install
[0/1] Installing files.
Installing subdir /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/viewer/icons/gnome to /usr/local/share/icons/hicolor
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/viewer/icons/gnome/128x128/apps/aravis-0.8.png to /usr/local/share/icons/hicolor/128x128/apps
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/viewer/icons/gnome/22x22/apps/aravis-0.8.png to /usr/local/share/icons/hicolor/22x22/apps
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/viewer/icons/gnome/256x256/apps/aravis-0.8.png to /usr/local/share/icons/hicolor/256x256/apps
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/viewer/icons/gnome/32x32/apps/aravis-0.8.png to /usr/local/share/icons/hicolor/32x32/apps
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/viewer/icons/gnome/48x48/apps/aravis-0.8.png to /usr/local/share/icons/hicolor/48x48/apps
Installing src/libaravis-0.8.so.0.8.27 to /usr/local/lib64
Installing src/arv-tool-0.8 to /usr/local/bin
Installing src/arv-test-0.8 to /usr/local/bin
Installing src/arv-camera-test-0.8 to /usr/local/bin
Installing src/arv-fake-gv-camera-0.8 to /usr/local/bin
Installing src/Aravis-0.8.gir to /usr/local/share/gir-1.0
Installing src/Aravis-0.8.typelib to /usr/local/lib64/girepository-1.0
Installing po/bs/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/bs/LC_MESSAGES
Installing po/ca/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/ca/LC_MESSAGES
Installing po/cs/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/cs/LC_MESSAGES
Installing po/da/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/da/LC_MESSAGES
Installing po/de/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/de/LC_MESSAGES
Installing po/el/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/el/LC_MESSAGES
Installing po/es/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/es/LC_MESSAGES
Installing po/eu/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/eu/LC_MESSAGES
Installing po/fi/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/fi/LC_MESSAGES
Installing po/fr/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/fr/LC_MESSAGES
Installing po/hu/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/hu/LC_MESSAGES
Installing po/id/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/id/LC_MESSAGES
Installing po/lv/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/lv/LC_MESSAGES
Installing po/oc/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/oc/LC_MESSAGES
Installing po/pl/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/pl/LC_MESSAGES
Installing po/pt/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/pt/LC_MESSAGES
Installing po/pt_BR/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/pt_BR/LC_MESSAGES
Installing po/ru/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/ru/LC_MESSAGES
Installing po/sl/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/sl/LC_MESSAGES
Installing po/sr/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/sr/LC_MESSAGES
Installing po/sr@latin/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/sr@latin/LC_MESSAGES
Installing po/sv/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/sv/LC_MESSAGES
Installing po/tr/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/tr/LC_MESSAGES
Installing po/zh_CN/LC_MESSAGES/aravis-0.8.mo to /usr/local/share/locale/zh_CN/LC_MESSAGES
Installing viewer/arv-viewer-0.8 to /usr/local/bin
Installing viewer/data/arv-viewer-0.8.desktop to /usr/local/share/applications
Installing gst/libgstaravis.0.8.so to /usr/local/lib64/gstreamer-1.0
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/src/arv.h to /usr/local/include/aravis-0.8
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/src/arvtypes.h to /usr/local/include/aravis-0.8
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/src/arvbuffer.h to /usr/local/include/aravis-0.8
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/src/arvcamera.h to /usr/local/include/aravis-0.8
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/src/arvchunkparser.h to /usr/local/include/aravis-0.8
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/src/arvdebug.h to /usr/local/include/aravis-0.8
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/src/arvdevice.h to /usr/local/include/aravis-0.8
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/src/arvdomcharacterdata.h to /usr/local/include/aravis-0.8
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/src/arvdomdocumentfragment.h to /usr/local/include/aravis-0.8
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/src/arvdomdocument.h to /usr/local/include/aravis-0.8
Installing /home/nelson/Projects/FLIR_Camera/aravis-0.8.27/src/arvdomelement.h to /usr/local/include/aravis-0.8
[... 71 lines cut ...]
Installing symlink pointing to libaravis-0.8.so.0 to /usr/local/lib64/libaravis-0.8.so

[nelson@LAB02 build]$ python
Python 3.11.2 (main, Feb  8 2023, 00:00:00) [GCC 12.2.1 20221121 (Red Hat 12.2.1-4)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gi
>>> gi.require_version( 'Aravis', '0.8' )\
>>> gi.require_version( 'Aravis', '0.8' )
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib64/python3.11/site-packages/gi/__init__.py", line 126, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace Aravis not available
>>> quit()

[nelson@LAB02 FLIR_Camera]$ ls /usr/local/lib64/ -l
total 1660
drwxr-xr-x. 1 root root       0 Aug  9  2022 bpf
drwxr-xr-x. 1 root root      36 Jul  9 18:21 girepository-1.0
drwxr-xr-x. 1 root root      38 Jul  9 18:21 gstreamer-1.0
lrwxrwxrwx. 1 root root      18 Jul  9 18:21 libaravis-0.8.so -> libaravis-0.8.so.0
lrwxrwxrwx. 1 root root      23 Jul  9 18:21 libaravis-0.8.so.0 -> libaravis-0.8.so.0.8.27
-rwxr-xr-x. 1 root root 1688768 Jul  9 17:36 libaravis-0.8.so.0.8.27
drwxr-xr-x. 1 root root      26 Jul  9 18:21 pkgconfig
drwxr-xr-x. 1 root root      26 Jan 24  2022 python3.10

EmmanuelP, 2023-07-11:
You should set `GI_TYPELIB_PATH` to  `/usr/local/lib64/girepository-1.0`, or install aravis in `/usr`, not `/usr/local/`, using meson prefix configuration.

drmcnelson, 2023-07-11:
Thank you.  I believe I tried the first. It produces the same error.

Can you be more specific on the second?

How do I "install aravis in /usr/local using meson prefix configuration"?

Aside, two things that may be of interest:

/usr/local is the result following the installation instructions exactly.

And, on another machine, same os, it installed into /usr/local and runs
fine.

EmmanuelP, 2023-07-11:
Using `GI_TYPELIB_PATH` works for me on fedora 38 (my install prefix is `/home/pacaud/Sources/INSTALL-meson/`):

```
export GI_TYPELIB_PATH=/home/pacaud/Sources/INSTALL-meson/lib64/girepository-1.0/
Sources/aravis/build on  wip/bindtodevice [$] via 🐍 v3.11.4 ❯ python3
Python 3.11.4 (main, Jun  7 2023, 00:00:00) [GCC 13.1.1 20230511 (Red Hat 13.1.1-2)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gi
>>> gi.require_version( 'Aravis', '0.8' )
>>> 
```

EmmanuelP, 2023-07-11:
> How do I "install aravis in /usr/local using meson prefix configuration"?

Installation prefix is set using `meson configure`. For example, for an installation in `/usr`:

```
meson configure --prefix=/usr
```

I don't recommend installing in `/usr`, as it mixes user compiled software wih the one provided by your distribution. Correctly setting the environment variable is better.

drmcnelson, 2023-07-11:
Thank you.

Will try it again.

EmmanuelP, 2023-07-17:
Did you succeed in using aravis from python ?

drmcnelson, 2023-07-17:
Sorry, I have not yet had time to try it again in that machine.   Planning to try again later today or tomorrow.

drmcnelson, 2023-07-17:
[LAB02 ~]$ export GI_TYPELIB_PATH=/usr/local/lib64/girepository-1.0
[LAB02 ~]$ python
Python 3.11.2 (main, Feb  8 2023, 00:00:00) [GCC 12.2.1 20221121 (Red Hat 12.2.1-4)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gi
>>> gi.require_version( 'Aravis', '0.8' )
>>> 

Seems to work, thank you,

So, the export should go into my bashrc for that machine?

It remains a curiosity why on that machine, the export is needed and on the other machines, all with the same software installation, it is not needed.

EmmanuelP, 2023-07-18:
> Seems to work, thank you,

Great. I'm closing this issue then.

> So, the export should go into my bashrc for that machine?

Yes.
 
> It remains a curiosity why on that machine, the export is needed and on the other machines, all with the same software installation, it is not needed.

Same distribution ? Which one ?

closed 2023-07-18 by EmmanuelP

drmcnelson, 2023-07-18:
> Fedora Cinnamon spin, 37

Fedora Cinnamon, 37

But to.be more specific, upgraded from 36, from 35, from... etc..   The
fresh installs were quite a ways back, the other machine might be a bit
less far back (after a disk problem) now that I think about it, and the hw
is a little different. This one is a gen 1 i7, the other is a 3770k, I
think.

So they're not really so identicsl