AravisProject/aravis #813: Packet resend problem.

KamiJW, 2023-08-15:
**Describe the bug**
Some cameras such as Hikvision cameras will send incorrect tailer packet if last_packet_id = n_packet - 1. the tailer(0x02) becomes payload(0x03) packet with status not equals success.  Those cameras will send the correct tailer packet if the request last_packet_id = 0xFFFFFF(or 0xFFFFFFFF).

**Camera description:**
 - Manufacturer Hikvision
 - Interface gige

EmmanuelP, 2023-08-16:
Hi,

Thanks for the report.

A quick fix would be to replace last_packet_id by 0xffffffff or 0xffffff if last_packet_id is equal to n_expected_packets - 1. Unfortunately, the computation of n_expected_packets is a bit fragile, and in some cases not even predictable.

I suspect the behaviour of the HikVision camera is not expected by the specification. I would suggest to report this to HikVision.

KamiJW, 2023-08-18:
Thank you for the reply. I have communicated with the HikVision support, I can not persuade them to optimize this. I can not find a clear statement in the standard saying last_packet_id must be or maybe  0xffffff if the receiver want the tailer packet. I did implement the quick fix you said in my code since sending 0xffffff is stated in the standard.

closed 2023-08-21 by KamiJW