AravisProject/aravis #826: Access Mode on acquisition

CynthiaLiLi, 2023-10-10:
**Describe the bug**
When I start the acquisition of a camera, some features are still marked as writable when I check their actual Access Mode while it’s supposed to be read only.
For example, I expect the feature “Width” to be read only after I start the acquisition, but it’s writable.

**To Reproduce**
When I call arv_camera_set_access_check_policy after creating the camera object using arv_camera_new to set the access check policy to enable.
Then start acquisition, and then use arv_gc_feature_node_get_actual_access_mode to return the result still RW.

**Expected behavior**
When the camera is in acquisition state, some feature nodes obtained using arv_gc_feature_node_get_actual_access_mode should be read-only.

**Camera description:**
 - Manufacturer [Hikrobot]
 - Interface [GigE]

**Platform description:**
 - Aravis version [0.8.29]
 - OS [windows]
 - Hardware [x86_64]
 - Development Framework [QT]

**Additional context**
ARV-DEBUG as flows:

CynthiaLiLi, 2023-10-10:
[16:12:01.816] 馃叧 genicam> [Gc::register_feature_node] Register node 'Root' [Category]
[16:12:01.816] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceControl' [Category]
[16:12:01.816] 馃叧 genicam> [Gc::register_feature_node] Register node 'ImageFormatControl' [Category]
[16:12:01.816] 馃叧 genicam> [Gc::register_feature_node] Register node 'AcquisitionControl' [Category]
[16:12:01.816] 馃叧 genicam> [Gc::register_feature_node] Register node 'DigitalIOControl' [Category]
[16:12:01.816] 馃叧 genicam> [Gc::register_feature_node] Register node 'ActionControl' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'CounterAndTimerControl' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'FileAccessControl' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventControl' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventAcquisitionStartData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventAcquisitionEndData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventFrameStartData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventFrameEndData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventFrameBurstStartData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventFrameBurstEndData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventExposureStartData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventExposureEndData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventLine0RisingEdgeData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventLine0FallingEdgeData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventErrorData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventTestData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventOverTemperatureData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventOverRunData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventFrameStartOverTriggerData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventStreamTransferOverflowData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventAcquisitionTriggerData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventAcquisitionTriggerEndData' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'ChunkDataControl' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'ColorTransformationControl' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'Debug' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'AnalogControl' [Category]
[16:12:01.817] 馃叧 genicam> [Gc::register_feature_node] Register node 'LUTControl' [Category]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'EncoderControl' [Category]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'FrequencyConverterControl' [Category]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'ShadingCorrection' [Category]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'TransferControl' [Category]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'UserSetControl' [Category]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'TransportLayerControl' [Category]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'Device' [Port]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'One_Int' [Integer]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'Zero_Int' [Integer]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'AcquisitionStart_RegAddr' [Integer]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'AcquisitionStop_RegAddr' [Integer]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceTemperature_RegAddr' [Integer]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceReset_RegAddr' [Integer]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'FindMe_RegAddr' [Integer]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'SensorChannelVersion_RegAddr' [Integer]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceLinkThroughputLimitMode_RegAddr' [Integer]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceLinkThroughputLimit_RegAddr' [Integer]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceLinkHeartbeatMode_RegAddr' [Integer]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceLinkHeartbeatTimeout_RegAddr' [Integer]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'DevicePJNumber_RegAddr' [Integer]
[16:12:01.818] 馃叧 genicam> [Gc::register_feature_node] Register node 'AlgorithmAbility_RegAddr' [Integer]
[... cut ...]
[16:12:01.829] 馃叧 genicam> [Gc::register_feature_node] Register node 'UCCIndexIncInq_RegAddr' [Integer]
[16:12:01.829] 馃叧 genicam> [Gc::register_feature_node] Register node 'UCCValueIndexMinInq_RegAddr' [Integer]
[16:12:01.829] 馃叧 genicam> [Gc::register_feature_node] Register node 'UCCValueIndexMaxInq_RegAddr' [Integer]
[16:12:01.829] 馃叧 genicam> [Gc::register_feature_node] Register node 'UCCValueIndexIncInq_RegAddr' [Integer]
[16:12:01.829] 馃叧 genicam> [Gc::register_feature_node] Register node 'PRNUCTargetMinInq_RegAddr' [Integer]

CynthiaLiLi, 2023-10-10:
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'PRNUCTargetMaxInq_RegAddr' [Integer]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'PRNUCTargetIncInq_RegAddr' [Integer]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventSelectorInq_RegAddr' [Integer]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventMCTTMaxInq_RegAddr' [Integer]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'CtrlTransCtrlInq_RegAddr' [Integer]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'ColorTransformationValueMaxInq_RegAddr' [Integer]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'ColorTransformationValueMinInq_RegAddr' [Integer]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'ColorTransformationValueUnitInq_RegAddr' [Integer]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'LineRateRatioMinInq_RegAddr' [Integer]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'LineRateRatioMaxInq_RegAddr' [Integer]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'PixelFormatSupHBInq0_RegAddr' [Integer]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'PixelFormatSupHBInq_RegAddr' [Integer]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'CounterAndTimerControl_Category_Inq' [StructEntry]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'EncoderControl_Category_Inq' [StructEntry]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'FrequencyConverter_Category_Inq' [StructEntry]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'ShadingCorrection_Category_Inq' [StructEntry]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'FileAccessControl_Category_Inq' [StructEntry]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'EventControl_Category_Inq' [StructEntry]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'ChunkDataControl_Category_Inq' [StructEntry]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'ActionControl_Category_Inq' [StructEntry]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'ColorTransformationControl_Category_Inq' [StructEntry]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'TransferControl_Category_Inq' [StructEntry]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceType' [Enumeration]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceType_Reg' [StructEntry]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceCharacterSet_Reg' [StructEntry]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceScanType' [Enumeration]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceScanType_Reg' [IntReg]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceVendorName' [StringReg]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceModelName' [StringReg]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceManufacturerInfo' [StringReg]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceVersion' [StringReg]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceFirmwareVersion' [StringReg]
[16:12:01.830] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceSerialNumber' [StringReg]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceID' [StringReg]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceUserID' [StringReg]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceUptime' [Integer]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceUptime_Reg' [IntReg]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'BoardDeviceType' [Integer]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'BoardDeviceType_Reg' [IntReg]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceTemperatureSelector_Inq' [StructEntry]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceTemperatureSelector_Sensor_Inq' [StructEntry]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceTemperatureSelector_Mainboard_Inq' [StructEntry]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceLinkThroughputLimit_Inq' [StructEntry]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceMaxThroughput_Inq' [StructEntry]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'DeviceLinkHeartbeatTimeout_Inq' [StructEntry]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'DevicePacketUnorderSupport_Inq' [StructEntry]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'SensorChannelVersion_Inq' [StructEntry]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'TG_Inq' [StructEntry]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'TDIMode_Inq' [StructEntry]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'SupUserOutputValue_Inq' [StructEntry]
[16:12:01.831] 馃叧 genicam> [Gc::register_feature_node] Register node 'CHAP_Inq' [StructEntry]
[... cut ...]
[16:12:01.898] 馃叧 genicam> [Gc::register_feature_node] Register node 'TransferQueueMode_Reg' [IntReg]
[16:12:01.898] 馃叧 genicam> [Gc::register_feature_node] Register node 'TransferStart_Reg' [IntReg]
[16:12:01.898] 馃叧 genicam> [Gc::register_feature_node] Register node 'TransferStop_Reg' [IntReg]
[16:12:01.898] 馃叧 genicam> [Gc::register_feature_node] Register node 'TransferAbort_Reg' [IntReg]
[16:12:01.898] 馃叧 genicam> [Gc::register_feature_node] Register node 'TransferPause_Reg' [IntReg]
[16:12:01.898] 馃叧 genicam> [Gc::register_feature_node] Register node 'TransferResume_Reg' [IntReg]
[1

CynthiaLiLi, 2023-10-10:
[16:12:01.917] 馃叧 genicam> [GcRegisterNode::get] 0x48,32
[16:12:01.919] 馃叧 genicam> [GcRegisterNode::get] 0x68,32
[16:12:01.920] 馃叧 genicam> [GcRegisterNode::_get_integer_value] reglsb = 31, regmsb, 16, lsb = 0, msb = 15
[16:12:01.920] 馃叧 genicam> [GcRegisterNode::_get_integer_value] value = 0x40001f04
[16:12:01.920] 馃叧 genicam> [GcRegisterNode::_get_integer_value] mask  = 0x0000ffff
[16:12:01.920] 馃叧 genicam> [GcRegisterNode::_get_integer_value] address = 0xd04, value = 0x1f04
[16:12:01.920] 馃叧 genicam> [GcRegisterNode::_get_integer_value] address = 0x10124, value = 0xdc
[16:12:01.921] 馃叧 genicam> [GcRegisterNode::_get_integer_value] address = 0x10120, value = 0x23c4
[16:12:01.922] 馃叧 genicam> [GcRegisterNode::_get_integer_value] address = 0x10144, value = 0x4
[16:12:01.922] 馃叧 genicam> [GcRegisterNode::_set_integer_value] address = 0xd18, value = 0xc0a8016b
[16:12:01.924] 馃叧 genicam> [GcRegisterNode::_set_integer_value] reglsb = 31, regmsb, 16, lsb = 0, msb = 15
[16:12:01.924] 馃叧 genicam> [GcRegisterNode::_set_integer_value] value = 0x0000ea9b
[16:12:01.924] 馃叧 genicam> [GcRegisterNode::_set_integer_value] mask  = 0x0000ffff
[16:12:01.924] 馃叧 genicam> [GcRegisterNode::_set_integer_value] address = 0xd00, value = 0xea9b
[16:12:01.926] 馃叧 genicam> [GcRegisterNode::_get_integer_value] reglsb = 1, regmsb, 1, lsb = 30, msb = 30
[16:12:01.926] 馃叧 genicam> [GcRegisterNode::_get_integer_value] value = 0x40001f04
[16:12:01.926] 馃叧 genicam> [GcRegisterNode::_get_integer_value] mask  = 0x40000000
[16:12:01.926] 馃叧 genicam> [GcRegisterNode::_get_integer_value] address = 0xd04, value = 0x1
[16:12:01.927] 馃叧 genicam> [GcRegisterNode::_set_integer_value] reglsb = 1, regmsb, 1, lsb = 30, msb = 30
[16:12:01.927] 馃叧 genicam> [GcRegisterNode::_set_integer_value] value = 0x00000001
[16:12:01.927] 馃叧 genicam> [GcRegisterNode::_set_integer_value] mask  = 0x40000000
[16:12:01.927] 馃叧 genicam> [GcRegisterNode::_set_integer_value] address = 0xd04, value = 0x40001f04
[16:12:01.931] 馃叧 genicam> [GcRegisterNode::_set_integer_value] reglsb = 0, regmsb, 0, lsb = 31, msb = 31
[16:12:01.931] 馃叧 genicam> [GcRegisterNode::_set_integer_value] value = 0x00000001
[16:12:01.931] 馃叧 genicam> [GcRegisterNode::_set_integer_value] mask  = 0x80000000
[16:12:01.931] 馃叧 genicam> [GcRegisterNode::_set_integer_value] address = 0xd04, value = 0xc0001f04
[16:12:01.940] 馃叧 genicam> [GcCommand::execute] ArvGevSCPSFireTestPacket (0x1)
[16:12:01.941] 馃叧 genicam> [GcRegisterNode::_set_integer_value] reglsb = 31, regmsb, 16, lsb = 0, msb = 15
[16:12:01.941] 馃叧 genicam> [GcRegisterNode::_set_integer_value] value = 0x00001f04
[16:12:01.941] 馃叧 genicam> [GcRegisterNode::_set_integer_value] mask  = 0x0000ffff
[16:12:01.941] 馃叧 genicam> [GcRegisterNode::_set_integer_value] address = 0xd04, value = 0x40001f04
[16:12:01.944] 馃叧 genicam> [GcRegisterNode::_get_integer_value] reglsb = 31, regmsb, 16, lsb = 0, msb = 15
[16:12:01.944] 馃叧 genicam> [GcRegisterNode::_get_integer_value] value = 0x40001f04
[16:12:01.944] 馃叧 genicam> [GcRegisterNode::_get_integer_value] mask  = 0x0000ffff
[16:12:01.944] 馃叧 genicam> [GcRegisterNode::_get_integer_value] address = 0xd04, value = 0x1f04
[16:12:01.945] 馃叧 genicam> [GcRegisterNode::_set_integer_value] reglsb = 0, regmsb, 0, lsb = 31, msb = 31
[16:12:01.945] 馃叧 genicam> [GcRegisterNode::_set_integer_value] value = 0x00000001
[16:12:01.945] 馃叧 genicam> [GcRegisterNode::_set_integer_value] mask  = 0x80000000
[16:12:01.945] 馃叧 genicam> [GcRegisterNode::_set_integer_value] address = 0xd04, value = 0xc0001f04
[16:12:01.950] 馃叧 genicam> [GcCommand::execute] ArvGevSCPSFireTestPacket (0x1)
[16:12:01.952] 馃叧 genicam> [GcRegisterNode::_set_integer_value] reglsb = 31, regmsb, 16, lsb = 0, msb = 15
[16:12:01.952] 馃叧 genicam> [GcRegisterNode::_set_integer_value] value = 0x00002164
[16:12:01.952] 馃叧 genicam> [GcRegisterNode::_set_integer_value] mask  = 0x0000ffff
[16:12:01.952] 馃叧 genicam> [GcRegisterNode::_set_integer_value] address = 0xd04, value = 0x40002164
[16:12:01.955] 馃叧 genicam> [GcRegisterNode::_get_integer_value] reglsb = 31, regmsb, 16, lsb = 0, msb = 15
[16:12:01.955] 馃叧 genicam> [GcRegisterNode::_get_integer_value] value = 0x40002164
[16:12:01.955] 馃叧 genicam> [GcRegisterNode::_get_integer_value] mask  = 0x0000ffff
[16:12:01.955] 馃叧 genicam> [GcRegisterNode::_get_integer_value] address = 0xd04, value = 0x2164
[16:12:01.956] 馃叧 genicam> [GcRegisterNode::_set_integer_value] reglsb = 0, regmsb, 0, lsb = 31, msb = 31
[16:12:01.956] 馃叧 genicam> [GcRegisterNode::_set_integer_value] value = 0x00000001
[16:12:01.956] 馃叧 genicam> [GcRegisterNode::_set_integer_value] mask  = 0x80000000
[16:12:01.956] 馃叧 genicam> [GcRegisterNode::_set_integer_value] address = 0xd04, value = 0xc0002164
[16:12:01.960] 馃叧 genicam> [GcCommand::execute] ArvGevSCPSFireTestPacket (0x1)
[16:12:01.961] 馃叧 genicam> [GcRegisterNode::_set_integer_value] reglsb = 31, regmsb, 16, lsb = 0, msb = 15
[16:12:01.961] 馃叧 genicam> [GcRegisterNode::_set_integer_value] value = 0x00002294
[16:12:01.961] 馃叧 genicam> [GcRegisterNode::_set_integer_value] mask  = 0x0000ffff
[16:12:01.961] 馃叧 genicam> [GcRegisterNode::_set_integer_value] address = 0xd04, value = 0x40002294
[16:12:01.964] 馃叧 genicam> [GcRegisterNode::_get_integer_value] reglsb = 31, regmsb, 16, lsb = 0, msb = 15
[16:12:01.964] 馃叧 genicam> [GcRegisterNode::_get_integer_value] value = 0x40002294
[16:12:01.964] 馃叧 genicam> [GcRegisterNode::_get_integer_value] mask  = 0x0000ffff
[... cut ...]
[16:12:02.103] 馃叧 genicam> [GcRegisterNode::_get_integer_value] value = 0x03680000
[16:12:02.103] 馃叧 genicam> [GcRegisterNode::_get_integer_value] mask  = 0x00400000
[16:12:02.103] 馃叧 genicam> [GcRegisterNode::_get_integer_value] address = 0x10150, value = 0x1
[16:12:02.104] 馃叧 genicam> [GcRegisterNode::_get_integer_value] address = 0x304a0, value = 0x10001
[16:12:02.104] 馃叧 genicam> [GcEnumeration::get_string_value] value = 1 - string = BinningHorizontal1
BinningHorizontal acceess  2 val  "BinningHorizontal1"

EmmanuelP, 2023-10-10:
Please attach the Genicam data obtained with `arv-tool-0.8 genicam`.

CynthiaLiLi, 2023-10-10:
这是来自QQ邮箱的假期自动回复邮件。
 
你好，我最近正在休假中，无法亲自回复你的邮件。我将在假期结束后，尽快给你回复。

CynthiaLiLi, 2023-10-11:
**arv-tool-0.8 genicam output as flows:**
Hikrobot-MV-CB060-10GM-S-W-K05153419 (192.168.1.11)
<?xml version="1.0" encoding="UTF-8"?>

<RegisterDescription xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns="http://www.genicam.org/GenApi/Version_1_0" xmlns:mstns="http://www.genicam.org/GenApi/Version_1_0" ModelName="HIKCamera" VendorName="HIKROBOT" StandardNameSpace="GEV" SchemaMajorVersion="1" SchemaMinorVersion="0" SchemaSubMinorVersion="0" MajorVersion="1" MinorVersion="2" SubMinorVersion="0" ToolTip="HIKROBOT GigE Vision Camera Interface" ProductGuid="EE4B7E09-DA29-4956-A1EC-212263331CFC" VersionGuid="79ace398-62e3-44a6-8845-f81aaf104382" xsi:schemaLocation="http://www.genicam.org/GenApi/Version_1_0 GenApiSchema_Version_1_0.xsd">
  <Category Name="Root" NameSpace="Standard">
    <pFeature>DeviceControl</pFeature>
    <pFeature>ImageFormatControl</pFeature>
    <pFeature>AcquisitionControl</pFeature>
    <pFeature>AnalogControl</pFeature>
    <pFeature>ColorTransformationControl</pFeature>
    <pFeature>LUTControl</pFeature>
    <pFeature>EncoderControl</pFeature>
    <pFeature>FrequencyConverterControl</pFeature>
    <pFeature>ShadingCorrection</pFeature>
    <pFeature>DigitalIOControl</pFeature>
    <pFeature>ActionControl</pFeature>
    <pFeature>CounterAndTimerControl</pFeature>
    <pFeature>FileAccessControl</pFeature>
    <pFeature>EventControl</pFeature>
    <pFeature>ChunkDataControl</pFeature>
    
    <pFeature>TransportLayerControl</pFeature>
    <pFeature>TransferControl</pFeature>
    <pFeature>UserSetControl</pFeature>
  </Category>
  <Category Name="DeviceControl" NameSpace="Standard">
    <ToolTip>Category for device information and control.</ToolTip>
    <Description>Category for device information and control.</Description>
    <DisplayName>Device Control</DisplayName>
    <Visibility>Beginner</Visibility>
    <ImposedAccessMode>RO</ImposedAccessMode>
    <pFeature>DeviceType</pFeature>
    <pFeature>DeviceScanType</pFeature>
    <pFeature>DeviceVendorName</pFeature>
    <pFeature>DeviceModelName</pFeature>
    <pFeature>DeviceManufacturerInfo</pFeature>
    <pFeature>DeviceVersion</pFeature>
    <pFeature>DeviceFirmwareVersion</pFeature>
    <pFeature>DeviceSerialNumber</pFeature>
    <pFeature>DeviceID</pFeature>
    <pFeature>DeviceUserID</pFeature>
    <pFeature>DeviceUptime</pFeature>
    <pFeature>BoardDeviceType</pFeature>
    <pFeature>DeviceConnectionSelector</pFeature>
    <pFeature>DeviceConnectionSpeed</pFeature>
    <pFeature>DeviceConnectionStatus</pFeature>
    <pFeature>DeviceLinkSelector</pFeature>
    <pFeature>DeviceLinkSpeed</pFeature>
    
    
    <pFeature>DeviceLinkConnectionCount</pFeature>
    <pFeature>DeviceLinkHeartbeatMode</pFeature>
    
    
    <pFeature>DeviceStreamChannelCount</pFeature>
    <pFeature>DeviceStreamChannelSelector</pFeature>
    <pFeature>DeviceStreamChannelType</pFeature>
    <pFeature>DeviceStreamChannelLink</pFeature>
    <pFeature>DeviceStreamChannelEndianness</pFeature>
    <pFeature>DeviceStreamChannelPacketSize</pFeature>
    <pFeature>DeviceEventChannelCount</pFeature>
    <pFeature>DeviceCharacterSet</pFeature>
    <pFeature>DeviceReset</pFeature>
    <pFeature>DeviceTemperatureSelector</pFeature>
    <pFeature>DeviceTemperature</pFeature>
    <pFeature>FindMe</pFeature>
    <pFeature>SensorChannelVersion</pFeature>
    <pFeature>DeviceMaxThroughput</pFeature>
    <pFeature>DevicePJNumber</pFeature>
[... 4737 lines cut ...]
      <EnumEntry Name="YUV422Packed" NameSpace="Standard">
        <ToolTip>Pixel format set to YUV 422 Packed</ToolTip>
        <Description>Pixel format set to YUV 422 Packed.</Description>
        <DisplayName>YUV 422 Packed</DisplayName>
        <pIsImplemented>YUV422PackedInq_Reg</pIsImplemented>
        <Value>0x0210001F</Value>
      </EnumEntry>
      <EnumEntry Name="YUV8_UYV" NameSpace="Standard">
        <ToolTip>Pixel format set to YUV 444 Packed</ToolTip>
        <Description>Pixel format set to YUV 444 Packed.</Description>
        <DisplayName>YUV8 UYV</DisplayName>
        <pIsImplemented>YUV444PackedInq_Reg</pIsImplemented>
        <Value>0x02180020</Value>
      </EnumEntry>
      <EnumEntry Name="YUV411_8_UYYVYY" NameSpace="Standard">
        <ToolTip>Pixel format set to YUV411 Packed.</ToolTip>
        <Description>Pixel format set YUV411 Packed.</Description>
        <DisplayName>YUV411 8 UYYVYY</DisplayName>
        <pIsImplemented>YUV411PackedInq_Reg</pIsImplemented>
        <Value>0x020C001E</Value>
      </EnumEntry>
      <EnumEntry Name="BayerGR8" NameSpace="Standard">
        <ToolTip>Pixel format set  to Bayer GR 8</ToolTip>
        <Description>Pixel format set to Bayer GR 8.</Description>
        <DisplayName>Bayer GR 8</DisplayName>

CynthiaLiLi, 2023-10-11:
Hikrobot-MV-CB060-10GM-S-W-K05153419 (192.168.1.11)
<?xml version="1.0" encoding="UTF-8"?>

<RegisterDescription xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns="http://www.genicam.org/GenApi/Version_1_0" xmlns:mstns="http://www.genicam.org/GenApi/Version_1_0" ModelName="HIKCamera" VendorName="HIKROBOT" StandardNameSpace="GEV" SchemaMajorVersion="1" SchemaMinorVersion="0" SchemaSubMinorVersion="0" MajorVersion="1" MinorVersion="2" SubMinorVersion="0" ToolTip="HIKROBOT GigE Vision Camera Interface" ProductGuid="EE4B7E09-DA29-4956-A1EC-212263331CFC" VersionGuid="79ace398-62e3-44a6-8845-f81aaf104382" xsi:schemaLocation="http://www.genicam.org/GenApi/Version_1_0 GenApiSchema_Version_1_0.xsd">
  <Category Name="Root" NameSpace="Standard">
    <pFeature>DeviceControl</pFeature>
    <pFeature>ImageFormatControl</pFeature>
    <pFeature>AcquisitionControl</pFeature>
    <pFeature>AnalogControl</pFeature>
    <pFeature>ColorTransformationControl</pFeature>
    <pFeature>LUTControl</pFeature>
    <pFeature>EncoderControl</pFeature>
    <pFeature>FrequencyConverterControl</pFeature>
    <pFeature>ShadingCorrection</pFeature>
    <pFeature>DigitalIOControl</pFeature>
    <pFeature>ActionControl</pFeature>
    <pFeature>CounterAndTimerControl</pFeature>
    <pFeature>FileAccessControl</pFeature>
    <pFeature>EventControl</pFeature>
    <pFeature>ChunkDataControl</pFeature>
    
    <pFeature>TransportLayerControl</pFeature>
    <pFeature>TransferControl</pFeature>
    <pFeature>UserSetControl</pFeature>
  </Category>
  <Category Name="DeviceControl" NameSpace="Standard">
    <ToolTip>Category for device information and control.</ToolTip>
    <Description>Category for device information and control.</Description>
    <DisplayName>Device Control</DisplayName>
    <Visibility>Beginner</Visibility>
    <ImposedAccessMode>RO</ImposedAccessMode>
    <pFeature>DeviceType</pFeature>
    <pFeature>DeviceScanType</pFeature>
    <pFeature>DeviceVendorName</pFeature>
    <pFeature>DeviceModelName</pFeature>
    <pFeature>DeviceManufacturerInfo</pFeature>
    <pFeature>DeviceVersion</pFeature>
    <pFeature>DeviceFirmwareVersion</pFeature>
    <pFeature>DeviceSerialNumber</pFeature>
    <pFeature>DeviceID</pFeature>
    <pFeature>DeviceUserID</pFeature>
    <pFeature>DeviceUptime</pFeature>
    <pFeature>BoardDeviceType</pFeature>
    <pFeature>DeviceConnectionSelector</pFeature>
    <pFeature>DeviceConnectionSpeed</pFeature>
    <pFeature>DeviceConnectionStatus</pFeature>
    <pFeature>DeviceLinkSelector</pFeature>
    <pFeature>DeviceLinkSpeed</pFeature>
    
    
    <pFeature>DeviceLinkConnectionCount</pFeature>
    <pFeature>DeviceLinkHeartbeatMode</pFeature>
    
    
    <pFeature>DeviceStreamChannelCount</pFeature>
    <pFeature>DeviceStreamChannelSelector</pFeature>
    <pFeature>DeviceStreamChannelType</pFeature>
    <pFeature>DeviceStreamChannelLink</pFeature>
    <pFeature>DeviceStreamChannelEndianness</pFeature>
    <pFeature>DeviceStreamChannelPacketSize</pFeature>
    <pFeature>DeviceEventChannelCount</pFeature>
    <pFeature>DeviceCharacterSet</pFeature>
    <pFeature>DeviceReset</pFeature>
    <pFeature>DeviceTemperatureSelector</pFeature>
    <pFeature>DeviceTemperature</pFeature>
    <pFeature>FindMe</pFeature>
    <pFeature>SensorChannelVersion</pFeature>
    <pFeature>DeviceMaxThroughput</pFeature>
    <pFeature>DevicePJNumber</pFeature>
    <pFeature>CHAP</pFeature>
    <pFeature>HBAbnormalMonitor</pFeature>
    <pFeature>HBVersion</pFeature>
    
  </Category>

CynthiaLiLi, 2023-10-11:
```xml
Hikrobot-MV-CB060-10GM-S-W-K05153419 (192.168.1.11)
<?xml version="1.0" encoding="UTF-8"?>
<!--===========================================================
Copyright (c) 2014 Hangzhou Hikrobot Technology Co., Ltd. All Rights Reserved..

This software is the confidential and proprietary information of
Hangzhou Hikrobot Technology Co., Ltd..  
You shall not disclose such Confidential Information and shall
use it only in accordance with the terms of the license
agreement you entered into with HIKROBOT.

HIKROBOT MAKES NO REPRESENTATIONS OR WARRANTIES ABOUT THE SUITABILITY 
OF THE SOFTWARE, EITHER EXPRESS OR IMPLIED, INCLUDING BUT NOT 
LIMITED TO THE IMPLIED WARRANTIES OF MERCHANTABILITY, FITNESS
FOR A PARTICULAR PURPOSE, OR NON-INFRINGEMENT. HIKROBOT SHALL NOT BE 
LIABLE FOR ANY DAMAGES SUFFERED BY LICENSEE AS A RESULT OF USING,
MODIFYING OR DISTRIBUTING THIS SOFTWARE OR ITS DERIVATIVES.
===========================================================-->
<RegisterDescription xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns="http://www.genicam.org/GenApi/Version_1_0" xmlns:mstns="http://www.genicam.org/GenApi/Version_1_0" ModelName="HIKCamera" VendorName="HIKROBOT" StandardNameSpace="GEV" SchemaMajorVersion="1" SchemaMinorVersion="0" SchemaSubMinorVersion="0" MajorVersion="1" MinorVersion="2" SubMinorVersion="0" ToolTip="HIKROBOT GigE Vision Camera Interface" ProductGuid="EE4B7E09-DA29-4956-A1EC-212263331CFC" VersionGuid="79ace398-62e3-44a6-8845-f81aaf104382" xsi:schemaLocation="http://www.genicam.org/GenApi/Version_1_0 GenApiSchema_Version_1_0.xsd">
  <Category Name="Root" NameSpace="Standard">
    <pFeature>DeviceControl</pFeature>
    <pFeature>ImageFormatControl</pFeature>
    <pFeature>AcquisitionControl</pFeature>
    <pFeature>AnalogControl</pFeature>
    <pFeature>ColorTransformationControl</pFeature>
    <pFeature>LUTControl</pFeature>
    <pFeature>EncoderControl</pFeature>
    <pFeature>FrequencyConverterControl</pFeature>
    <pFeature>ShadingCorrection</pFeature>
    <pFeature>DigitalIOControl</pFeature>
    <pFeature>ActionControl</pFeature>
    <pFeature>CounterAndTimerControl</pFeature>
    <pFeature>FileAccessControl</pFeature>
    <pFeature>EventControl</pFeature>
    <pFeature>ChunkDataControl</pFeature>
    <!--<pFeature>Debug</pFeature>-->
    <pFeature>TransportLayerControl</pFeature>
    <pFeature>TransferControl</pFeature>
    <pFeature>UserSetControl</pFeature>
  </Category>
  <Category Name="DeviceControl" NameSpace="Standard">
    <ToolTip>Category for device information and control.</ToolTip>
    <Description>Category for device information and control.</Description>
    <DisplayName>Device Control</DisplayName>
    <Visibility>Beginner</Visibility>
    <ImposedAccessMode>RO</ImposedAccessMode>
    <pFeature>DeviceType</pFeature>
    <pFeature>DeviceScanType</pFeature>
    <pFeature>DeviceVendorName</pFeature>
    <pFeature>DeviceModelName</pFeature>
    <pFeature>DeviceManufacturerInfo</pFeature>
    <pFeature>DeviceVersion</pFeature>
    <pFeature>DeviceFirmwareVersion</pFeature>
    <pFeature>DeviceSerialNumber</pFeature>
    <pFeature>DeviceID</pFeature>
    <pFeature>DeviceUserID</pFeature>
    <pFeature>DeviceUptime</pFeature>
    <pFeature>BoardDeviceType</pFeature>
    <pFeature>DeviceConnectionSelector</pFeature>
    <pFeature>DeviceConnectionSpeed</pFeature>
    <pFeature>DeviceConnectionStatus</pFeature>
    <pFeature>DeviceLinkSelector</pFeature>
    <pFeature>DeviceLinkSpeed</pFeature>
    <!--<pFeature>DeviceLinkThroughputLimitMode</pFeature>-->
    <!--<pFeature>DeviceLinkThroughputLimit</pFeature>-->
    <pFeature>DeviceLinkConnectionCount</pFeature>
    <pFeature>DeviceLinkHeartbeatMode</pFeature>
    <!--<pFeature>DeviceLinkHeartbeatTimeout</pFeature>-->
    <!--<pFeature>DeviceCommandTimeout</pFeature>-->
    <pFeature>DeviceStreamChannelCount</pFeature>
    <pFeature>DeviceStreamChannelSelector</pFeature>
    <pFeature>DeviceStreamChannelType</pFeature>
    <pFeature>DeviceStreamChannelLink</pFeature>
    <pFeature>DeviceStreamChannelEndianness</pFeature>
    <pFeature>DeviceStreamChannelPacketSize</pFeature>
    <pFeature>DeviceEventChannelCount</pFeature>
    <pFeature>DeviceCharacterSet</pFeature>
    <pFeature>DeviceReset</pFeature>
    <pFeature>DeviceTemperatureSelector</pFeature>
    <pFeature>DeviceTemperature</pFeature>
    <pFeature>FindMe</pFeature>
    <pFeature>SensorChannelVersion</pFeature>
    <pFeature>DeviceMaxThroughput</pFeature>
    <pFeature>DevicePJNumber</pFeature>
    <pFeature>CHAP</pFeature>
    <pFeature>HBAbnormalMonitor</pFeature>
    <pFeature>HBVersion</pFeature>
    <!--<pFeature>AlgorithmAbility</pFeature>-->
  </Category>
```

CynthiaLiLi, 2023-10-11:
**arv-tool-0.8 genicam output as flows:**
```xml
Hikrobot-MV-CB060-10GM-S-W-K05153419 (192.168.1.11)
<?xml version="1.0" encoding="UTF-8"?>
<!--===========================================================
Copyright (c) 2014 Hangzhou Hikrobot Technology Co., Ltd. All Rights Reserved..

This software is the confidential and proprietary information of
Hangzhou Hikrobot Technology Co., Ltd..  
You shall not disclose such Confidential Information and shall
use it only in accordance with the terms of the license
agreement you entered into with HIKROBOT.

HIKROBOT MAKES NO REPRESENTATIONS OR WARRANTIES ABOUT THE SUITABILITY 
OF THE SOFTWARE, EITHER EXPRESS OR IMPLIED, INCLUDING BUT NOT 
LIMITED TO THE IMPLIED WARRANTIES OF MERCHANTABILITY, FITNESS
FOR A PARTICULAR PURPOSE, OR NON-INFRINGEMENT. HIKROBOT SHALL NOT BE 
LIABLE FOR ANY DAMAGES SUFFERED BY LICENSEE AS A RESULT OF USING,
MODIFYING OR DISTRIBUTING THIS SOFTWARE OR ITS DERIVATIVES.
===========================================================-->
<RegisterDescription xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns="http://www.genicam.org/GenApi/Version_1_0" xmlns:mstns="http://www.genicam.org/GenApi/Version_1_0" ModelName="HIKCamera" VendorName="HIKROBOT" StandardNameSpace="GEV" SchemaMajorVersion="1" SchemaMinorVersion="0" SchemaSubMinorVersion="0" MajorVersion="1" MinorVersion="2" SubMinorVersion="0" ToolTip="HIKROBOT GigE Vision Camera Interface" ProductGuid="EE4B7E09-DA29-4956-A1EC-212263331CFC" VersionGuid="79ace398-62e3-44a6-8845-f81aaf104382" xsi:schemaLocation="http://www.genicam.org/GenApi/Version_1_0 GenApiSchema_Version_1_0.xsd">
  <Category Name="Root" NameSpace="Standard">
    <pFeature>DeviceControl</pFeature>
    <pFeature>ImageFormatControl</pFeature>
    <pFeature>AcquisitionControl</pFeature>
    <pFeature>AnalogControl</pFeature>
    <pFeature>ColorTransformationControl</pFeature>
    <pFeature>LUTControl</pFeature>
    <pFeature>EncoderControl</pFeature>
    <pFeature>FrequencyConverterControl</pFeature>
    <pFeature>ShadingCorrection</pFeature>
    <pFeature>DigitalIOControl</pFeature>
    <pFeature>ActionControl</pFeature>
    <pFeature>CounterAndTimerControl</pFeature>
    <pFeature>FileAccessControl</pFeature>
    <pFeature>EventControl</pFeature>
    <pFeature>ChunkDataControl</pFeature>
    <!--<pFeature>Debug</pFeature>-->
    <pFeature>TransportLayerControl</pFeature>
    <pFeature>TransferControl</pFeature>
    <pFeature>UserSetControl</pFeature>
  </Category>
  <Category Name="DeviceControl" NameSpace="Standard">
    <ToolTip>Category for device information and control.</ToolTip>
    <Description>Category for device information and control.</Description>
    <DisplayName>Device Control</DisplayName>
    <Visibility>Beginner</Visibility>
    <ImposedAccessMode>RO</ImposedAccessMode>
    <pFeature>DeviceType</pFeature>
    <pFeature>DeviceScanType</pFeature>
    <pFeature>DeviceVendorName</pFeature>
    <pFeature>DeviceModelName</pFeature>
    <pFeature>DeviceManufacturerInfo</pFeature>
    <pFeature>DeviceVersion</pFeature>
    <pFeature>DeviceFirmwareVersion</pFeature>
    <pFeature>DeviceSerialNumber</pFeature>
    <pFeature>DeviceID</pFeature>
    <pFeature>DeviceUserID</pFeature>
    <pFeature>DeviceUptime</pFeature>
    <pFeature>BoardDeviceType</pFeature>
    <pFeature>DeviceConnectionSelector</pFeature>
    <pFeature>DeviceConnectionSpeed</pFeature>
    <pFeature>DeviceConnectionStatus</pFeature>
    <pFeature>DeviceLinkSelector</pFeature>
    <pFeature>DeviceLinkSpeed</pFeature>
    <!--<pFeature>DeviceLinkThroughputLimitMode</pFeature>-->
    <!--<pFeature>DeviceLinkThroughputLimit</pFeature>-->
    <pFeature>DeviceLinkConnectionCount</pFeature>
    <pFeature>DeviceLinkHeartbeatMode</pFeature>
    <!--<pFeature>DeviceLinkHeartbeatTimeout</pFeature>-->
[... 4819 lines cut ...]
      <EnumEntry Name="YUV422Packed" NameSpace="Standard">
        <ToolTip>Pixel format set to YUV 422 Packed</ToolTip>
        <Description>Pixel format set to YUV 422 Packed.</Description>
        <DisplayName>YUV 422 Packed</DisplayName>
        <pIsImplemented>YUV422PackedInq_Reg</pIsImplemented>
        <Value>0x0210001F</Value>
      </EnumEntry>
      <EnumEntry Name="YUV8_UYV" NameSpace="Standard">
        <ToolTip>Pixel format set to YUV 444 Packed</ToolTip>
        <Description>Pixel format set to YUV 444 Packed.</Description>
        <DisplayName>YUV8 UYV</DisplayName>
        <pIsImplemented>YUV444PackedInq_Reg</pIsImplemented>
        <Value>0x02180020</Value>
      </EnumEntry>
      <EnumEntry Name="YUV411_8_UYYVYY" NameSpace="Standard">
        <ToolTip>Pixel format set to YUV411 Packed.</ToolTip>
        <Description>Pixel format set YUV411 Packed.</Description>
        <DisplayName>YUV411 8 UYYVYY</DisplayName>
        <pIsImplemented>YUV411PackedInq_Reg</pIsImplemented>
        <Value>0x020C001E</Value>
      </EnumEntry>
      <EnumEntry Name="BayerGR8" NameSpace="Standard">
        <ToolTip>Pixel format set  to Bayer GR 8</ToolTip>
        <Description>Pixel format set to Bayer GR 8.</Description>
        <DisplayName>Bayer GR 8</DisplayName

CynthiaLiLi, 2023-10-11:
```xml
<IntReg Name="BitRegionSelector_Reg" NameSpace="Custom">
      <pAddress>BitRegionSelector_RegAddr</pAddress>
      <Length>4</Length>
      <AccessMode>RW</AccessMode>
      <pPort>Device</pPort>
      <pInvalidator>PixelFormat</pInvalidator>
      <pInvalidator>UserSetLoad</pInvalidator>
      <Sign>Unsigned</Sign>
      <Endianess>BigEndian</Endianess>
    </IntReg>
    <IntSwissKnife Name="Is_Pixel_Mono8_or_Bayer8">
      <pVariable Name="FMT">PixelFormat_Reg</pVariable>
      <Formula>((FMT = 0x01080001)
             || (FMT = 0x01080008)
             || (FMT = 0x01080009)
             || (FMT = 0x0108000a)
             || (FMT = 0x0108000b)) ? 1:0</Formula>
    </IntSwissKnife>
    <Enumeration Name="ImageCompressionMode" NameSpace="Standard">
      <ToolTip>Enable a specific image compression mode as the base mode for image transfer.</ToolTip>
      <Description>Enable a specific image compression mode as the base mode for image transfer.</Description>
      <DisplayName>Image Compression Mode</DisplayName>
      <pIsImplemented>ImageCompressionMode_Inq</pIsImplemented>
      <pIsAvailable>JPEG_Ctrl_By_Width_Avail</pIsAvailable>
      <pIsLocked>TLParamsLocked</pIsLocked>
      <Streamable>Yes</Streamable>
      <EnumEntry Name="Off" NameSpace="Standard">
        <ToolTip>Image compression is disabled. Images are transmitted uncompressed.</ToolTip>
        <Description>Image compression is disabled. Images are transmitted uncompressed.</Description>
        <DisplayName>Off</DisplayName>
        <Value>0</Value>
      </EnumEntry>
      <EnumEntry Name="JPEG" NameSpace="Standard">
        <ToolTip>JPEG compression is selected.</ToolTip>
        <Description>JPEG compression is selected.</Description>
        <DisplayName>JPEG</DisplayName>
        <pIsImplemented>JPEGMode_Inq</pIsImplemented>
        <Value>1</Value>
      </EnumEntry>
      <EnumEntry Name="HB" NameSpace="Standard">
        <ToolTip>High Bandwidth Mode is selected.</ToolTip>
        <Description>High Bandwidth Mode is selected.</Description>
        <DisplayName>HB</DisplayName>
        <pIsImplemented>HBMode_Inq</pIsImplemented>
        <pIsAvailable>FMT_SUP_HB_Avail</pIsAvailable>
        <Value>2</Value>
      </EnumEntry>
      <pValue>ImageCompressionMode_Reg</pValue>
    </Enumeration>
    <IntReg Name="ImageCompressionMode_Reg" NameSpace="Custom">
      <pAddress>ImageCompressionMode_RegAddr</pAddress>
      <Length>4</Length>
      <AccessMode>RW</AccessMode>
      <pPort>Device</pPort>
      <pInvalidator>UserSetLoad</pInvalidator>
      <pInvalidator>GevGVSPExtendedIDMode</pInvalidator>
      <pInvalidator>PixelFormat</pInvalidator>
      <pInvalidator>WidthMax</pInvalidator>
      <pInvalidator>Width</pInvalidator>
      <Sign>Unsigned</Sign>
      <Endianess>BigEndian</Endianess>
    </IntReg>
    <IntSwissKnife Name="FMT_SUP_HB_Avail" NameSpace="Custom">
      <pVariable Name="FMT">PixelFormat_Reg</pVariable>
      <pVariable Name="RGB8_HB">RGB8HBInq_Reg</pVariable>
      <pVariable Name="MONO8_HB">Mono8HBInq_Reg</pVariable>
      <pVariable Name="MONO10_HB">Mono10HBInq_Reg</pVariable>
      <pVariable Name="MONO10P_HB">Mono10PackedHBInq_Reg</pVariable>
      <pVariable Name="MONO12_HB">Mono12HBInq_Reg</pVariable>
[... 4079 lines cut ...]
      <pAddress>LineTriggerSoftware_RegAddr</pAddress>
      <Length>4</Length>
      <AccessMode>WO</AccessMode>
      <pPort>Device</pPort>
      <pInvalidator>LineMode</pInvalidator>
      <pInvalidator>LineSelector</pInvalidator>
      <pInvalidator>LineSource</pInvalidator>
      <Sign>Unsigned</Sign>
      <Endianess>BigEndian</Endianess>
    </IntReg>
    <IntSwissKnife Name="LineTriggerSoftware_Ava">
      <pVariable Name="SRC">LineSource_Reg</pVariable>
      <pVariable Name="AVA">LineSourceAvailable_Ava</pVariable>
      <Formula>( ((SRC = 5) || (SRC = 8)) &amp;&amp; (AVA = 1)) ? 1 : 0</Formula>
    </IntSwissKnife>
    <Boolean Name="StrobeEnable">
      <ToolTip>Enables the strobe signal to be output to the selected line.</ToolTip>
      <Description>Enables the strobe signal to be output to the selected line.</Description>
      <DisplayName>Strobe Enable</DisplayName>
      <Visibility>Expert</Visibility>
      <pIsImplemented>GPIOImplemented_Int</pIsImplemented>
      <pIsAvailable>StrobeEnabledAvailable_Int</pIsAvailable>
      <ImposedAccessMode>RW</ImposedAccessMode>
      <Streamable>Yes</Streamable>
      <pValue>StrobeLineO

CynthiaLiLi, 2023-10-11:
```xml
<Enumeration Name="CounterSelector" NameSpace="Standard">
      <!-- Recommended element -->
      <ToolTip>Selects which Counter to configure.</ToolTip>
      <Description>Selects which Counter to configure.</Description>
      <DisplayName>Counter Selector</DisplayName>
      <Visibility>Expert</Visibility>
      <pIsImplemented>CounterSelector_Inq</pIsImplemented>
      <ImposedAccessMode>RW</ImposedAccessMode>
      <Streamable>Yes</Streamable>
      <EnumEntry Name="Counter0" NameSpace="Standard">
        <ToolTip>Selects the counter 0.</ToolTip>
        <Description>Selects the counter 0.</Description>
        <DisplayName>Counter 0</DisplayName>
        <pIsImplemented>Counter0_Inq</pIsImplemented>
        <Value>0</Value>
      </EnumEntry>
      <EnumEntry Name="Counter1" NameSpace="Standard">
        <ToolTip>Selects the counter 1.</ToolTip>
        <Description>Selects the counter 1.</Description>
        <DisplayName>Counter 1</DisplayName>
        <pIsImplemented>Counter1_Inq</pIsImplemented>
        <Value>1</Value>
      </EnumEntry>
      <EnumEntry Name="Counter2" NameSpace="Standard">
        <ToolTip>Selects the counter 2.</ToolTip>
        <Description>Selects the counter 2.</Description>
        <DisplayName>Counter 2</DisplayName>
        <pIsImplemented>Counter2_Inq</pIsImplemented>
        <Value>2</Value>
      </EnumEntry>
      <pValue>CounterSelector_Int</pValue>
      <pSelected>CounterEventSource</pSelected>
      <pSelected>CounterResetSource</pSelected>
      <pSelected>CounterReset</pSelected>
      <pSelected>CounterValue</pSelected>
      <pSelected>CounterValueAtReset</pSelected>
    </Enumeration>
    <Enumeration Name="CounterEventSource" NameSpace="Standard">
      <!-- Recommended element -->
      <ToolTip>Select the events that will be the source to increment the Counter.</ToolTip>
      <Description>Select the events that will be the source to increment the Counter.</Description>
      <DisplayName>Counter Event Source</DisplayName>
      <Visibility>Expert</Visibility>
      <pIsImplemented>CounterEventSource_Inq</pIsImplemented>
      <ImposedAccessMode>RW</ImposedAccessMode>
      <Streamable>Yes</Streamable>
      <EnumEntry Name="Off" NameSpace="Standard">
        <ToolTip>Counter is stopped.</ToolTip>
        <Description>Counter is stopped.</Description>
        <DisplayName>Off</DisplayName>
        <pIsImplemented>CounterEventSourceOff_Inq</pIsImplemented>
        <Value>0</Value>
      </EnumEntry>
      <EnumEntry Name="Line0" NameSpace="Standard">
        <ToolTip>Counts the number of transitions on the chosen I/O Line.</ToolTip>
        <Description>Counts the number of transitions on the chosen I/O Line.</Description>
        <DisplayName>Line 0</DisplayName>
        <pIsImplemented>CounterEventSourceLine0_Inq</pIsImplemented>
        <Value>11</Value>
      </EnumEntry>
      <EnumEntry Name="Line1" NameSpace="Standard">
        <ToolTip>Counts the number of transitions on the chosen I/O Line.</ToolTip>
        <Description>Counts the number of transitions on the chosen I/O Line.</Description>
        <DisplayName>Line 1</DisplayName>
        <pIsImplemented>CounterEventSourceLine1_Inq</pIsImplemented>
        <Value>12</Value>
      </EnumEntry>
      <EnumEntry Name="Line2" NameSpace="Standard">
        <ToolTip>Counts the number of transitions on the chosen I/O Line.</ToolTip>
[... 4504 lines cut ...]
        <pMax>FPNCIndexMax_Reg</pMax>
        <pInc>FPNCIndexInc_Reg</pInc>
        <Representation>Linear</Representation>
        <pSelected>FPNCXValue</pSelected>
      </Integer>
      <Integer Name="FPNCXIndex_CtrlValue" NameSpace="Custom">
        <Value>1</Value>
      </Integer>
      <Integer Name="FPNCXValue" NameSpace="Custom">
        <DisplayName>FPNC X Value</DisplayName>
        <Visibility>Guru</Visibility>
        <pIsImplemented>FPNCXSelector_Inq</pIsImplemented>
        <pValue>FPNCXValue_Reg</pValue>
        <pMin>FPNCValueMin_Reg</pMin>
        <pMax>FPNCValueMax_Reg</pMax>
        <pInc>FPNCValueInc_Reg</pInc>
        <Representation>Linear</Representation>
      </Integer>
      <IntSwissKnife Name="FPNCXValue_RegAddrOffSet">
        <pVariable Name="OFFSET">FPNCXIndex</pVariable>
        <pVariable Name="SEL">FPNCXSelector_Int</pVariable>
        <Formula>( OFFSET - 1 ) * 4 + (SEL * 0x8000)</Formula>
      </IntSwissKnife>
      <IntSwissKnife Name="FPNCXValue_BaseRegAddrOffSet">
        <pVariab

CynthiaLiLi, 2023-10-11:
```xml
<Register Name="ClampLevelValueALL" NameSpace="Custom">
      <DisplayName>Clamp Level Value All</DisplayName>
      <Visibility>Guru</Visibility>
      <pAddress>ClampLevelValueALL_RegAddr</pAddress>
      <pLength>ClampLevelValueALL_Len</pLength>
      <AccessMode>RW</AccessMode>
      <pPort>Device</pPort>
    </Register>
    <IntSwissKnife Name="ClampLevelValueALL_Len">
      <pVariable Name="INDEX">UCCIndexMax_Reg</pVariable>
      <Formula>INDEX * 4 * 4</Formula>
    </IntSwissKnife>
    <Register Name="UCCValueALL" NameSpace="Custom">
      <DisplayName>UCC Value All</DisplayName>
      <Visibility>Guru</Visibility>
      <pAddress>UCCValueALL_Reg</pAddress>
      <pLength>UCCValueALL_Len</pLength>
      <AccessMode>RW</AccessMode>
      <pPort>Device</pPort>
    </Register>
    <Integer Name="UCCValueALL_Reg" NameSpace="Custom">
      <pValue>UCCValue_RegAddr</pValue>
    </Integer>
    <IntSwissKnife Name="UCCValueALL_Len">
      <pVariable Name="VALUE_INDEX">UCCValueIndexMax_Reg</pVariable>
      <pVariable Name="UCC_INDEX_MAX">UCCIndexMax_Reg</pVariable>
      <Formula>UCC_INDEX_MAX * VALUE_INDEX * 4 / 2</Formula>
    </IntSwissKnife>
    <Command Name="ClampLevelSave" NameSpace="Custom">
      <ToolTip>Save the clamplevel table into flash</ToolTip>
      <Description>Save the clamplevel table into flash</Description>
      <DisplayName>Clamp Level Save</DisplayName>
      <Visibility>Guru</Visibility>
      <ImposedAccessMode>WO</ImposedAccessMode>
      <pValue>ClampLevelSave_Reg</pValue>
      <CommandValue>1</CommandValue>
    </Command>
    <IntReg Name="ClampLevelSave_Reg">
      <pAddress>ClampLevelSave_RegAddr</pAddress>
      <Length>4</Length>
      <AccessMode>WO</AccessMode>
      <pPort>Device</pPort>
      <Sign>Unsigned</Sign>
      <Endianess>BigEndian</Endianess>
    </IntReg>
    <Command Name="UCCValueSave" NameSpace="Custom">
      <ToolTip>Save the UCC table into flash</ToolTip>
      <Description>Save the UCC table into flash</Description>
      <DisplayName>UCC Value Save</DisplayName>
      <Visibility>Guru</Visibility>
      <ImposedAccessMode>WO</ImposedAccessMode>
      <pValue>UCCValueSave_Reg</pValue>
      <CommandValue>1</CommandValue>
    </Command>
    <IntReg Name="UCCValueSave_Reg">
      <pAddress>UCCValueSave_RegAddr</pAddress>
      <Length>4</Length>
      <AccessMode>WO</AccessMode>
      <pPort>Device</pPort>
      <Sign>Unsigned</Sign>
      <Endianess>BigEndian</Endianess>
    </IntReg>
    <Float Name="VESValue" NameSpace="Custom">
      <DisplayName>VES value</DisplayName>
      <Visibility>Guru</Visibility>
      <pIsImplemented>VESCorrection_Inq</pIsImplemented>
      <pValue>VESValue_Float</pValue>
      <Min>0</Min>
      <Max>10</Max>
[... 4231 lines cut ...]
      <ImposedAccessMode>WO</ImposedAccessMode>
      <pValue>UserSetSave_Reg</pValue>
      <pCommandValue>UserSetSave_CmdValueFao</pCommandValue>
    </Command>
    <IntSwissKnife Name="UserSetSave_CmdValueFao">
      <pVariable Name="VALUE">UserSetSelector_Reg</pVariable>
      <Formula>VALUE</Formula>
    </IntSwissKnife>
    <IntSwissKnife Name="IsDefaultSet">
      <pVariable Name="VAL">UserSetSelector_Reg</pVariable>
      <Formula>(VAL = 0)? 0:1</Formula>
    </IntSwissKnife>
    <IntReg Name="UserSetSave_Reg" NameSpace="Custom">
      <pAddress>UserSetSave_RegAddr</pAddress>
      <Length>4</Length>
      <AccessMode>WO</AccessMode>
      <pPort>Device</pPort>
      <Cachable>WriteAround</Cachable>
      <pInvalidator>UserSetLoad</pInvalidator>
      <Sign>Unsigned</Sign>
      <Endianess>BigEndian</Endianess>
    </IntReg>
    <Enumeration Name="UserSetSelector" NameSpace="Standard">
      <ToolTip>Selects the configuration set to load, save, or configure. Once a configuration set has been selected, all changes to the configuration set settings will be applied to the selected configuration set.</ToolTip>
      <Description>This enumeration selects the configuration set to load, save or configure. P

CynthiaLiLi, 2023-10-11:
```xml
<Enumeration Name="GevCCP" NameSpace="Standard">
      <ToolTip>Controls the device access privilege of an application.</ToolTip>
      <Description>Controls the device access privilege of an application.</Description>
      <DisplayName>GEV CCP</DisplayName>
      <Visibility>Guru</Visibility>
      <ImposedAccessMode>RW</ImposedAccessMode>
      <EnumEntry Name="OpenAccess" NameSpace="Standard">
        <DisplayName>Open Access</DisplayName>
        <Value>0</Value>
      </EnumEntry>
      <EnumEntry Name="ExclusiveAccess" NameSpace="Standard">
        <DisplayName>Exclusive Access</DisplayName>
        <Value>1</Value>
      </EnumEntry>
      <EnumEntry Name="ControlAccess" NameSpace="Standard">
        <DisplayName>Control Access</DisplayName>
        <Value>2</Value>
      </EnumEntry>
      <pValue>GevCCP_Reg</pValue>
    </Enumeration>
    <Enumeration Name="GevSupportedOptionSelector" NameSpace="Standard">
      <ToolTip>Selects the GEV option to interrogate for existing support.</ToolTip>
      <Description>Selects the GEV option to interrogate for existing support.</Description>
      <DisplayName>GEV Supported Option Selector</DisplayName>
      <Visibility>Expert</Visibility>
      <ImposedAccessMode>RW</ImposedAccessMode>
      <EnumEntry Name="UserDefinedName" NameSpace="Standard">
        <DisplayName>User Defined Name</DisplayName>
        <Value>31</Value>
      </EnumEntry>
      <EnumEntry Name="SerialNumber" NameSpace="Standard">
        <DisplayName>Serial Number</DisplayName>
        <Value>30</Value>
      </EnumEntry>
      <EnumEntry Name="HeartbeatDisable" NameSpace="Standard">
        <DisplayName>Heartbeat Disable</DisplayName>
        <Value>29</Value>
      </EnumEntry>
      <EnumEntry Name="LinkSpeed" NameSpace="Standard">
        <DisplayName>Link Speed</DisplayName>
        <Value>28</Value>
      </EnumEntry>
      <EnumEntry Name="CCPApplicationSocket" NameSpace="Standard">
        <DisplayName>CCP Application Socket</DisplayName>
        <Value>27</Value>
      </EnumEntry>
      <EnumEntry Name="ManifestTable" NameSpace="Standard">
        <DisplayName>Manifest Table</DisplayName>
        <Value>26</Value>
      </EnumEntry>
      <EnumEntry Name="TestData" NameSpace="Standard">
        <DisplayName>Test Data</DisplayName>
        <Value>25</Value>
      </EnumEntry>
      <EnumEntry Name="DiscoveryAckDelay" NameSpace="Standard">
        <DisplayName>Discovery Ack Delay</DisplayName>
        <Value>24</Value>
      </EnumEntry>
      <EnumEntry Name="DiscoveryAckDelayWritable" NameSpace="Standard">
        <DisplayName>Discovery Ack Delay Writable</DisplayName>
        <Value>23</Value>
      </EnumEntry>
      <EnumEntry Name="ExtendedStatusCodes" NameSpace="Standard">
        <DisplayName>Extended Status Codes</DisplayName>
        <Value>22</Value>
      </EnumEntry>
      <EnumEntry Name="PrimaryApplicationSwitchover" NameSpace="Standard">
        <DisplayName>Primary Application Switchover (PAS) </DisplayName>
        <Value>21</Value>
[... 2583 lines cut ...]
      </IntReg>
      <IntReg Name="SaturationAbsVal_Reg">
        <pAddress>SaturationAbsVal_RegAddr</pAddress>
        <Length>4</Length>
        <AccessMode>RW</AccessMode>
        <pPort>Device</pPort>
        <pInvalidator>UserSetLoad</pInvalidator>
        <pInvalidator>ColorTransformationEnable</pInvalidator>
        <Sign>Unsigned</Sign>
        <Endianess>BigEndian</Endianess>
      </IntReg>
      <IntReg Name="SaturationCtrl_Reg">
        <pAddress>SaturationCtrl_RegAddr</pAddress>
        <Length>4</Length>
        <AccessMode>RW</AccessMode>
        <pPort>Device</pPort>
        <pInvalidator>UserSetLoad</pInvalidator>
        <pInvalidator>ColorTransformationEnable</pInvalidator>
        <Sign>Unsigned</Sign>
        <Endianess>BigEndian</Endianess>
      </IntReg>
    </Group>
  </Group>
</RegisterDescription>
```

CynthiaLiLi, 2023-10-11:
> Please attach the Genicam data obtained with `arv-tool-0.8 genicam`.

Hi, Emmanuel.
The output of `arv-tool-0.8 genicam` has been submitted to comments.

EmmanuelP, 2023-10-11:
Hi Cynthia,

Thanks a lot for the data.

```
    <Integer Name="Width" NameSpace="Standard">
      <ToolTip>Width of the image provided by the device (in pixels).</ToolTip>
      <Description>Width of the image provided by the device (in pixels).</Description>
      <DisplayName>Width</DisplayName>
      <Visibility>Beginner</Visibility>
      <pIsLocked>KEYI_TLParamsLocked</pIsLocked>
      <ImposedAccessMode>RW</ImposedAccessMode>
      <Streamable>Yes</Streamable>
      <pValue>Width_Reg</pValue>
      <pMin>WidthMin_Int</pMin>
      <pMax>ROIWidthMax_Int</pMax>
      <pInc>WidthInc_Int</pInc>
      <Representation>Linear</Representation>
    </Integer>
```

The problem comes from the fact IsLocked is currently not checked in the register access code. This need to be implemented in arvgcregisternode.c.

The next time, please don't paste the required data in comments, attached them as file. It makes the issue report easier to read.

CynthiaLiLi, 2023-10-11:
> Hi Cynthia,
> 
> Thanks a lot for the data.
> 
> ```
>     <Integer Name="Width" NameSpace="Standard">
>       <ToolTip>Width of the image provided by the device (in pixels).</ToolTip>
>       <Description>Width of the image provided by the device (in pixels).</Description>
>       <DisplayName>Width</DisplayName>
>       <Visibility>Beginner</Visibility>
>       <pIsLocked>KEYI_TLParamsLocked</pIsLocked>
>       <ImposedAccessMode>RW</ImposedAccessMode>
>       <Streamable>Yes</Streamable>
>       <pValue>Width_Reg</pValue>
>       <pMin>WidthMin_Int</pMin>
>       <pMax>ROIWidthMax_Int</pMax>
>       <pInc>WidthInc_Int</pInc>
>       <Representation>Linear</Representation>
>     </Integer>
> ```
> 
> The problem comes from the fact `pIsLocked` is currently not checked in the features access code. This need to be implemented in `arvgcfeaturenodeprivate.h`.
> 
> The next time, please don't paste the required data in comments, attached them as file. It makes the issue report easier to read.

I'm glad to be able to contribute a little to Aravis.
Regarding the issue of attached files, I will definitely pay attention next time.