AravisProject/aravis #827: Aravis can't find Gige Camera with Ethernet port in centos 8 Linux

PeiyuQuan, 2023-10-10:
I installed Aravis at Centos 8 linux computer. I can ping camera IP address, But arv-tool-0.8 can't find camera. 
Could you help me fix this issue?

$ ping 192.168.0.1
PING 192.168.0.1 (192.168.0.1) 56(84) bytes of data.
64 bytes from 192.168.0.1: icmp_seq=1 ttl=255 time=0.546 ms
64 bytes from 192.168.0.1: icmp_seq=2 ttl=255 time=0.522 ms
64 bytes from 192.168.0.1: icmp_seq=3 ttl=255 time=0.482 ms
64 bytes from 192.168.0.1: icmp_seq=4 ttl=255 time=0.525 ms
64 bytes from 192.168.0.1: icmp_seq=5 ttl=255 time=0.639 ms

./arv-camera-test --debug=all:3
Looking for the first available camera
Found 0 USB3Vision device (among 12 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 134.79.37.160
[GvDiscoverSocket::new] Add interface 192.168.10.6
[GvDiscoverSocket::new] Add interface 192.168.122.1
No camera found: No supported device found

I have installed the main database for it, and use command "ninja reconfigure"
$ ldd arv-tool-0.8
	linux-vdso.so.1 (0x00007ffe27c2e000)
	libaravis-0.8.so.0 => /usr/local/aravis/build/src/./libaravis-0.8.so.0 (0x00007f5625252000)
	libglib-2.0.so.0 => /lib64/libglib-2.0.so.0 (0x00007f5624f38000)
	libgobject-2.0.so.0 => /lib64/libgobject-2.0.so.0 (0x00007f5624ce5000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f5624920000)
	libgio-2.0.so.0 => /lib64/libgio-2.0.so.0 (0x00007f5624576000)
	libxml2.so.2 => /lib64/libxml2.so.2 (0x00007f562420e000)
	libz.so.1 => /lib64/libz.so.1 (0x00007f5623ff6000)
	libm.so.6 => /lib64/libm.so.6 (0x00007f5623c74000)
	libusb-1.0.so.0 => /lib64/libusb-1.0.so.0 (0x00007f5623a59000)
	libgnutls.so.30 => /lib64/libgnutls.so.30 (0x00007f5623668000)
	libpcre.so.1 => /lib64/libpcre.so.1 (0x00007f56233f7000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f56231d7000)
	libffi.so.6 => /lib64/libffi.so.6 (0x00007f5622fce000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f56254dd000)
	libgmodule-2.0.so.0 => /lib64/libgmodule-2.0.so.0 (0x00007f5622dca000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007f5622bc6000)
	libselinux.so.1 => /lib64/libselinux.so.1 (0x00007f562299b000)
	libresolv.so.2 => /lib64/libresolv.so.2 (0x00007f5622783000)
	libmount.so.1 => /lib64/libmount.so.1 (0x00007f5622529000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f5622311000)
	liblzma.so.5 => /lib64/liblzma.so.5 (0x00007f56220ea000)
	libudev.so.1 => /lib64/libudev.so.1 (0x00007f5621e4e000)
	libp11-kit.so.0 => /lib64/libp11-kit.so.0 (0x00007f5621b24000)
	libidn2.so.0 => /lib64/libidn2.so.0 (0x00007f5621906000)
	libunistring.so.2 => /lib64/libunistring.so.2 (0x00007f5621585000)
	libtasn1.so.6 => /lib64/libtasn1.so.6 (0x00007f5621372000)
	libnettle.so.6 => /lib64/libnettle.so.6 (0x00007f5621138000)
	libhogweed.so.4 => /lib64/libhogweed.so.4 (0x00007f5620f08000)
	libgmp.so.10 => /lib64/libgmp.so.10 (0x00007f5620c70000)
	libpcre2-8.so.0 => /lib64/libpcre2-8.so.0 (0x00007f56209ec000)
	libblkid.so.1 => /lib64/libblkid.so.1 (0x00007f5620799000)
	libuuid.so.1 => /lib64/libuuid.so.1 (0x00007f5620591000)
	librt.so.1 => /lib64/librt.so.1 (0x00007f5620389000)

$ sudo ninja reconfigure
[0/1] Regenerating build files.
The Meson build system
Version: 0.58.2
Source dir: /usr/local/aravis
Build dir: /usr/local/aravis/build
Build type: native build
Project name: aravis
Project version: 0.8.1
C compiler for the host machine: cc (gcc 8.5.0 "cc (GCC) 8.5.0 20210514 (Red Hat 8.5.0-20)")
C linker for the host machine: cc ld.bfd 2.30-123
C++ compiler for the host machine: c++ (gcc 8.5.0 "c++ (GCC) 8.5.0 20210514 (Red Hat 8.5.0-20)")
C++ linker for the host machine: c++ ld.bfd 2.30-123
Host machine cpu family: x86_64
Host machine cpu: x86_64
Compiler for C supports arguments -Wduplicated-branches: YES (cached)
Compiler for C supports arguments -Wimplicit-fallthrough: YES (cached)
Compiler for C supports arguments -Wmisleading-indentation: YES (cached)
Compiler for C supports arguments -Wstrict-prototypes: YES (cached)
Compiler for C supports arguments -Wunused: YES (cached)
Compiler for C supports arguments -Wdeclaration-after-statement: YES (cached)
Compiler for C supports arguments -Wformat=2: YES (cached)
Compiler for C supports arguments -Wimplicit-function-declaration: YES (cached)
Compiler for C supports arguments -Winit-self: YES (cached)
Compiler for C supports arguments -Wmissing-include-dirs: YES (cached)
Compiler for C supports arguments -Wmissing-prototypes: YES (cached)
Compiler for C supports arguments -Wpointer-arith: YES (cached)
Dependency glib-2.0 found: YES 2.56.4 (cached)
Dependency gobject-2.0 found: YES 2.56.4 (cached)
Dependency gio-2.0 found: YES 2.56.4 (cached)
Dependency libxml-2.0 found: YES 2.9.7 (cached)
Dependency zlib found: YES 1.2.11 (cached)
Dependency libusb-1.0 found: YES 1.0.23 (cached)
Library m found: YES
Has header "linux/if_packet.h" : YES (cached)
Program glib-mkenums found: YES (/bin/glib-mkenums)
Program glib-mkenums found: YES (/bin/glib-mkenums)
Program glib-mkenums found: YES (/bin/glib-mkenums)
Program glib-mkenums found: YES (/bin/glib-mkenums)
Configuring arvfeatures.h using configuration
Configuring arvversion.h using configuration
Program g-ir-scanner found: YES (/bin/g-ir-scanner)
Dependency gobject-introspection-1.0 found: YES 1.56.1 (cached)
Dependency gobject-introspection-1.0 found: YES 1.56.1 (cached)
Dependency gobject-introspection-1.0 found: YES 1.56.1 (cached)
Program g-ir-scanner found: YES (/usr/bin/g-ir-scanner)
Dependency gobject-introspection-1.0 found: YES 1.56.1 (cached)
Program g-ir-compiler found: YES (/usr/bin/g-ir-compiler)
Program python3 found: YES (/bin/python3)
Configuring pylaunch using configuration
Configuring pylaunch-dbg using configuration
Configuring jslaunch using configuration
Configuring jslaunch-dbg using configuration
Dependency gtk+-3.0 found: YES 3.22.30 (cached)
Dependency gstreamer-base-1.0 found: YES 1.16.1 (cached)
Dependency gstreamer-app-1.0 found: YES 1.16.1 (cached)
Dependency gstreamer-video-1.0 found: YES 1.16.1 (cached)
Dependency libnotify found: YES 0.7.7 (cached)
Found pkg-config: /bin/pkg-config (1.4.2)
Program glib-compile-resources found: YES (/bin/glib-compile-resources)
Configuring arv-viewer-0.8.appdata.xml using configuration
Configuring arv-viewer.desktop.in using configuration
Dependency gstreamer-base-1.0 found: YES 1.16.1 (cached)
Dependency gstreamer-app-1.0 found: YES 1.16.1 (cached)
Configuring gst-aravis-launch using configuration
Configuring gst-aravis-inspect using configuration
Program gtkdoc-scan found: YES (/bin/gtkdoc-scan)
Program xsltproc found: YES (/bin/xsltproc)
Configuring gtkdocentities.ent using configuration
Program gtkdoc-scan found: YES (/bin/gtkdoc-scan)
Program gtkdoc-scangobj found: YES (/bin/gtkdoc-scangobj)
Program gtkdoc-mkdb found: YES (/bin/gtkdoc-mkdb)
Program gtkdoc-mkhtml found: YES (/bin/gtkdoc-mkhtml)
Program gtkdoc-fixxref found: YES (/bin/gtkdoc-fixxref)
Build targets in project: 41

Found ninja-1.8.2 at /bin/ninja

EmmanuelP, 2023-10-11:
```
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 134.79.37.160
[GvDiscoverSocket::new] Add interface 192.168.10.6
[GvDiscoverSocket::new] Add interface 192.168.122.1
```

Your camera IP address is 192.168.0.1. Does one of your interface include this address in its mask ?
Are you running a firewall on your machine ?

PeiyuQuan, 2023-10-11:
Hello, correct. The Issue is from firewall. Do you know which firewall ports is for arv-tool? It is a lab computer; we should keep turn on firewall. But I can't disable arv-tool's firewall port.
Thank you

EmmanuelP, 2023-10-12:
The port used by arv-tool is dynamically allocated. The protocol is based on udp. I can not help about firewall configuration. I just know aravis works fine with the default rule of fedora firewall.

EmmanuelP, 2023-10-21:
Not an Aravis issue, I'm closing it.

Thanks.

closed 2023-10-21 by EmmanuelP