AravisProject/aravis #829: arv_dom_document_get_url is always NULL

eudoxos, 2023-10-15:
**Describe the bug**
There is a function to retrieve URL of the genicam XML: `arv_dom_document_get_url`; it returns `ArvDomDocumentPrivate`'s `url` attribute, assigned in `arv_dom_document_new_from_url`.

This codepath is not currently used at all, though. Instead, `arv_gc_new` receives already loaded XML; the loading happens in `_load_genicam` which only returns the XML content but does not store its URL (obtained with `arv_gv_device_read_memory` at `ARV_GVBS_XML_URL_SIZE`). That is why `arv_dom_document_get_url` return NULL (as far as I understand).

**Expected behavior**
`arv_dom_document_get_url` returns URL (such as `Local:genicam.xml;123;456`, 123 being address and 456 size) which can be parsed and used to retrieve the XML from memory again (for GigeV, USB and Fake cameras).

**Platform description:**
 - Aravis version: git
 - OS: Mint 21.1 
 - Hardware: x86_64

**Additional context**
This is needed for the GenTL provider so that it does not duplicate code from other parts of Aravis. GenTL consumers download and interpret the XML themselves, and use URL to request them via memory access.

EmmanuelP, 2023-11-03:
Please try #835.

eudoxos, 2023-11-04:
Excellent, really helped!

PR #835 (2023-11-05): genicam: store genicam URL in ArvDomDocument URL property
Fixes #829
commit 76819cb4 (2023-11-05): genicam: store genicam URL in ArvDomDocument URL property

Fixes #829
closed 2023-11-05 by EmmanuelP