AravisProject/aravis #830: Aravis failing to stream from Daheng Imaging 4k camera on Windows 10

TAldhous, 2023-10-16:
I would like to stream video from two Daheng Imaging 4k cameras using the Aravis Gstreamer plugin on Windows 10. One camera is a GIGE network camera model MER2-630-18GC-P and the other is a USB3 camera model ME2P-900-43U3C. Both work with Galaxy Viewer, the sample application provided by Daheng Imaging.

I have built Aravis 0.8.29 as described in the documentation, and tried connecting to the cameras using the Aravis viewer.

The USB3 camera is not detected by the Aravis viewer at all.

The GIGE camera is detected by the Aravis viewer, but when selected, it fails to show any video.

I also tried running arv-camera-test. The output is as follows:

    $ arv-camera-test-0.8.exe -d all
    Looking for the first available camera
    [23:31:31.959] ­ƒà© interface> Found 0 USB3Vision device (among 4 USB devices)
    [23:31:31.967] ­ƒà© interface> [GvDiscoverSocket::new] Add interface xxx.xxx.xxx.xxx (xxx.xxx.xxx.xxx)
    [23:31:31.970] ­ƒà© interface> [GvDiscoverSocket::new] Add interface xxx.xxx.xxx.xxx (xxx.xxx.xxx.xxx)
    [23:31:31.970] ­ƒà© interface> [GvDiscoverSocket::new] Add interface xxx.xxx.xxx.xxx (xxx.xxx.xxx.xxx)
    [23:31:31.971] ­ƒà© interface> [GvDiscoverSocket::new] Add interface xxx.xxx.xxx.xxx (xxx.xxx.xxx.xxx)
    [23:31:31.973] ­ƒà© interface> [GvInterface::discovery] Device 'Daheng Imaging-MER2-630-18GC-P-XXXXX
    XXXXXX' found (interface xxx.xxx.xxx.xxx) user_id 'CY1000-1-1' - MAC 'xx:xx:xx:xx:xx:xx'
    [23:31:31.974] ­ƒà© device> [GvDevice::new] Interface address = xxx.xxx.xxx.xxx
    [23:31:31.974] ­ƒà© device> [GvDevice::new] Device address = xxx.xxx.xxx.xxx
    [23:31:31.978] ­ƒà© device> [GvDevice::load_genicam] xml url = 'Local:MER2_GIGE_V2.5.4.zip;41a30000;
    8810' at 0x200
    [23:31:31.979] ­ƒà© device> [GvDevice::load_genicam] Xml address = 0x41a30000 - size = 0x8810 - MER2
    _GIGE_V2.5.4.zip
    [23:31:32.136] ­ƒà© device> [GvDevice::load_genicam] Zipped xml data
    [23:31:32.174] ­ƒà© dom> [Genicam::create_element] Unknown tag (SwapEndianess)
    [23:31:32.175] ­ƒà© genicam> [Gc::set_default_node_data] Add 'DeviceManufacturerInfo'
    [23:31:32.176] ­ƒà© genicam> [Gc::set_default_node_data] Add 'DeviceID'
    [23:31:32.177] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationLLA'
    [23:31:32.178] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationDHCP'
    [23:31:32.179] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationPersistent
    IP'
    [23:31:32.180] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPAddress'
    [23:31:32.180] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentSubnetMask'
    [23:31:32.181] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentDefaultGateway'
    [23:31:32.181] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevNumberOfNetworkInterfaces'
    [23:31:32.182] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentIPAddress'
    [23:31:32.182] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentSubnetMask'
    [23:31:32.183] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentDefaultGateway'
    [23:31:32.183] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelCount'
    [23:31:32.184] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelSelector'
    [23:31:32.184] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevSCPAddrCalc'
    [23:31:32.185] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevTimestampTickFrequency'
    [23:31:32.185] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevSCPHostPort'
    [23:31:32.185] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSFireTestPacket'
    [23:31:32.186] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSDoNotFragment'
    [23:31:32.186] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSBigEndian'
    [23:31:32.187] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSPacketSize'
    [23:31:32.188] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevSCPD'
    [23:31:32.188] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevSCDA'
    [23:31:32.189] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevSCSP'
    [23:31:32.190] ­ƒà© genicam> [Gc::set_default_node_data] Add 'ArvGevSCCFGMultipart'
    [23:31:32.198] ­ƒà© device> [GvDevice::new] Device endianness = big
    [23:31:32.199] ­ƒà© device> [GvDevice::new] Packet resend     = yes
    [23:31:32.199] ­ƒà© device> [GvDevice::new] Write memory      = yes
    [23:31:32.200] ­ƒà© device> [GvDevice::new] Legacy endianness handling = yes
    vendor name            = Daheng Imaging
    model name             = MER2-630-18GC-P
    device serial number   = XXXXXXXXXXX
    image width            = 3088
    image height           = 2064
    horizontal binning     = 1
    vertical binning       = 1
    exposure               = 50000 ┬Ás
    gain                   = 0 dB
    payload                = 6373632 bytes
    gv n_stream channels   = 1
    gv current channel     = 0
    gv packet delay        = 0 ns
[... 47 lines cut ...]
                             counter     :           0:           0:           0
    [23:31:45.322] ­ƒà© stream> [GvStream::finalize] n_completed_buffers    = 0
    [23:31:45.322] ­ƒà© stream> [GvStream::finalize] n_failures             = 0
    [23:31:45.322] ­ƒà© stream> [GvStream::finalize] n_underruns            = 0
    [23:31:45.322] ­ƒà© stream> [GvStream::finalize] n_timeouts             = 0
    [23:31:45.323] ­ƒà© stream> [GvStream::finalize] n_aborted              = 0
    [23:31:45.323] ­ƒà© stream> [GvStream::finalize] n_missing_frames       = 0
    [23:31:45.323] ­ƒà© stream> [GvStream::finalize] n_size_mismatch_errors = 0
    [23:31:45.324] ­ƒà© stream> [GvStream::finalize] n_received_packets     = 0
    [23:31:45.324] ­ƒà© stream> [GvStream::finalize] n_missing_packets      = 0
    [23:31:45.324] ­ƒà© stream> [GvStream::finalize] n_error_packets        = 0
    [23:31:45.324] ­ƒà© stream> [GvStream::finalize] n_ignored_packets      = 0
    [23:31:45.325] ­ƒà© stream> [GvStream::finalize] n_resend_requests      = 0
    [23:31:45.325] ­ƒà© stream> [GvStream::finalize] n_resent_packets       = 0
    [23:31:45.325] ­ƒà© stream> [GvStream::finalize] n_resend_ratio_reached = 0
    [23:31:45.326] ­ƒà© stream> [GvStream::finalize] n_resend_disabled      = 0
    [23:31:45.326] ­ƒà© stream> [GvStream::finalize] n_duplicated_packets   = 0
    [23:31:45.326] ­ƒà© stream> [GvStream::finalize] n_transferred_bytes    = 0
    [23:31:45.326] ­ƒà© stream> [GvStream::finalize] n_ignored_bytes        = 0
    [23:31:45.327] ­ƒà© stream> [Stream::finalize] Flush 50 buffer[s] in input queue
    [23:31:45.327] ­ƒà© stream> [Stream::finalize] Flush 0 buffer[s] in output queue

Do you have any suggestions on how I can get the cameras to work with Aravis?

Many Thanks

EmmanuelP, 2023-11-02:
The GigEvision camera does not receive any stream packet at all. Do you have a firewall that would block the stream packets ? Or an exotic network configuration ?

Did you check the cameras are working using linux ?

For the USB camera, aravis relies on libusb which seems to be tricky to set up under windows. Please have a look at aravis discourse instance, I recall some discussions about it happening there.

TAldhous, 2023-11-08:
Both cameras work on Windows with GalaxyViewer, the example viewer provided by Daheng Imaging.

We get the same problem on two different PC's in different environments - one is an office machine, the other is a home laptop. I have tested it with the firewall turned off on the home laptop, but will test it again just to make sure. I have not tested it on a linux machine, but will do so and report my findings.

I will investigate the setup of libusb to see if that helps to solve the issue.

Thanks for responding!

TAldhous, 2023-11-08:
I have retested, and the firewall was definitely part of the problem. Allowing arv-camera-test-0.8.exe through the firewall on the laptop at home connected via wifi, gives this:

```
$ arv-camera-test-0.8.exe
Looking for the first available camera
vendor name            = Daheng Imaging
model name             = MER2-630-18GC-P
device serial number   = XXXXXXXXXXX
image width            = 3088
image height           = 2064
horizontal binning     = 1
vertical binning       = 1
exposure               = 2500 ┬Ás
gain                   = 0 dB
payload                = 6373632 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 0 ns
gv packet size         = 1500 bytes
  0 frame/s  -       0 MiB/s - 8 errors
  0 frame/s  -       0 MiB/s - 16 errors
  0 frame/s  -       0 MiB/s - 16 errors
  0 frame/s  -       0 MiB/s - 18 errors
  0 frame/s  -       0 MiB/s - 18 errors
  0 frame/s  -       0 MiB/s - 16 errors
  0 frame/s  -       0 MiB/s - 17 errors
  0 frame/s  -       0 MiB/s - 17 errors
  0 frame/s  -       0 MiB/s - 17 errors
  0 frame/s  -       0 MiB/s - 16 errors
Control lost
  0 frame/s  -       0 MiB/s - 18 errors
n_completed_buffers    = 0
n_failures             = 177
n_underruns            = 0
n_timeouts             = 177
n_aborted              = 0
n_missing_frames       = 0
n_size_mismatch_errors = 0
n_received_packets     = 16575
n_missing_packets      = 769922
n_error_packets        = 0
n_ignored_packets      = 2185
n_resend_requests      = 213887
n_resent_packets       = 0
n_resend_ratio_reached = 173
n_resend_disabled      = 0
n_duplicated_packets   = 0
n_transferred_bytes    = 21124308
n_ignored_bytes        = 17480

```
Changing to a wired connection gives this:

```
$  arv-camera-test-0.8.exe
Looking for the first available camera
vendor name            = Daheng Imaging
model name             = MER2-630-18GC-P
device serial number   = FBJ22120380
image width            = 3088
image height           = 2064
horizontal binning     = 1
vertical binning       = 1
exposure               = 2500 ┬Ás
gain                   = 0 dB
payload                = 6373632 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 0 ns
gv packet size         = 1500 bytes
 15 frames/s -    95.6 MiB/s
 16 frames/s -     102 MiB/s
 18 frames/s -     115 MiB/s
 17 frames/s -     108 MiB/s
 16 frames/s -     102 MiB/s
 16 frames/s -     102 MiB/s
 17 frames/s -     108 MiB/s
 17 frames/s -     108 MiB/s
 17 frames/s -     108 MiB/s
 16 frames/s -     102 MiB/s
 16 frames/s -     102 MiB/s
 17 frames/s -     108 MiB/s
 16 frames/s -     102 MiB/s
 18 frames/s -     115 MiB/s
 16 frames/s -     102 MiB/s
 17 frames/s -     108 MiB/s
n_completed_buffers    = 267
n_failures             = 0
n_underruns            = 0
n_timeouts             = 0
n_aborted              = 0
n_missing_frames       = 0
n_size_mismatch_errors = 0
n_received_packets     = 1166474
n_missing_packets      = 0
n_error_packets        = 0
n_ignored_packets      = 0
n_resend_requests      = 7557
n_resent_packets       = 6233
n_resend_ratio_reached = 0
n_resend_disabled      = 0
n_duplicated_packets   = 0
n_transferred_bytes    = 1716316272
n_ignored_bytes        = 0

```
a huge improvement!

However, arv-viewer-0.8.exe does not work. It detects the camera
[image]
but when the play button is clicked, it just shows a blank window:
[image]
which eventually just becomes a frame:
[image]

Do you have any suggestions as to what is failing, and how I might fix it.

Many thanks for responding.

VisionShilin, 2025-02-19:
Did you find a solution for this issue?
I met the same problem too.

TAldhous, 2025-02-19:
Hi, I'm afraid we never got this to work.

Regards

Tony Aldhous

EmmanuelP, 2025-02-20:
> However, arv-viewer-0.8.exe does not work. It detects the camera ! but when the play button is clicked, it just shows a blank window

> Do you have any suggestions as to what is failing, and how I might fix it.

Could you attach the console output of `arv-viewer-0.8 -d all` ?

VisionShilin, 2025-02-20:
It works fine after I disable the firewall or add the program into the exception list. The firewall of Windows did cause the problem I think.
The arv-viewer-0.8.exe still didn't work while the gst-launch-1.0.exe aravissrc works fine.