AravisProject/aravis #836: ARAVIS Compatiblity with SONY-XCG-CP510

AmmarkoV, 2023-11-03:
Hello everyone,
I am trying to interface a SONY-XCG-CP510 camera with ARAVIS
https://www.image-sensing-solutions.eu/XCG-CP510.html
The camera is picked up by arv-tool and arv-viewer, unfortunately does not work streaming images.. :(

Testing it with : arv-test-0.8 
```
Found 1 device
Testing 'SONY:XCG-CP510'
Genicam:Load                        SUCCESS 
Genicam:Schema                      SUCCESS Schema validation disabled
Properties:SensorSizeReadout        FAILURE [SensorWidth] [GEV_BOOTSTRAP_SensorWidth] GigEVision read_memory error (invalid-access)
Properties:SensorSizeCheck          IGNORED 
Properties:GainAvailable            SUCCESS 
Properties:GainReadout              FAILURE [Gain] [GainValueConverter] [GainSelectorReg] GigEVision read_memory error (invalid-access)
Properties:ExposureTimeAvailable    SUCCESS 
Properties:ExposureTimeReadout      FAILURE [ExposureTime] [ExposureTimeConverter] [ExposureTimeConvertCoeffReg] [ExposureTimeConvertCoeffReg] GigEVision read_memory error (invalid-access)
MultipleAcquisitionA:BufferCheck    FAILURE [AcquisitionMode] [AcquisitionMode] [GEV_BOOTSTRAP_AcquisitionMode] GigEVision write_memory error (invalid-access)
SingleAcquisition:BufferCheck       FAILURE [ArvGevSCPSPacketSize] [ArvGevSCPSPacketSizeReg] GigEVision write_memory error (invalid-access)
SoftwareTrigger:BufferCheck         FAILURE [AcquisitionMode] [AcquisitionMode] [GEV_BOOTSTRAP_AcquisitionMode] GigEVision write_memory error (invalid-access)
MultipleAcquisitionB:BufferCheck    FAILURE [AcquisitionMode] [AcquisitionMode] [GEV_BOOTSTRAP_AcquisitionMode] GigEVision write_memory error (invalid-access)
Multipart:NoSupport                 SUCCESS 
Chunks:BufferCheck                  FAILURE [ChunkModeActive] [GEV_BOOTSTRAP_ChunkModeActive] GigEVision write_memory error (invalid-access)
GigEVision:NNetworkInterfaces       SUCCESS 1
GigEVision:NStreamChannels          SUCCESS 1
GigEVision:StreamChannel            SUCCESS 
```

Giving  arv-camera-test-0.8 -d all --duration 1
```
Looking for the first available camera
[15:28:55.009] 🅸 interface> Found 0 USB3Vision device (among 11 USB devices)
[15:28:55.009] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[15:28:55.009] 🅸 interface> [GvDiscoverSocket::new] Add interface 139.91.185.16 (139.91.185.255)
[15:28:55.009] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.17.0.1 (172.17.255.255)
[15:28:55.014] 🅸 interface> [GvInterface::discovery] Device 'SONY-XCG-CP510-3205040' found (interface 139.91.185.16) user_id '' - MAC 'f8:4e:17:e9:25:e9'
[15:28:55.014] 🅸 device> [GvDevice::new] Interface address = 139.91.185.16
[15:28:55.014] 🅸 device> [GvDevice::new] Device address = 139.91.185.46
[15:28:55.023] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:XCG-CPSeries_v_1_1_7.zip;400000;90ec' at 0x200
[15:28:55.023] 🅸 device> [GvDevice::load_genicam] Xml address = 0x400000 - size = 0x90ec - XCG-CPSeries_v_1_1_7.zip
[15:28:56.362] 🅸 device> [GvDevice::load_genicam] Zipped xml data
[15:28:56.407] 🅸 dom> [Genicam::create_element] Unknown tag (SwapEndianess)
[15:28:56.407] 🅸 genicam> [Gc::set_default_node_data] Add 'DeviceID'
[15:28:56.407] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationLLA'
[15:28:56.407] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationDHCP'
[15:28:56.408] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationPersistentIP'
[15:28:56.408] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPAddress'
[15:28:56.408] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentSubnetMask'
[15:28:56.408] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentDefaultGateway'
[15:28:56.408] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevNumberOfNetworkInterfaces'
[15:28:56.408] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentIPAddress'
[15:28:56.408] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentSubnetMask'
[15:28:56.408] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentDefaultGateway'
[15:28:56.408] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelCount'
[15:28:56.408] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelSelector'
[15:28:56.408] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPAddrCalc'
[15:28:56.408] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevTimestampTickFrequency'
[15:28:56.408] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPHostPort'
[15:28:56.409] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSFireTestPacket'
[15:28:56.409] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSDoNotFragment'
[15:28:56.409] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSBigEndian'
[15:28:56.409] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSPacketSize'
[15:28:56.409] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPD'
[15:28:56.409] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCDA'
[15:28:56.409] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCSP'
[15:28:56.409] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCCFGMultipart'
[15:28:56.417] 🅸 device> [GvDevice::new] Device endianness = big
[15:28:56.417] 🅸 device> [GvDevice::new] Packet resend     = yes
[15:28:56.417] 🅸 device> [GvDevice::new] Write memory      = yes
[15:28:56.417] 🅸 device> [GvDevice::new] Legacy endianness handling = no
Failed to configure the device: [ChunkModeActive] [GEV_BOOTSTRAP_ChunkModeActive] GigEVision write_memory error (invalid-access)
```

**Describe the solution you'd like**
I would like the camera to work with linux

**Describe alternatives you've considered**
Using the official SDK on windows 

**Additional context**
Any ideas or mods to aravis to receive raw images from the sensor ?

EmmanuelP, 2023-11-03:
Probably another instance of a camera needing the legacy endianness mechanism. Please try to add your device to this list:

https://github.com/AravisProject/aravis/blob/ba0d728d0952dd8258ddfefba523d11d8f9b8510/src/arvgcport.c#L62-L72

AmmarkoV, 2023-11-03:
@EmmanuelP  Thank you very much!
You were right!

Adding : 
   { .vendor_selection = "Sony",                        .model_selection = "XCG_CPSeries"},

Fixed the issue!
I am closing the issue as resolved!

closed 2023-11-03 by AmmarkoV