AravisProject/aravis #844: aravis python assertion errors

nagyilles, 2023-11-22:
I am importing the package as follows:

from gi.repository import Aravis
self.cam = Aravis.Camera.new(name)

The camera opened successfully (Baser acA1920-155um USB3) :

camera name:  acA1920-155um                                                                                                                                                                                                                   
[12:23:48.714] 🅸 stream> SIRM_INFO             = 0x02000000                                                                                                                                                                                   
[12:23:48.714] 🅸 stream> SIRM_REQ_PAYLOAD_SIZE = 0x0000000000232800                                                                                                                                                                           
[12:23:48.714] 🅸 stream> SIRM_REQ_LEADER_SIZE  = 0x00000400                                                                                                                                                                                   
[12:23:48.714] 🅸 stream> SIRM_REQ_TRAILER_SIZE = 0x00000400                                                                                                                                                                                   
[12:23:48.714] 🅸 stream> Required alignment    = 4                                                                                                                                                                                            
[12:23:48.715] 🅸 stream> SIRM_PAYLOAD_SIZE     = 0x00100000                                                                                                                                                                                   
[12:23:48.715] 🅸 stream> SIRM_PAYLOAD_COUNT    = 0x00000002                                                                                                                                                                                   
[12:23:48.716] 🅸 stream> SIRM_TRANSFER1_SIZE   = 0x00032800                                                                                                                                                                                   
[12:23:48.716] 🅸 stream> SIRM_TRANSFER2_SIZE   = 0x00000000                                                                                                                                                                                   
[12:23:48.716] 🅸 stream> SIRM_MAX_LEADER_SIZE  = 0x00000400                                                                                                                                                                                   
[12:23:48.716] 🅸 stream> SIRM_MAX_TRAILER_SIZE = 0x00000400                                                                                                                                                                                   
                                                                                                                                                                                                                           
I am able to grab images, but I am getting the following error messages:

** (python:9975): CRITICAL **: 12:23:50.795: arv_buffer_get_part_pixel_format: assertion 'arv_buffer_part_is_image (buffer, part_id)' failed                                                                                                  
                                                                                                                                                                                                                                              
** (python:9975): CRITICAL **: 12:23:50.795: arv_buffer_get_part_height: assertion 'arv_buffer_part_is_image (buffer, part_id)' failed                                                                                                        
                                                                                                                                                                                                                                              
** (python:9975): CRITICAL **: 12:23:50.795: arv_buffer_get_part_width: assertion 'arv_buffer_part_is_image (buffer, part_id)' failed                                                                                                         
Callback[10004]                                                                                                                                                                                                                               
                                                                                                                                                                                                                                              
** (python:9975): CRITICAL **: 12:23:50.796: arv_buffer_get_part_pixel_format: assertion 'arv_buffer_part_is_image (buffer, part_id)' failed                                                                                                  
                                                                                                                                                                                                                                              
** (python:9975): CRITICAL **: 12:23:50.796: arv_buffer_get_part_height: assertion 'arv_buffer_part_is_image (buffer, part_id)' failed                                                                                                        
                                                                                                                                                                                                                                              
** (python:9975): CRITICAL **: 12:23:50.796: arv_buffer_get_part_width: assertion 'arv_buffer_part_is_image (buffer, part_id)' 

I have tried on 2 different machine with Ubuntu 20.04.1 LTS and different cameras.

EmmanuelP, 2023-11-27:
Could you attach a short sample software that shows your issue ?

ayushbisht2001, 2024-08-21:
@nagyilles Did you figure out a way to resolve this issue.. ? I'm also facing similar issue while using basler and allied vision cameras..

nagyilles, 2024-08-22:
@ayushbisht2001 Unfortunately not, but I did not worked on this code since and I did not see the EmmanuelP's message either. Could you answer his question?