AravisProject/aravis #859: Found 0 USB3Vision device even though the camera is plugged into the USB3 port

xulingtian, 2024-01-19:
**Describe the bug**
camera is plugged into the USB3 port, but not detected. Aravis compiled with  ```--enable-usb```

**Camera description:**
 - IDS Imaging Development Systems GmbH U3-356xXLE-C-Service (not detected in LaptopA)
 - different Basler Cameras (detected in Laptop A and B)
 - Allied Vision_ALVIUM 1800 U-1240m (detected in Laptop A, not detected in Laptop B)
 - all cameras use USB3.0

**Platform description:**
 - Aravis 0.8.28
 - OS: Ubuntu 22.04

**Testing Information**
For example with IDS Imaging Development Systems GmbH U3-356xXLE-C-Service:

 **arv-test-0.8 -d all** :

```
[16:19:03.604] 🅸 interface> Found 0 USB3Vision device (among 22 USB devices)
[16:19:03.605] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[16:19:03.605] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.24.5.31 (172.24.5.255)
[16:19:03.605] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.23.0.1 (172.23.0.15)
Found 0 device
```

**arv-tool-0.8 --debug=all**:
```
[16:21:50.205] 🅸 interface> Found 0 USB3Vision device (among 22 USB devices)
[16:21:50.205] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[16:21:50.205] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.24.5.31 (172.24.5.255)
[16:21:50.205] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.23.0.1 (172.23.0.15)
No device found
```

**lsusb -v**
```
Bus 004 Device 012: ID 1409:8000 IDS Imaging Development Systems GmbH U3-356xXLE-C-Service
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x1409 IDS Imaging Development Systems GmbH
  idProduct          0x8000 
  bcdDevice            0.00
  iManufacturer           1 IDS Imaging Development Systems GmbH
  iProduct                2 U3-356xXLE-C-Service
  iSerial                 3 4104220014
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005e
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration         11 USB3 Vision
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              488mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass        239 Miscellaneous Device
      bFunctionSubClass       5 USB3 Vision
      bFunctionProtocol       0 
      iFunction               6 USB3 Vision Device
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      0 
      iInterface             12 Control
      ** UNRECOGNIZED:  14 24 01 03 00 01 00 01 00 01 00 08 01 02 04 07 09 03 05 0c
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      1 
      iInterface             14 Event
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000006
      BESL Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000c
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   2
      Lowest fully-functional device speed is High Speed (480Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        2047 micro seconds

```

Thank you

xulingtian, 2024-02-01:
Hi @EmmanuelP Could you please review the problem?

EmmanuelP, 2024-02-02:
Hi,

The lsusb output strangely only shows a control and an event interface, no streaming interface. That explains why Aravis is not detecting it.

EmmanuelP, 2024-02-02:
I'm closing this issue as I don't know what to do about the missing streaming interface. Don't hesitate to reopen if you have additional informations that may help.

Cheers.

closed 2024-02-02 by EmmanuelP

xulingtian, 2024-02-08:
I still don't know why some cameras have no streaming interface in some laptops. But we find a "solution". I write it here for other users
For example "Allied Vision_ALVIUM 1800 U-1240m". After installation of the camera SDK VimbaX, and then the streaming interface is found. It seems like it is activated.  And then we can use aravis