AravisProject/aravis #863: I have tried the ip setting function, without error, but the ip has not set correctly

JimXu1989, 2024-01-31:
**Describe the bug**
I use the follow code to set gigEcam IP
ArvCamera *camera;
    ArvBuffer *buffer;
    GError *error = NULL;

    std::string filename="test.png";

    /* Connect to the first available camera */
    camera = arv_camera_new(NULL, &error);
    ArvGvIpConfigurationMode mode;
    arv_camera_gv_set_ip_configuration_mode (camera, ARV_GV_IP_CONFIGURATION_MODE_PERSISTENT_IP,&error);
    arv_camera_gv_set_persistent_ip_from_string(camera,"192.168.100.54","255.255.255.0","192.168.100.1",&error);

    GInetAddress *ip = NULL;
    GInetAddressMask *mask = NULL;
    GInetAddress *gateway = NULL;
    arv_camera_gv_get_persistent_ip(camera, &ip, &mask, &gateway, &error);

while, the IP did not change, I check the cam ip from the camera manage software provide by the manufacturer. 
How to set the IP and how to print it out?

Thanks a lot.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Camera description:**
 - Manufacturer Hik robot
 - Model 
 - Interface [GigE]

**Platform description:**
 - Aravis version 0.8.30
 - OS: [ubuntu 22.04]
 - Hardware [x86_64]

JimXu1989, 2024-01-31:
I read one of the solution:
https://aravis-project.discourse.group/t/is-it-possible-to-change-the-ip-of-a-connected-camera/226
But I do not know exactly how to code.

JimXu1989, 2024-01-31:
I use python package ipaddress to convert ip address to int, and set the camera with the following code, but the ip of camera still 192.168.100.3
arv_camera_set_integer(camera,"GevCurrentIPConfigurationDHCP",0,&error);
arv_camera_set_integer(camera,"GevCurrentIPConfigurationPersistentIP",1,&error);
arv_camera_set_integer(camera,"GevPersistentIPAddress",3215483944,&error);
arv_camera_set_integer(camera,"GevPersistentSubnetMask",4294967040,&error);

EmmanuelP, 2024-02-02:
Hi,

Are you able to use the camera and get a video stream in arv-viewer ?

JimXu1989, 2024-02-02:
> Hi,
> 
> Are you able to use the camera and get a video stream in arv-viewer ?

When I set the ip by the software provide by the camera manufacture correctly, I can get image from arv-viewer

EmmanuelP, 2024-02-02:
Did you try arv-tool ? For example:

```
arv-tool-0.8 network ip=192.168.0.1 mask=255.255.255.0 gateway=192.168.0.254
```

JimXu1989, 2024-02-03:
> Did you try arv-tool ? For example:
> 
> ```
> arv-tool-0.8 network ip=192.168.0.1 mask=255.255.255.0 gateway=192.168.0.254
> ```
it shows read_memory timeout
arv-tool-0.8 network ip=192.168.100.15 mask=255.255.255.0 gateway=192.168.100.254
GEV-SIA445-00E48218249 (192.16.100.6)
Failed to open device 'GEV-SIA445-00E48218249': GigEVision read_memory timeout

JimXu1989, 2024-02-03:
> Did you try arv-tool ? For example:
> 
> ```
> arv-tool-0.8 network ip=192.168.0.1 mask=255.255.255.0 gateway=192.168.0.254
> ```
My computer ip is 192.168.100.2
when I set the camera ip 192.16.100.6,it shows:
arv-tool-0.8 network ip=192.168.100.15 mask=255.255.255.0 gateway=192.168.100.254
GEV-SIA445-00E48218249 (192.16.100.6)
Failed to open device 'GEV-SIA445-00E48218249': GigEVision read_memory timeout

when I set the camera ip 192.168.100.6,it shows:
arv-tool-0.8 network ip=192.168.100.15 mask=255.255.255.0 gateway=192.168.100.254
GEV-SIA445-00E48218249 (192.168.100.6)

All the situation, the camera ip is not changed