AravisProject/aravis #865: ARV_GV_STREAM_SOCKET_BUFFER_AUTO causes frame timeout on small images

hackel6, 2024-02-01:
**Describe the bug**
When using ARV_GV_STREAM_SOCKET_BUFFER_AUTO option with small image sizes (images that are transferred in only one packet, a TIMEOUT is generated even though packets seem to be properly received. Removing the ARV_GV_STREAM_SOCKET_BUFFER_AUTO option makes everything work again (using default socket buffer size).

**To Reproduce**
Only tried this on a SI-1900 camera, but simply executing the arv-camera-tester using a small width and height and setting the '-a' option reproduces.

**Expected behavior**
Images are streamed and received as expected.

**Camera description:**
 - Manufacturer: Spectral Instruments
 - Model: 1900
 - Interface: GigE

**Platform description:**
 - Aravis version: 0.8.22
 - OS: Gentoo Linux
 - Hardware x86_64

**Additional context**

> arv-camera-test-0.8 -a --width=4 --height=4
Looking for the first available camera
vendor name            = Spectral Instruments, Inc.
model name             = 1900
device serial number   = XXX
image width            = 4
image height           = 4
horizontal binning     = 1
vertical binning       = 1
payload                = 32 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 0 ns
[16:11:42.628] ?? stream> [GvStream::stream_new] Destination stream port = 48540
[16:11:42.628] ?? stream> [GvStream::stream_new] Source stream port = 49344
[16:11:42.628] ?? stream> [GvStream::loop] Standard socket method
gv packet size         = 1500 bytes
[16:11:42.629] ?? stream-thread> [GvStream::update_socket] Socket buffer size set to 32
[16:11:42.629] ?? stream-thread> [GvStream::find_frame_data] Start frame 1
[16:11:42.629] ?? sp> packet_type  =       ok (0x0000)
                     content_type = data-leader (0x0001)
                     frame_id     =        1
                     packet_id    =        0
                     data_size    =       36
                     payload_type = image
                     pixel format = video/x-raw, format=(string)GRAY16_LE
                     width        = 4
                     height       = 4
                     x_offset     = 2046
                     y_offset     = 2054
  0 frame/s  -       0 MiB/s
[16:11:44.626] ?? stream-thread> [GvStream::check_frame_completion] Timeout for frame 1 at dt = 1996584
[16:11:44.626] ?? stream-thread> [GvStream::close_frame] Close frame 1
  0 frame/s  -       0 MiB/s - 1 error
  0 frame/s  -       0 MiB/s
[16:11:46.421] ?? stream-thread> [GvStream::find_frame_data] Start frame 2
[16:11:46.523] ?? stream-thread> [GvStream::check_frame_completion] Timeout for frame 2 at dt = 101852
[16:11:46.523] ?? stream-thread> [GvStream::close_frame] Close frame 2
  0 frame/s  -       0 MiB/s - 1 error
  0 frame/s  -       0 MiB/s
[16:11:48.317] ?? stream-thread> [GvStream::find_frame_data] Start frame 3
[16:11:48.419] ?? stream-thread> [GvStream::check_frame_completion] Timeout for frame 3 at dt = 101857
[16:11:48.419] ?? stream-thread> [GvStream::close_frame] Close frame 3
  0 frame/s  -       0 MiB/s - 1 error
  0 frame/s  -       0 MiB/s
[16:11:50.212] ?? stream-thread> [GvStream::find_frame_data] Start frame 4
[16:11:50.314] ?? stream-thread> [GvStream::check_frame_completion] Timeout for frame 4 at dt = 101867
[16:11:50.314] ?? stream-thread> [GvStream::close_frame] Close frame 4
^C  0 frame/s  -       0 MiB/s - 1 error
[... 72 lines cut ...]
                         <          0;           0;           0;           0
                         min         :      101852;           0; 1.89577e+06
                         max         : 1.99658e+06; 1.89577e+06; 1.89577e+06
                         last max at :           0;           1;           0
                         counter     :           4:           5:           1
[16:11:50.639] ?? stream> [GvStream::finalize] n_completed_buffers    = 0
[16:11:50.639] ?? stream> [GvStream::finalize] n_failures             = 4
[16:11:50.639] ?? stream> [GvStream::finalize] n_underruns            = 0
[16:11:50.639] ?? stream> [GvStream::finalize] n_timeouts             = 4
[16:11:50.639] ?? stream> [GvStream::finalize] n_aborteds             = 0
[16:11:50.640] ?? stream> [GvStream::finalize] n_missing_frames       = 0
[16:11:50.640] ?? stream> [GvStream::finalize] n_size_mismatch_errors = 0
[16:11:50.640] ?? stream> [GvStream::finalize] n_received_packets     = 5
[16:11:50.640] ?? stream> [GvStream::finalize] n_missing_packets      = 10
[16:11:50.640] ?? stream> [GvStream::finalize] n_error_packets        = 0
[16:11:50.640] ?? stream> [GvStream::finalize] n_ignored_packets      = 0
[16:11:50.640] ?? stream> [GvStream::finalize] n_resend_requests      = 0
[16:11:50.640] ?? stream> [GvStream::finalize] n_resent_packets       = 0
[16:11:50.640] ?? stream> [GvStream::finalize] n_resend_ratio_reached = 0
[16:11:50.640] ?? stream> [GvStream::finalize] n_resend_disabled      = 0
[16:11:50.640] ?? stream> [GvStream::finalize] n_duplicated_packets   = 0
[16:11:50.640] ?? stream> [GvStream::finalize] n_transferred_bytes    = 204
[16:11:50.641] ?? stream> [GvStream::finalize] n_ignored_bytes        = 0
[16:11:50.641] ?? stream> [Stream::finalize] Flush 50 buffer[s] in input queue
[16:11:50.641] ?? stream> [Stream::finalize] Flush 0 buffer[s] in output queue

EmmanuelP, 2024-02-02:
Hi,

Thanks. The auto buffer size code should have a lower limit in the value applied to the socket size.

hackel6, 2024-02-05:
Thanks for your reply. Any thoughts on what that lower limit should be?

EmmanuelP, 2024-02-08:
No idea yet. I have to check the code. The current one probably does not take into account the udp packet format cost.

closed 2024-03-11 by EmmanuelP

hackel6, 2024-03-19:
this appears to have worked! thanks!

PR #873 (2024-03-11): gvstream: put a lower limit to socket buffer size
Fix #865
commit b09b4ed8 (2024-03-11): gvstream: put a lower limit to socket buffer size

Fix #865
commit 248cebb1 (2024-03-12): gvstream: put a lower limit to socket buffer size

Fix #865