AravisProject/aravis #866: Internal data stream error (Aravis + GStreamer + Daheng USB3)

lamres, 2024-02-06:
**Describe the bug**
I have the Jetson Xavier NX hardware and Daheng MER2-503 USB3 camera.
Try to use aravis via gstreamer pipeline according to this [article](https://gist.github.com/nitheeshkl/5cbf1a0777801a7e9b8e12f8252d465e), but my code got this error "Internal data stream error".
[image]

**To Reproduce**
Steps to reproduce the behavior [here](https://gist.github.com/nitheeshkl/5cbf1a0777801a7e9b8e12f8252d465e).

Aravis works fine.
[image]

GStreamer plugin is also installed.

**Expected behavior**
I'd like to pipe a stream from USB3 camera into the gstreamer's pipeline.

**Camera description:**
 - Manufacturer: Daheng
 - Model: MER2-503
 - Interface: USB3

**Platform description:**
 - Aravis version: 0.10
 - OS: Ubuntu 20.04 LTS
 - Hardware: Jetson Xavier NX
 - JetPack: 5.1.2

EmmanuelP, 2024-02-06:
Try `gst-launch-1.0 aravissrc ! video/x-raw,format=GRAY8 ! videoconvert ! autovideosink`

lamres, 2024-02-06:
@EmmanuelP it looks like this

m2m@m2m-desktop:~$ ARV_DEBUG=all gst-launch-1.0 aravissrc ! video/x-raw,format=GRAY8 ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
[17:36:21.883] 🅸 interface> Found 1 USB3Vision device (among 7 USB devices)
[17:36:21.909] 🅸 device> [UvDevice::new] GUID    = 2BA200005949
[17:36:21.912] 🅸 device> [UvDevice::new] Using control endpoint 1, interface 0
[17:36:21.912] 🅸 device> [UvDevice::new] Using data endpoint 3, interface 2
[17:36:21.913] 🅸 device> Get genicam
[17:36:21.914] 🅸 device> MANUFACTURER_NAME =        'Daheng Imaging'
[17:36:21.920] 🅸 device> MAX_DEVICE_RESPONSE_TIME = 0x0000012c
[17:36:21.920] 🅸 device> DEVICE_CAPABILITY        = 0x0000000000000f0d
[17:36:21.920] 🅸 device> SRBM_ADDRESS =             0x0000000000010000
[17:36:21.921] 🅸 device> MANIFEST_TABLE_ADDRESS =   0x0000000000034000
[17:36:21.926] 🅸 device> U3VCP_CAPABILITY =         0x0000000000000003
[17:36:21.926] 🅸 device> MAX_CMD_TRANSFER =         0x00000400
[17:36:21.926] 🅸 device> MAX_ACK_TRANSFER =         0x00000400
[17:36:21.926] 🅸 device> SIRM_OFFSET =              0x0000000000020000
[17:36:21.937] 🅸 device> SIRM_INFO =                0x0a000000
[17:36:21.937] 🅸 device> SIRM_CONTROL =             0x00000001
[17:36:21.938] 🅸 device> SIRM_REQ_PAYLOAD_SIZE =    0x00000000004c8000
[17:36:21.938] 🅸 device> SIRM_REQ_LEADER_SIZE =     0x00000034
[17:36:21.938] 🅸 device> SIRM_REQ_TRAILER_SIZE =    0x00000020
[17:36:21.938] 🅸 device> SIRM_MAX_LEADER_SIZE =     0x00000400
[17:36:21.939] 🅸 device> SIRM_PAYLOAD_SIZE =        0x00100000
[17:36:21.939] 🅸 device> SIRM_PAYLOAD_COUNT =       0x00000004
[17:36:21.939] 🅸 device> SIRM_TRANSFER1_SIZE =      0x000c8000
[17:36:21.939] 🅸 device> SIRM_TRANSFER2_SIZE =      0x00000000
[17:36:21.940] 🅸 device> SIRM_MAX_TRAILER_SIZE =    0x00000400
[17:36:21.942] 🅸 device> MANIFEST_N_ENTRIES =       0x0000000000000001
[17:36:21.942] 🅸 device> MANIFEST ENTRY
                         00000000 02 00 04 01 00 04 01 01 00 00 d0 00 00 00 00 00  ................
                         00000010 a8 9e 00 00 00 00 00 00 15 55 df 0c 23 8e f4 08  .........U..#...
                         00000020 c6 2d 6f f5 82 21 ca 87 42 fb 56 f0 00 00 00 00  .-o..!..B.V.....
                         00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[17:36:21.942] 🅸 device> genicam address =          0x0000000000d00000
[17:36:21.943] 🅸 device> genicam size    =          0x0000000000009ea8
[17:36:22.820] 🅸 device> zip file =                 MER-U3X_XML_V1.4.2.xml
[17:36:23.072] 🆆 misc> [PixelFormat::to_gst_caps_string] 0x0110000d not found
Pipeline is live and does not need PREROLL ...
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis0:
streaming stopped, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...

EmmanuelP, 2024-02-06:
`arv-tool-0.8 features PixelFormat` ?

lamres, 2024-02-06:
@EmmanuelP the output is next

`Daheng Imaging-2BA200005949-FCY22120557 (USB3)
        Enumeration  : [RW] 'PixelFormat'
            EnumEntry   : 'BayerRG10'
            EnumEntry   : 'BayerRG8'`

alvinahmadov, 2024-02-07:
Hi @EmmanuelP.

I have the same problems with streams.
Sometimes stream not starting. Especially when I develop an rtsp streaming app.

Thanks!

alvinahmadov, 2024-02-07:
The output:

`Looking for the first available camera
[12:57:13.343] 🅸 interface> Found 1 USB3Vision device (among 7 USB devices)
[12:57:13.365] 🅸 device> [UvDevice::new] GUID    = 2BA200005949
[12:57:13.367] 🅸 device> [UvDevice::new] Using control endpoint 1, interface 0
[12:57:13.367] 🅸 device> [UvDevice::new] Using data endpoint 3, interface 2
[12:57:13.368] 🅸 device> Get genicam
[12:57:13.370] 🅸 device> MANUFACTURER_NAME =        'Daheng Imaging'
[12:57:13.373] 🅸 device> MAX_DEVICE_RESPONSE_TIME = 0x0000012c
[12:57:13.374] 🅸 device> DEVICE_CAPABILITY        = 0x0000000000000f0d
[12:57:13.374] 🅸 device> SRBM_ADDRESS =             0x0000000000010000
[12:57:13.374] 🅸 device> MANIFEST_TABLE_ADDRESS =   0x0000000000034000
[12:57:13.378] 🅸 device> U3VCP_CAPABILITY =         0x0000000000000003
[12:57:13.378] 🅸 device> MAX_CMD_TRANSFER =         0x00000400
[12:57:13.378] 🅸 device> MAX_ACK_TRANSFER =         0x00000400
[12:57:13.379] 🅸 device> SIRM_OFFSET =              0x0000000000020000
[12:57:13.388] 🅸 device> SIRM_INFO =                0x0a000000
[12:57:13.388] 🅸 device> SIRM_CONTROL =             0x00000000
[12:57:13.388] 🅸 device> SIRM_REQ_PAYLOAD_SIZE =    0x00000000004c8000
[12:57:13.389] 🅸 device> SIRM_REQ_LEADER_SIZE =     0x00000034
[12:57:13.389] 🅸 device> SIRM_REQ_TRAILER_SIZE =    0x00000020
[12:57:13.389] 🅸 device> SIRM_MAX_LEADER_SIZE =     0x00000400
[12:57:13.389] 🅸 device> SIRM_PAYLOAD_SIZE =        0x00100000
[12:57:13.389] 🅸 device> SIRM_PAYLOAD_COUNT =       0x00000004
[12:57:13.389] 🅸 device> SIRM_TRANSFER1_SIZE =      0x000c8000
[12:57:13.390] 🅸 device> SIRM_TRANSFER2_SIZE =      0x00000000
[12:57:13.390] 🅸 device> SIRM_MAX_TRAILER_SIZE =    0x00000400
[12:57:13.392] 🅸 device> MANIFEST_N_ENTRIES =       0x0000000000000001
[12:57:13.393] 🅸 device> MANIFEST ENTRY
                         00000000 02 00 04 01 00 04 01 01 00 00 d0 00 00 00 00 00  ................
                         00000010 a8 9e 00 00 00 00 00 00 15 55 df 0c 23 8e f4 08  .........U..#...
                         00000020 c6 2d 6f f5 82 21 ca 87 42 fb 56 f0 00 00 00 00  .-o..!..B.V.....
                         00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[12:57:13.393] 🅸 device> genicam address =          0x0000000000d00000
[12:57:13.393] 🅸 device> genicam size    =          0x0000000000009ea8
[12:57:14.267] 🅸 device> zip file =                 MER-U3X_XML_V1.4.2.xml
vendor        = Daheng Imaging
model         = MER2-503-36U3C

** (arv-device-test:6068): CRITICAL **: 12:57:14.426: arv_gc_string_get_value: assertion 'ARV_IS_GC_STRING (gc_string)' failed
device id     = (null)
sensor width  = 2448
sensor height = 2048
image width   = 2448 (max:2448)
image height  = 2048 (max:2048)
horizontal binning  = 1 (min:1 - max:4)
vertical binning    = 1 (min:1 - max:4)
exposure auto mode  = Continuous
gain auto mode      = Off
trigger selector    = FrameStart
reverse x           = FALSE
payload size  = 5013504 (0x4c8000)
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
^CFrame rate = 0 Hz
Processed buffers = 0
Failures          = 0
Underruns         = 0
[12:57:40.545] 🅸 stream> [UvStream::finalize] n_completed_buffers    = 0
[12:57:40.545] 🅸 stream> [UvStream::finalize] n_failures             = 0
[12:57:40.546] 🅸 stream> [UvStream::finalize] n_underruns            = 0
[12:57:40.546] 🅸 stream> [UvStream::finalize] n_aborted              = 0
[12:57:40.546] 🅸 stream> [UvStream::finalize] n_transferred_bytes    = 0
[12:57:40.546] 🅸 stream> [UvStream::finalize] n_ignored_bytes        = 0
[12:57:40.547] 🅸 stream> [Stream::delete_buffers] Delete 30 buffer[s] in input queue
[12:57:40.547] 🅸 stream> [Stream::delete_buffers] Delete 0 buffer[s] in output queue`

alvinahmadov, 2024-02-07:
Output of arv-device-test below:

```
Looking for the first available camera
[12:57:13.343] 🅸 interface> Found 1 USB3Vision device (among 7 USB devices)
[12:57:13.365] 🅸 device> [UvDevice::new] GUID    = 2BA200005949
[12:57:13.367] 🅸 device> [UvDevice::new] Using control endpoint 1, interface 0
[12:57:13.367] 🅸 device> [UvDevice::new] Using data endpoint 3, interface 2
[12:57:13.368] 🅸 device> Get genicam
[12:57:13.370] 🅸 device> MANUFACTURER_NAME =        'Daheng Imaging'
[12:57:13.373] 🅸 device> MAX_DEVICE_RESPONSE_TIME = 0x0000012c
[12:57:13.374] 🅸 device> DEVICE_CAPABILITY        = 0x0000000000000f0d
[12:57:13.374] 🅸 device> SRBM_ADDRESS =             0x0000000000010000
[12:57:13.374] 🅸 device> MANIFEST_TABLE_ADDRESS =   0x0000000000034000
[12:57:13.378] 🅸 device> U3VCP_CAPABILITY =         0x0000000000000003
[12:57:13.378] 🅸 device> MAX_CMD_TRANSFER =         0x00000400
[12:57:13.378] 🅸 device> MAX_ACK_TRANSFER =         0x00000400
[12:57:13.379] 🅸 device> SIRM_OFFSET =              0x0000000000020000
[12:57:13.388] 🅸 device> SIRM_INFO =                0x0a000000
[12:57:13.388] 🅸 device> SIRM_CONTROL =             0x00000000
[12:57:13.388] 🅸 device> SIRM_REQ_PAYLOAD_SIZE =    0x00000000004c8000
[12:57:13.389] 🅸 device> SIRM_REQ_LEADER_SIZE =     0x00000034
[12:57:13.389] 🅸 device> SIRM_REQ_TRAILER_SIZE =    0x00000020
[12:57:13.389] 🅸 device> SIRM_MAX_LEADER_SIZE =     0x00000400
[12:57:13.389] 🅸 device> SIRM_PAYLOAD_SIZE =        0x00100000
[12:57:13.389] 🅸 device> SIRM_PAYLOAD_COUNT =       0x00000004
[12:57:13.389] 🅸 device> SIRM_TRANSFER1_SIZE =      0x000c8000
[12:57:13.390] 🅸 device> SIRM_TRANSFER2_SIZE =      0x00000000
[12:57:13.390] 🅸 device> SIRM_MAX_TRAILER_SIZE =    0x00000400
[12:57:13.392] 🅸 device> MANIFEST_N_ENTRIES =       0x0000000000000001
[12:57:13.393] 🅸 device> MANIFEST ENTRY
                         00000000 02 00 04 01 00 04 01 01 00 00 d0 00 00 00 00 00  ................
                         00000010 a8 9e 00 00 00 00 00 00 15 55 df 0c 23 8e f4 08  .........U..#...
                         00000020 c6 2d 6f f5 82 21 ca 87 42 fb 56 f0 00 00 00 00  .-o..!..B.V.....
                         00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[12:57:13.393] 🅸 device> genicam address =          0x0000000000d00000
[12:57:13.393] 🅸 device> genicam size    =          0x0000000000009ea8
[12:57:14.267] 🅸 device> zip file =                 MER-U3X_XML_V1.4.2.xml
vendor        = Daheng Imaging
model         = MER2-503-36U3C

** (arv-device-test:6068): CRITICAL **: 12:57:14.426: arv_gc_string_get_value: assertion 'ARV_IS_GC_STRING (gc_string)' failed
device id     = (null)
sensor width  = 2448
sensor height = 2048
image width   = 2448 (max:2448)
image height  = 2048 (max:2048)
horizontal binning  = 1 (min:1 - max:4)
vertical binning    = 1 (min:1 - max:4)
exposure auto mode  = Continuous
gain auto mode      = Off
trigger selector    = FrameStart
reverse x           = FALSE
payload size  = 5013504 (0x4c8000)
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
^CFrame rate = 0 Hz
Processed buffers = 0
Failures          = 0
Underruns         = 0
[12:57:40.545] 🅸 stream> [UvStream::finalize] n_completed_buffers    = 0
[12:57:40.545] 🅸 stream> [UvStream::finalize] n_failures             = 0
[12:57:40.546] 🅸 stream> [UvStream::finalize] n_underruns            = 0
[12:57:40.546] 🅸 stream> [UvStream::finalize] n_aborted              = 0
[12:57:40.546] 🅸 stream> [UvStream::finalize] n_transferred_bytes    = 0
[12:57:40.546] 🅸 stream> [UvStream::finalize] n_ignored_bytes        = 0
[12:57:40.547] 🅸 stream> [Stream::delete_buffers] Delete 30 buffer[s] in input queue
[12:57:40.547] 🅸 stream> [Stream::delete_buffers] Delete 0 buffer[s] in output queue
```

lamres, 2024-02-07:
@EmmanuelP  the current message is 

> [PixelFormat::to_gst_caps_string] 0x0110000d not found

I checked the code of Aravis and found this
[image]

that is similar to our formats

> 'BayerRG10’, 'BayerRG8'

What is the best next step to investigate it?

EmmanuelP, 2024-02-08:
Gstreamer videoconvert element does not take bayer formats as its sink. You have to insert a bayer converter, something like:

```
gst-launch-1.0 aravissrc ! video/x-bayer,format=rggb ! bayer2rgb ! videoconvert ! autovideosink
```

lamres, 2024-02-10:
@EmmanuelP 
It's like the same one

$ ARV_DEBUG=all gst-launch-1.0 aravissrc ! video/x-bayer,format=rggb ! bayer2rgb ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
[21:03:12.322] 🅸 interface> Found 1 USB3Vision device (among 7 USB devices)
[21:03:12.352] 🅸 device> [UvDevice::new] GUID    = 2BA200005949
[21:03:12.354] 🅸 device> [UvDevice::new] Using control endpoint 1, interface 0
[21:03:12.354] 🅸 device> [UvDevice::new] Using data endpoint 3, interface 2
[21:03:12.355] 🅸 device> Get genicam
[21:03:12.356] 🅸 device> MANUFACTURER_NAME =        'Daheng Imaging'
[21:03:12.360] 🅸 device> MAX_DEVICE_RESPONSE_TIME = 0x0000012c
[21:03:12.360] 🅸 device> DEVICE_CAPABILITY        = 0x0000000000000f0d
[21:03:12.361] 🅸 device> SRBM_ADDRESS =             0x0000000000010000
[21:03:12.361] 🅸 device> MANIFEST_TABLE_ADDRESS =   0x0000000000034000
[21:03:12.364] 🅸 device> U3VCP_CAPABILITY =         0x0000000000000003
[21:03:12.365] 🅸 device> MAX_CMD_TRANSFER =         0x00000400
[21:03:12.365] 🅸 device> MAX_ACK_TRANSFER =         0x00000400
[21:03:12.365] 🅸 device> SIRM_OFFSET =              0x0000000000020000
[21:03:12.375] 🅸 device> SIRM_INFO =                0x0a000000
[21:03:12.375] 🅸 device> SIRM_CONTROL =             0x00000000
[21:03:12.375] 🅸 device> SIRM_REQ_PAYLOAD_SIZE =    0x00000000004c8000
[21:03:12.375] 🅸 device> SIRM_REQ_LEADER_SIZE =     0x00000034
[21:03:12.376] 🅸 device> SIRM_REQ_TRAILER_SIZE =    0x00000020
[21:03:12.376] 🅸 device> SIRM_MAX_LEADER_SIZE =     0x00000400
[21:03:12.376] 🅸 device> SIRM_PAYLOAD_SIZE =        0x00100000
[21:03:12.376] 🅸 device> SIRM_PAYLOAD_COUNT =       0x00000004
[21:03:12.376] 🅸 device> SIRM_TRANSFER1_SIZE =      0x000c8000
[21:03:12.376] 🅸 device> SIRM_TRANSFER2_SIZE =      0x00000000
[21:03:12.377] 🅸 device> SIRM_MAX_TRAILER_SIZE =    0x00000400
[21:03:12.379] 🅸 device> MANIFEST_N_ENTRIES =       0x0000000000000001
[21:03:12.379] 🅸 device> MANIFEST ENTRY
                         00000000 02 00 04 01 00 04 01 01 00 00 d0 00 00 00 00 00  ................
                         00000010 a8 9e 00 00 00 00 00 00 15 55 df 0c 23 8e f4 08  .........U..#...
                         00000020 c6 2d 6f f5 82 21 ca 87 42 fb 56 f0 00 00 00 00  .-o..!..B.V.....
                         00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[21:03:12.380] 🅸 device> genicam address =          0x0000000000d00000
[21:03:12.380] 🅸 device> genicam size    =          0x0000000000009ea8
[21:03:13.259] 🅸 device> zip file =                 MER-U3X_XML_V1.4.2.xml
[21:03:13.532] 🆆 misc> [PixelFormat::to_gst_caps_string] 0x0110000d not found
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Could not set caps on camera "": No stream to stop
Additional debug info:
../gst/gstaravis.c(402): gst_aravis_set_caps (): /GstPipeline:pipeline0/GstAravis:aravis0
Execution ended after 0:00:00.000854181
Setting pipeline to NULL ...
Freeing pipeline ...

kamatsuoka, 2025-05-15:
fyi the `video/x-bayer,format=rggb ! bayer2rgb` converters also got my Flir Blackfly S working