AravisProject/aravis #870: Value for constant  ARV_PIXEL_FORMAT_MONO_10_PACKED is wrong

schneider-coder, 2024-02-14:
**Describe the bug**
I am using a Basler camera which is set to pixelformat MONO10p.
When I am accessing the camera using the aravis client I get the right value 0x010A0046 according to the spec.

https://www.emva.org/wp-content/uploads/GenICam_PFNC_1_1_01.pdf

In the code the value of this pixelformat MONO10p is set to 0x010c0004 which is not defined in the spec a all.

`#define ARV_PIXEL_FORMAT_MONO_10_PACKED		((ArvPixelFormat) 0x010c0004u)`

So, the bottom line is, the value I get from the camera is correct and the value defined in ARV_PIXEL_FORMAT_MONO_10_PACKED is wrong. The value 0x010c0004 doesn't even exist in the spec.

**To Reproduce**
Either check the spec and compare the values or connect to a GIGE Vision cam which is set to pixelformat MONO10p and check the value you get using 

`ArvPixelFormat pixelFormat = arv_buffer_get_image_pixel_format(buffer);`

**Expected behavior**
The value for the constant must be this
`#define ARV_PIXEL_FORMAT_MONO_10_PACKED		((ArvPixelFormat) 0x010A0046u)`

**Camera description:**
 - Basler
 - ace2
 - GIGE Vision

**Platform description:**
 - Aravais sdk v0.8
 - OS: Ubuntu22
 - x86_64

charliebudd, 2024-04-19:
Just bumped into this too, I just switched to using the string `Mono10p`. It's a shame the standard changed something like this, I'm sure there was a good reason. I don't think there is anything in the camera to indicate which version of the standard was used. 

I guess a fix could be implemented in the aravis API. `ARV_PIXEL_FORMAT_MONO_10_PACKED_LEGACY` could be defined and If setting to `ARV_PIXEL_FORMAT_MONO_10_PACKED`, `arv_buffer_set_image_pixel_format` could catch the error if thrown and try the legacy value before returning the error to the user.

charliebudd, 2024-04-22:
Actually, my camera at least exposes some versioning info for the standard in use...
```
DeviceGenCPVersionMajor [Integer]
DeviceGenCPVersionMinor [Integer]
DeviceU3VVersionMajor [Integer]
DeviceU3VVersionMinor [Integer]
```
If exposing this is part of the standard, it could be a way to determine which value should be used.